MichaIng/DietPi #5085: Cannot add dietpi.txt, dietpi-wifi.txt, Automation_Custom_Script.sh

tolidano, 2021-12-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.8
- Distro version | bullsye
- Kernel version | Linux DietPi 4.4.143-65-rockchip-g58431d38f8f3 #1 SMP PREEMPT Sat Aug 14 09:31:07 UTC 2021 aarch64 GNU/Linux
- SBC model | ROCK Pi S (aarch64)
- Power supply used | Laptop USBC
- SDcard used | On-board NAND Flash

#### Additional Information (if applicable)
Unlike normal installation to SD Card, Rock PI S NAND Flash uses rkdeveloptool to write .img to NAND Flash. 
I am unable to mount the NAND Flash after the image is written and have to reboot.
I am attempting to write dietpi.txt, dietpi-wifi.txt, and Automation_Custom_Script.sh to the img before flashing the NAND.
See here:
```
% mount | grep DietPi
/home/tolidano/Downloads/DietPi_ROCKPiS-ARMv8-Bullseye.img on /home/tolidano/Downloads/dpi type ext4 (rw,relatime)
tolidano@1mix4 ~/Downloads/dpi/boot
 % ls -al
total 32
drwxr-xr-x  2 root root  4096 Dec 13 00:03 .
drwxr-xr-x 18 root root  4096 Oct 22 23:07 ..
-rw-r--r--  1 root root  1570 Dec 13 00:03 Automation_Custom_Script.sh
-rw-r--r--  1 root root 15038 Dec 12 23:27 dietpi.txt
-rw-r--r--  1 root root   257 Dec 12 23:58 dietpi-wifi.txt
```
Among other changes to dietpi.txt, I did this, let us imagine this was the only change:
```
 % cat dietpi.txt | grep ^AUTO_SETUP_AUTOMATED        
AUTO_SETUP_AUTOMATED=1
```
Along with setting the root/dietpi password and:
```
AUTO_SETUP_NET_WIFI_ENABLED=1
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=US
```
With dietpi-wifi.txt file having the details for my wifi connection.

So upon reboot, the device does not join the wifi, and once I SSH in, I am presented with the license and the normal installation flow.

The boot folder looks like this: 
```
root@DietPi:/boot# ls
FSCK0000.REC				      config-4.4.143-65-rockchip-g58431d38f8f3	dietpi-wifi.txt  initrd.img-4.4.143-65-rockchip-g58431d38f8f3
System.map-4.4.143-65-rockchip-g58431d38f8f3  dietpi					dietpi.txt	 uEnv.txt
boot.cmd				      dietpi-LICENSE.txt			dtbs		 vmlinuz-4.4.143-65-rockchip-g58431d38f8f3
boot.scr				      dietpi-README.md				extlinux
```
And the dietpi-wifi.txt file is empty and the dietpi.txt file is stock.

#### Steps to reproduce
1. Mount image
2. Write dietpi.txt and dietpi-wifi.txt to /boot
3. Flash image
4. Boot device

#### Expected behaviour
- Device should pick up dietpi.txt and dietpi-wifi.txt in /boot (along with Automation_Custom_Script.sh)

#### Actual behaviour
- None of the settings in dietpi.txt or dietpi-wifi.txt are honored, files on Rock Pi S are stock files with defaults

#### Extra details
- dmesg shows no interesting messages.

tolidano, 2021-12-16:
I am attempting to write the img to an SD card using dd to see if this changes anything and to examine the /boot folder at that point (before a boot takes place)

tolidano, 2021-12-16:
I have tried to update the files to the ones generated in the /boot folder after the install as well as create the dietpi folder and touch a file called .installed which appeared to be desired by the installer. I copied the dietpi.txt and dietpi-wifi.txt files into the dietpi folder as well (but not the Automation file) and reflashed that image with the network plugged in.

tolidano, 2021-12-16:
This attempt also failed - there is no dietpi.txt in the boot/dietpi folder, and there are the stock files again in /boot so I am at a loss.

MichaIng, 2021-12-17:
Many thanks for your report.

You were/are not skeptical that the `/boot` directory is completely empty where you add your custom configs, especially no kernel, initramfs or anything else necessary to boot, and after you boot it, all this is magically there? 😄

You need to mount and add your configs to the first FAT partition of the image, where all those files are present already (aside of `Automation_Custom_Script.sh`.

tolidano, 2021-12-17:
I was interested in how that happens, and scanned the entire image thinking you copied it in from somewhere. It did not occur to me to check the other partition. Will give that a shot and report back. Thanks!

tolidano, 2021-12-18:
Ok, so great news. By installing this, it mostly did the thing. I flashed the NAND with the modified image and called rkdeveloptool rd to reset the device. It did configure the wifi automatically and join the wifi network, which was good.

However, once I connected to the device, only then did it proceed to complete the installation steps (in an automated fashion) - the update of diet pi, the switch to OpenSSH, the removal of analytics data and unused software, all took place in the tty when I first logged in (it did not prompt me for the license, and did complete all of these steps and execute the Automation script on it's own, though). 

I also learned the very important caveat that it could not handle Wifi AND Ethernet (I tried to turn on both in dietpi.txt, there is clearly a comment that says this will not work, although I don't understand why, it would be very helpful to have both, and it is possible to have both after the initial setup, so I'm confused here).

Overall, this is very close, and I will take the opportunity to thank you for creating/maintaining this great piece of software and for your prompt and helpful reply.

MichaIng, 2021-12-18:
The limitation of WiFi + Ethernet indeed is an issue. Our network setup done via `dietpi-config` and on first boot really is a very simple one, enabling an interface for the first found adapter of this type, in case the first one with an attached cable, and enabling internet access through it via DHCP or static IP/gateway info based on `dietpi.txt` resp. dialog/menu choice. When attempting this for two adapters (WiFi + Ethernet), naturally this causes issues as the system has either two gateways, no knowing there to send packets/requests to, or one interface config invalidates/overrides the other one. For a WiFi Hotspot + Ethernet internet setup we have the related "software" option in `dietpi-software`, but again with the limitation of being only WiFi hotspot + Ethernet internet and not WiFi + WiFi or Ethernet + Ethernet or Ethernet on LAN side and WiFi for internet. Another request often is using one interface as main internet interface and a second as fallback, like WiFi fallback when the cable gets disconnected.

In such more specific network setup aims, one needs to do it manually, via `/etc/network/intefaces.d/` or using a different network stack or even GUI, where plenty of options are available on Debian. For the fallback setup we have some guide on the forum, bug I need to dig it out when this is what you want to do.

We btw just had an internal talk that reworking our network setup tools/scripts to be more compatible and flexible is the first priority change for next DietPi versions, though I should be careful to make promises here. DietPi v8.0 in January will do mostly bug fixes and removal of Stretch support code overhead, but I hope to be able to start with a new `dietpi-network` tool (replacing the parts currently embedded in `dietpi-config`) for v8.1 in February.

closed 2022-02-20 by MichaIng