MichaIng/DietPi #5086: Is it possible let the services restart regularly?

cherrlee, 2021-12-16:
Is it possible let the services restart regularly.?
I found that I lost smb service every 48 hours and it would be back after I restart the samba service.

Joulinar, 2021-12-16:
Hi,

for a daily restart, you could put the service restart to `crontab` or create an own restart script and place it into `/etc/cron.daily`

MichaIng, 2021-12-17:
You can also define a custom 48h/2d schedule via `/etc/cron.d`. But what you plan is no solution, fragile and more like hammering against the symptoms 😉. I highly recommend to investigate why the Samba server is failing:
```sh
journalctl -u smbd -u nmbd
```

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng