MichaIng/DietPi #5087: qemu guest agent shutdown

jcz1, 2021-12-16:
Hello.

I'm not DietPi user, but i found this thread: https://dietpi.com/phpbb/viewtopic.php?t=9622

I think it is this bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1001795

Please write some info to forum, i don't have an account. Thanks.

J.K.

MichaIng, 2021-12-17:
Thanks for the heads up, however, nope this is not related. The problem was not an invalid/falsely parsed shutdown command but that the `qemu-guest-agent` since Bullseye relies on `dbus`/`systemd-logind` running in the guest, without having those explicitly defined as dependencies. Installing `dbus` and unmasking/starting `systemd-logind` in the guest fixed it.

closed 2021-12-17 by MichaIng

MichaIng, 2022-02-26:
Btw, I reported it in the meantime: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1004943
I believe whether systemd-logind is used depends on whether systemd is the init system, e.g. by checking whether /run/systemd/system exists. The latter however does not imply that dbus is installed or systemd-logind hence running, so that check could be enhanced in qemu-guest-agent. As stated by the Debian package maintainer, probably the new version from Bullseye backports solves it. To test, within the guest, run:
```sh
apt install -t bullseye-backports qemu-guest-agent
# and to mask again systemd-logind, if it was unmasked before
systemctl mask --now systemd-logind
```