MichaIng/DietPi #5089: Dietpi backup to external usb fails - claims read only system

helkav, 2021-12-16:
copied from system error report:

#### Details:
- Date           | Thu Dec 16 20:34:42 GMT 2021
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux 28g 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `touch /mnt/dietpi_x/dietpi-backup/data`
- Exit code      | 1
- Software title | DietPi-Backup
#### Steps to reproduce:

1. run dietpi-backup after setting up target location
#### Expected behaviour:

backup should be made to target location on USB?
#### Actual behaviour:

#### Extra details:
error about drive being readonly
- ...
#### Additional logs:
```
touch: setting times of '/mnt/dietpi_x/dietpi-backup/data': Read-only file system

helkav, 2021-12-16:
I have found a workaround. drive was RO when I checked in /etc/fstab but this drive was setup by drive_manager - I couldn't change to RW there so had to manually edit /etc/fstab

closed 2021-12-16 by helkav

MichaIng, 2021-12-17:
> I have found a workaround. drive was RO when I checked in /etc/fstab

You mean there was the `ro` option in `/etc/fstab`? There is a `Read Only` option in `dietpi-drive_manager` when you select that drive. But default is `rw`. If `ro` got applied automatically, this could indicate a filesystem corruption I/O error. So if you did not manually remounted it R/O or selected that option, I suggest to check for kernel errors, just to detect issues better earlier than later:
```sh
dmesg -l emerg,alert,crit,err
```

helkav, 2021-12-17:
This is interesting because at the moment I am having issues with dietpi-drive_manager and NTFS drives. I set up and next time I go in the drives aren't seen. I connect to windows 10 on my laptop and do a chkdsk and fix but dietpi repeats. I am at the point where I am going to set the drives up and manually write the fstab lines. I have a Ubuntu dual boot on my laptop so using that to make sure drives work. Unfortunately only one usb3 port on laptop 😢 so will have to do one at a time. There was a windows reserved partition at beginning of 4tb drive which I deleted because I didn't know what it was for. However I then tried to resize the following partition into that one. Thought it would be a quick job. Major mistake. Now I think I will have to cancel.

I have noticed some issues around NTFS and drive_manager in update although they seemed resolved? I wondered if the issue was linux kernal upgrade although Ubuntu would seem to suggest otherwise

MichaIng, 2021-12-18:
Did you compare the `/etc/fstab` entry for NTFS drives between how `dietpi-drive_manager` creates them and how you do? I'm not aware of any issues with them currently, but NTFS generally has limitations on Linux.

Expanding a filesystem to the start of a drive is indeed not trivially possible, but only expanding it to the end of a drive. But it depends on the tools, not sure whether/how it works with `ntfsresize`. Otherwise copy the data away, reformat it with a single partition/filesystem and move data back.

helkav, 2021-12-18:
Inline:
_________________

On 18 December 2021 00:05:44 GMT, MichaIng ***@***.***> wrote:
>Did you compare the `/etc/fstab` entry for NTFS drives between how
>`dietpi-drive_manager` creates them and how you do? I'm not aware of
>any issues with them currently, but NTFS generally has limitations on
>Linux.

No sorry I should have. At the moment I haven't proved any issue with dietpi-drive_manager and wouldn't be able to until comparing as you say

> Expanding a filesystem to the start of a drive is indeed not trivially
> possible, but only expanding it to the end of a drive. But it depends
> on the tools, not sure whether/how it works with [...]

I realised after setting this in motion that's what I should have done. Still I can cancel and if the data is corrupted I have elsewhere. Then I will do as you suggest above.

Thanks for your prompt replies on this. Dietpi is truly fab. I will have to setup a donation.

-- 
Sent from my Android device with monocles mail. Please excuse my brevity.

MichaIng, 2021-12-18:
It is definitely possible that mount options used by `dietpi-drive_manager` can be improved, we just need to find out whether this is really an issue our end or a general limitation with NTFS drives, and whether e.g. your mount options work fine/better in your case but cause issues or have downsides in other cases. Finally allowing to add/change mount options would be a nice feature 🙂.

About resizing filesystems in general, note that DietPi uses original Debian/GNU tools for this, where offered, so in case of bugs, we may forward them to the Debian bug tracker, and in case of limitations, those may be out of our control 😉.