MichaIng/DietPi #5090: Segmentation Fault with dig (causing pihole failure)

eng3, 2021-12-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B+ (armv7l)
- Power supply used | 5V 2A
- SDcard used | Unknown

#### Additional Information (if applicable)
- Software title | dig
- Was the software title installed freshly or updated/migrated? freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | 98c88e12-407e-4b95-83c2-ddf1f0dd2cac

#### Steps to reproduce

1. Install fresh DietPi 32-bit on rpi3 b+
2. Install unbound, pihole, wireguard
3. pihole fails to start based on dig error
4. running 'dig' results in "Segmentation Fault" immediately with no other message

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-12-17:
Many thanks for your report.

Tested and I wasn't able to replicate on an RPi ARMv7 Bullseye image, `dig` generally works fine on it.

I checked your bug report upload:
```
[    9.834837] Under-voltage detected! (0x00050005)
```
It is not necessarily related, but it can be the reason why the `dig` command got corrupted in the first place. Please try either switch the PSU, or the power cable to get those under-voltage warnings resolved. You can check via:
```sh
dmesg | grep voltage
```
or
```sh
vcgencmd get_throttled
```
_which should return `throttled=0x0` when everything is fine._

When done, you can try to repair `dig` by reinstalling the related APT packages:
```sh
apt update
apt install --reinstall bind9-dnsutils bind9-host bind9-libs dnsutils libedit2 libfstrm0 libicu67 liblmdb0 libmaxminddb0 libprotobuf-c1 libuv1 libxml2
```

eng3, 2021-12-17:
That seems to be the issue.  I got 0x50005.  When I tried reinstalling, it seemed to fix the issue. Sounds like I need to replace my power supply and maybe reinstall everything

closed 2021-12-17 by eng3

MichaIng, 2021-12-17:
If you have a different cable, try this first, often it is really those: There are extreme cases where the cable resistance drops voltage by 1V, which effectively breaks power supply despite the PSU is fine. But otherwise yes, the PSU should be one which is designed for powering devices, e.g. not a charger PSU: Charging does not require a strictly stable voltage, especially a lower voltage for short times does no harm, but when powering a device the needs are different and the voltage needs to be sufficiently stable. Hence not all mobile phone chargers work well.

eng3, 2021-12-18:
Yes that seems to be the issue.  I tried several cables and it didnt help, finally I just tried an extremely short cable and it worked.  I still get other errors sometimes (frequency lock) but I think that is just thermal issues when under heavy load of install/updates.  I'm not seeing anymore corruption.

MichaIng, 2021-12-18:
Okay, great that the original issue is solved now. A short able of course is always a good idea to minimise resistance, a good cable however shouldn't imply relevant resistance even with 2 meters length or so.