MichaIng/DietPi #5092: [Bug] AdGuard Home is not updating on Dietpi

Goldenfreddy0703, 2021-12-17:
#### Details:
- Date           | Fri Dec 17 15:56:42 EST 2021
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `cp -a AdGuardHome/. /mnt/dietpi_userdata/adguardhome/`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Go to dietpi-software and install AdGuard Home if you have not already have it. 
2. Follow the instructions on the DietPi Docs on how to update AdGuard Home: https://dietpi.com/docs/software/dns_servers/#adguard-home
[image]
3. Notice the error that says text file is busy.
#### Expected behaviour:

- AdGuardHome getting reinstalled and updated. 
#### Actual behaviour:

- Im getting an error when reinstalling and updating AdGuardHome when following the DietPi Docs.
#### Extra details:

- I do not really have any extra details but i think this is a software issue and im not sure what im doing wrong or how to update AdGuardHome on dietpi properly
#### Additional logs: Don't have any except a screenshot. 
```
![image](https://user-images.githubusercontent.com/62456796/146608371-726081fe-20e3-434d-803d-456e6b8f08ee.png)

cp: cannot create regular file '/mnt/dietpi_userdata/adguardhome/./AdGuardHome': Text file busy
```

Joulinar, 2021-12-17:
Hi, 

There should be a button inside AGH as well if a new version is available. Did you tried to use the internal update function?

MichaIng, 2021-12-17:
But many thanks for reporting. Looks like we need to stop the service before overwriting the files, and quickly restart it afterwards (in case AGH is used for the hosts own DNS resolution.

Or, if the internal updater works fine, skip updating it with a message that it should be done via internal updater.

Goldenfreddy0703, 2021-12-17:
> Hi,
> 
> There should be a button inside AGH as well if a new version is available. Did you tried to use the internal update function?

Hey there, i think i did find that button but when trying it, i think it had a problem with the version or something, its been a long time sense i tried that button.

Joulinar, 2021-12-17:
yup just tested, the service needs to be stopped on reinstall. Something we need to implement as users could run `dietpi-software reinstall 126` anyway

Joulinar, 2021-12-17:
> Hey there, i think i did find that button but when trying it, i think it had a problem with the version or something, its been a long time sense i tried that button.

if possible could you try the internal update function? that would be helpful.

Goldenfreddy0703, 2021-12-17:
Oh i actually just tried it and somehow it's updated to the latest version. I thought i had an old version.
The version im using is Version: v0.106.3.

Joulinar, 2021-12-17:
yes actual version is AdGuard Home v0.106.3 from mid of May 2021

Goldenfreddy0703, 2021-12-17:
oh ok, thank you very much. 
I remember last time when i tried the internal update function, it could not find the right version for some odd reason. Glad to be using the latest version and i hope this is a bug that could get fixed in the future.

MichaIng, 2021-12-17:
I'll also run a test to see whether there is any unintended result when running the internal updater on our setup, but actually when there is one available, this is always the preferred update method, to assure that possible migration steps are done. In theory if the  AGH startup itself does not do these migrations (when required) but only the updater, a simple binary replacement could cause issues.

Goldenfreddy0703, 2021-12-17:
ahhh ok i see, Thank you

Joulinar, 2021-12-17:
ok I guess I found why the internal update is not working. Because we create all fiels with `root` user permission. But we start the service als user `adguardhome`.

```
root@DietPi4:/mnt/dietpi_userdata/adguardhome# ls -la
total 21468
drwxr-xr-x  3 root        root            4096 Dec 17 22:36 .
drwxrwxr-x 11 dietpi      dietpi          4096 Dec 17 22:15 ..
-rwxr-xr-x  1 root        root        21889024 Mar 10  2021 AdGuardHome
-rw-r--r--  1 root        root             331 Mar 10  2021 AdGuardHome.sig
-rw-r--r--  1 root        root            2588 Dec 17 22:35 AdGuardHome.yaml.back
-rw-r--r--  1 root        root           10232 Mar 10  2021 CHANGELOG.md
-rw-r--r--  1 root        root           35149 Mar 10  2021 LICENSE.txt
-rw-r--r--  1 root        root           22442 Mar 10  2021 README.md
drwxr-xr-x  3 adguardhome adguardhome     4096 Dec 17 22:15 data
```

Means the update function is not able to overwrite files

```
Dec 17 22:43:08 DietPi4 AdGuardHome[17306]: 2021/12/17 22:43:08 [info] Updating from v0.105.2 to v0.106.3.  URL:https://static.adguard.com/adguardhome/release/AdGuardHome_linux_arm64.tar.gz
Dec 17 22:43:13 DietPi4 AdGuardHome[17306]: 2021/12/17 22:43:13 [info] copySupportingFiles(/mnt/dietpi_userdata/adguardhome/agh-update-v0.106.3, /mnt/dietpi_userdata/adguardhome) failed: open /mnt/dietpi_userdata/adguardhome/LICENSE.txt: permission denied
```

Changing the entire directory to user `adguardhome` will allow the update to work.

```
Dec 17 22:45:19 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:19 [info] Updating from v0.105.2 to v0.106.3.  URL:https://static.adguard.com/adguardhome/release/AdGuardHome_linux_arm64.tar.gz
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopping all tasks
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopping AdGuard Home
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopping HTTP server...
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopped HTTP server
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopping the DNS proxy server
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] udpListen.ReadFrom() returned because we're reading from a closed connection, exiting loop
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopped the DNS proxy server
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Stopped
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22 [info] Restarting: [/mnt/dietpi_userdata/adguardhome/AdGuardHome]
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.427232 [info] AdGuard Home, version v0.106.3
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.428790 [info] Upgrade yaml: 7 to 8
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.428825 [info] Upgrade yaml: 8 to 9
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.428839 [info] Upgrade yaml: 9 to 10
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.474340 [info] Initializing auth module: /mnt/dietpi_userdata/adguardhome/data/sessions.db
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.474766 [info] auth: initialized.  users:1  sessions:1
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.474800 [info] Initialize web module
Dec 17 22:45:22 DietPi4 AdGuardHome[17349]: 2021/12/17 22:45:22.522180 [info] AdGuard Home is available on the following addresses:
```

MichaIng, 2021-12-17:
Makes sense. We can apply correct owner with next update as regular patch, in the same turn skip file replacement if `/opt/adguardhome` exists already and refer to the internal updater. Docs then need to be updated as well, of course.

MichaIng, 2021-12-17:
The permissions are correct already, were set with initial merge of AdGuard Home: https://github.com/MichaIng/DietPi/commit/5bb8d60
@Joulinar 
Probably your install is from a in-dev stage of the related PR?

Joulinar, 2021-12-17:
nope was fresh install today for this issue. Will retest.

MichaIng, 2021-12-17:
Okay great, so we can update our docs directly 🙂. An upgrade is now skipped on reinstall as long as the binary/executable is present already: https://github.com/MichaIng/DietPi/commit/8d4f67e
I chose to use the binary here since removing the whole dir would remove data/configs and instructions are complicated when all other files need to be removed. Should be handy: If anything is not working, removing the binary, then doing a reinstall as repair step.

Joulinar, 2021-12-17:
should be fine this way.

MichaIng, 2021-12-18:
PR is up to fix the docs: https://github.com/MichaIng/DietPi-Docs/pull/644

closed 2021-12-18 by MichaIng