MichaIng/DietPi #5093: Dietpi-Software File browser won't start

jaguar489, 2021-12-18:
Hi,

I am running a freshly installed 64-bit dietpi OS on a raspberry Pi 4 4GB using a micro SD card. I am having the attached error when trying to access the file browser software. I have already attempted reinstalling the software and restarting the service but it continuosly says failed. 
Thanks 
[image]

Joulinar, 2021-12-18:
Hmm I did a test installation on RPi4 64bit without issue. Can you share following

```bash
journalctl -u filebrowser.service
```

jaguar489, 2021-12-18:
Here is the result 

root@DietPi:~# journalctl -u filebrowser.service
-- Journal begins at Fri 2021-12-17 19:23:01 EST, ends at Sat 2021-12-18 04:35:39 EST. --
Dec 17 19:23:13 DietPi systemd[1]: Started File Browser (DietPi).
Dec 17 19:23:15 DietPi filebrowser[841]: 2021/12/17 19:23:15 No config file used
Dec 17 19:23:15 DietPi filebrowser[841]: 2021/12/17 19:23:15 listen tcp 0.0.0.0:8084: bind: address already in use
Dec 17 19:23:15 DietPi systemd[1]: filebrowser.service: Main process exited, code=exited, status=1/FAILURE
Dec 17 19:23:15 DietPi systemd[1]: filebrowser.service: Failed with result 'exit-code'.

I'm not seeing anything that would conflict with the port. 
I also have the dietpi dashboard installed which also seems to have file browser.

Joulinar, 2021-12-18:
Can you share following

```bash
ss -tulpn | grep LISTEN
```

jaguar489, 2021-12-18:
root@DietPi:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      80                                          127.0.0.1:3306       0.0.0.0:*    users:(("mariadbd",pid=725,fd=21))  
tcp   LISTEN 0      511                                         127.0.0.1:6379       0.0.0.0:*    users:(("redis-server",pid=663,fd=7))
tcp   LISTEN 0      5                                             0.0.0.0:8080       0.0.0.0:*    users:(("python3",pid=828,fd=3))    
tcp   LISTEN 0      1024                                          0.0.0.0:80         0.0.0.0:*    users:(("lighttpd",pid=815,fd=4))   
tcp   LISTEN 0      500                                           0.0.0.0:8084       0.0.0.0:*    users:(("mono",pid=416,fd=5))       
tcp   LISTEN 0      10                                            0.0.0.0:8085       0.0.0.0:*    users:(("python3",pid=844,fd=5))    
tcp   LISTEN 0      1000                                          0.0.0.0:22         0.0.0.0:*    users:(("dropbear",pid=439,fd=4))   
tcp   LISTEN 0      128                                           0.0.0.0:8088       0.0.0.0:*    users:(("dietpi-dashboar",pid=41453,fd=6))
tcp   LISTEN 0      500                                           0.0.0.0:8989       0.0.0.0:*    users:(("mono",pid=831,fd=8))       
tcp   LISTEN 0      512                                           0.0.0.0:8096       0.0.0.0:*    users:(("jellyfin",pid=823,fd=288)) 
tcp   LISTEN 0      30                                 192.168.0.26%wlan0:6881       0.0.0.0:*    users:(("qbittorrent-nox",pid=820,fd=18))
tcp   LISTEN 0      30                                       127.0.0.1%lo:6881       0.0.0.0:*    users:(("qbittorrent-nox",pid=820,fd=16))
tcp   LISTEN 0      512                                                 *:7878             *:*    users:(("Radarr",pid=834,fd=146))   
tcp   LISTEN 0      511                                             [::1]:6379          [::]:*    users:(("redis-server",pid=663,fd=8))
tcp   LISTEN 0      1024                                             [::]:80            [::]:*    users:(("lighttpd",pid=815,fd=5))   
tcp   LISTEN 0      4096                                                *:22000            *:*    users:(("syncthing",pid=897,fd=17)) 
tcp   LISTEN 0      128                                                 *:21               *:*    users:(("proftpd",pid=660,fd=0))    
tcp   LISTEN 0      1000                                             [::]:22            [::]:*    users:(("dropbear",pid=439,fd=5))   
tcp   LISTEN 0      50                                                  *:1340             *:*    users:(("qbittorrent-nox",pid=820,fd=14))
tcp   LISTEN 0      4096                                                *:8384             *:*    users:(("syncthing",pid=897,fd=18)) 
tcp   LISTEN 0      30                   [fe80::e65f:1ff:fe5b:88ec]%wlan0:6881          [::]:*    users:(("qbittorrent-nox",pid=820,fd=26))
tcp   LISTEN 0      30       [2607:fea8:9600:f0:e65f:1ff:fe5b:88ec]%wlan0:6881          [::]:*    users:(("qbittorrent-nox",pid=820,fd=24))
tcp   LISTEN 0      30     [fd00:1cab:c00f:cfd2:e65f:1ff:fe5b:88ec]%wlan0:6881          [::]:*    users:(("qbittorrent-nox",pid=820,fd=22))
tcp   LISTEN 0      30                                           [::1]%lo:6881          [::]:*    users:(("qbittorrent-nox",pid=820,fd=20))

Joulinar, 2021-12-18:
There is a `mono` instance running. Do you use something like `sonarr`?

jaguar489, 2021-12-18:
Yes both sonarr and radarr

Joulinar, 2021-12-18:
ok we produced a port conflict between `mono` and `file browser`. As workaround you could do following

```
cd /mnt/dietpi_userdata/filebrowser
/opt/filebrowser/filebrowser config set -p 8091
systemctl restart filebrowser.service
```

`file browser` should be available on port 8091 now.

MichaIng, 2021-12-18:
Looks like. I'm wondering whether this is a Sonarr API on port 8084 or whether Mono itself always listens on this port for some reaons/runtime API. Can be tested with Radarr/Lidarr/Jacket on an ARMv6 RPi (1 and Zero), since all other architectures use the .NET core based versions without Mono.

MichaIng, 2021-12-19:
We solved it differently now: #5099
Previously we had `mono-complete` installed, which pulls in an additional webserver services running on this port `8084` while Sonarr and the Mono-builds of Radarr, Lidarr and Jackett do not use this port and do not require anything provided by `mono-complete`. So it is not installed anymore now in favour of a smaller subset of Mono runtime and development packages, which does not include this unnecessary Mono XSP4 webserver service. You can manually apply this change:
```sh
apt-mark manual mono-devel
apt --autoremove purge mono-complete
```

closed 2022-01-02 by MichaIng