MichaIng/DietPi #5094: mount the nextcloud raid failed

Jernel-dotcom, 2021-12-18:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='applied'
echo $G_DISTRO_NAME $G_RASPBIAN
Linux DietPi 5.10.63-v8
echo $G_HW_MODEL_NAME
- Power supply used 5V 1A RAVpower
- SDcard used SanDisk ultra

#### Additional Information (if applicable)
- Software title  Nextcloud
- Was the software title installed freshly or updated/migrated? Fresh installation
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce
1. Install nextcloud 
2. Use it for a few weeks
3. Get this issue from nowhere 

#### Expected behaviour
- This behavior shouldn't happen 

#### Actual behaviour
- Nextcloud didn't work 

#### Extra details
`Dec 18 16:10:19 DietPi systemd[1]: dev-disk-by\x2duuid-9fefd2a8\x2d39ae\x2d4aef\x2d8ad4\x2dbadb34d264ab.device: Job dev-disk-by\x2duuid-9fefd2a8\x2d39ae\x2d4aef\x2d8ad4\x2dbadb34d264ab.device/start timed out.
Dec 18 16:10:19 DietPi systemd[1]: Timed out waiting for device /dev/disk/by-uuid/9fefd2a8-39ae-4aef-8ad4-badb34d264ab.
Dec 18 16:10:19 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 16:10:19 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 16:10:19 DietPi systemd[1]: dev-disk-by\x2duuid-9fefd2a8\x2d39ae\x2d4aef\x2d8ad4\x2dbadb34d264ab.device: Job dev-disk-by\x2duuid-9fefd2a8\x2d39ae\x2d4aef\x2d8ad4\x2dbadb34d264ab.device/start failed with result 'timeout'.
`

Joulinar, 2021-12-18:
can you give some more details on this disk/raid setup. Is that a software raid? Or some external hardware? At the moment it seems our disks are not available/usable.

Jernel-dotcom, 2021-12-18:
The raid setup with mdadm.
4 hdds 2.5 seagate 
raid 5 
quad nas kit -
https://shop.allnetchina.cn/products/quad-sata-hat-case-for-raspberry-pi-4
```
ls: cannot access '/mnt/dietpi_userdata/nextcloud_data': No such device
total 28
drwxrwxr-x 8 dietpi dietpi 4096 Feb 10  2021 .
drwxr-xr-x 7 root   root   4096 Oct 19 12:06 ..
drwxrwxr-x 2 dietpi dietpi 4096 Feb 10  2021 Music
drwxrwxr-x 2 dietpi dietpi 4096 Feb 10  2021 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Feb 10  2021 Video
drwxrwxr-x 2 dietpi dietpi 4096 Feb 10  2021 downloads
drwxr-xr-x 5 mysql  mysql  4096 Dec 18 16:08 mysql
d????????? ? ?      ?         ?            ? nextcloud_data
```

Joulinar, 2021-12-18:
Did you tried to reboot? The software raid did not seems to be healthy. Can you check the status `mdadm`?

Jernel-dotcom, 2021-12-18:
Tried to reboot multiple times 
```
root@DietPi:~# mdadm -D /dev/md0 
/dev/md0:
           Version : 1.2
        Raid Level : raid0
     Total Devices : 4
       Persistence : Superblock is persistent

             State : inactive

    Number   Major   Minor   RaidDevice
```

Joulinar, 2021-12-18:
I'm not an expert for these kind of software raids but status on your device is `inactive`. You would need to find out why your software raid is not working.

MichaIng, 2021-12-18:
If it is an `fstab` entry, the following command may provide additional details about the failed mount:
```sh
journalctl -u mnt-dietpi_userdata-nextcloud_data.mount
```

Jernel-dotcom, 2021-12-18:
```
#journalctl -u mnt-dietpi_userdata-nextcloud_data.mount

root@DietPi:~# journalctl -u mnt-dietpi_userdata-nextcloud_data.mount
Journal file /var/log/journal/5bc8c3e1da734f11af934ab2c197a34e/system.journal is truncated, ignoring file.
-- Journal begins at Sat 2021-12-18 22:23:46 +02, ends at Sat 2021-12-18 22:33:11 +02. --
Dec 18 22:25:40 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:25:40 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 22:27:10 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:27:10 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 22:28:40 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:28:40 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 22:30:11 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:30:11 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 22:31:41 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:31:41 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.
Dec 18 22:33:11 DietPi systemd[1]: Dependency failed for /mnt/dietpi_userdata/nextcloud_data.
Dec 18 22:33:11 DietPi systemd[1]: mnt-dietpi_userdata-nextcloud_data.mount: Job mnt-dietpi_userdata-nextcloud_data.mount/start failed with result 'dependency'.

# dmesg -l emerg,alert,crit,err
[    0.564736] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.

```

Jernel-dotcom, 2021-12-18:
```
root@DietPi:~# cat /proc/mdstat 
Personalities : [raid6] [raid5] [raid4] 
md0 : active (auto-read-only) raid5 sdd[4] sdc[2] sdb[1] sda[0]
      2929890816 blocks super 1.2 level 5, 512k chunk, algorithm 2 [4/4] [UUUU]
      bitmap: 0/8 pages [0KB], 65536KB chunk

```

Joulinar, 2021-12-18:
some dependency is missing. Are you sure all drives are available?

`lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid`

MichaIng, 2022-01-04:
Does the issue persist or can this be closed?

MichaIng, 2022-02-06:
Marking as closed due to outstanding reply. Feel free to reopen when required.

closed 2022-02-06 by MichaIng