MichaIng/DietPi #5096: Unable To Connect To Deluge Web UI

AdamF100001, 2021-12-18:
Hi

I am unable to connect to Deluge Web UI in Deluge 1.3.15 Windows Client.

But I can connect to the Deluge Web UI on Andriod through an app called Deluge off the F-Driod app store.

Thanks    

Adam

MichaIng, 2021-12-18:
This means it is a client side issue. Can you connect via browser from Windows?

AdamF100001, 2021-12-18:
Yes I can connect via browser from Windows

Joulinar, 2021-12-18:
Maybe an issue between the client 1.3.15 and deluge 2. Maybe they are not fitting to each other anymore. If I'm not mistaken the client is from 2017 https://ftp.osuosl.org/pub/deluge/windows/?C=M;O=D

AdamF100001, 2021-12-18:
yeah client 2.0 is supported on everything apart from windows, but I have it working on Andriod so not all bad.

Joulinar, 2021-12-18:
ok I marked the issue as outside DietPi as it is most probably the client

MichaIng, 2021-12-18:
Ah right, I remember there was already an issue about that. At least one can always connect via browser 🙂.

AdamF100001, 2021-12-18:
Also true. thanks for the help.

Joulinar, 2022-01-02:
Going to close this as it is out of our scope. Feel free to reopen if needed.

closed 2022-01-02 by Joulinar