MichaIng/DietPi #5097: Update failed

erwin-frohsinn, 2021-12-19:
#### Details:
```
- Date           | Sun Dec 19 08:44:09 CET 2021
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux NodeRedMosq 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `mount -v -o rw,remount /`
- Exit code      | 32
- Software title | DietPi-Drive_Manager
```
#### Steps to reproduce:

1. run dietpi-update
2. ...
#### Expected behaviour:

execute update and return to normal operation
#### Actual behaviour:

Many messages of this type: 
```
dpkg: error processing archive /tmp/apt-dpkg-install-QVY6eQ/7-rpi-eeprom_13.3-1~buster_armhf.deb (--unpack):                                     
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system     
```
Finally terminating with this message:
```                                                                                                                                              
   E: Sub-process /usr/bin/dpkg returned an error code (2)                                                                                          
---------------------------------------------------------------------                                                                            
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
```

I tried to fix this as recommended with dietpi-drive_manager, but no success:
```
   sudo dietpi-drive_manager
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
df: /proc/sys/fs/binfmt_misc: No such device
[ INFO ] DietPi-Drive_Manager |  - Detected mounted physical drive: /dev/mmcblk0p2 > /
stty: 'standard input': Inappropriate ioctl for device
/boot/dietpi/func/dietpi-globals: line 474: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/boot/dietpi/func/dietpi-globals: line 480: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")
[FAILED] DietPi-Drive_Manager | mount -v -o rw,remount /
stty: 'standard input': Inappropriate ioctl for device
/boot/dietpi/func/dietpi-globals: line 474: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/boot/dietpi/func/dietpi-globals: line 480: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")
[FAILED] DietPi-Drive_Manager | mount -v -o rw,remount /
stty: 'standard input': Inappropriate ioctl for device
/boot/dietpi/func/dietpi-globals: line 474: ((: <= 120 : syntax error: operand expected (error token is "<= 120 ")
/boot/dietpi/func/dietpi-globals: line 480: ((: >= 7 : syntax error: operand expected (error token is ">= 7 ")

```                                                   
#### Extra details:

- ...
#### Additional logs:
```
mount: /: cannot remount /dev/mmcblk0p2 read-write, is write-protected.
```

Joulinar, 2021-12-19:
There seems to be some issues with your SD card. Could you check for kernel error messages 

`dmesg -l err,crit,alert,emerg`

erwin-frohsinn, 2021-12-19:
Am Sonntag, 19. Dezember 2021, 09:37:37 CET schrieb 
Joulinar:
> dmesg -l err,crit,alert,emerg

[    0.544465] bcm2708_fb soc:fb: Unable to determine 
number of FBs. Disabling driver.

Kind regards!

Joulinar, 2021-12-19:
do you have any physical write protection on your SD card?

erwin-frohsinn, 2021-12-19:
Am Sonntag, 19. Dezember 2021, 13:52:26 CET schrieb Joulinar:
> do you have any physical write protection on your SD card?
No. And after pysically switching the device off and on again, at 
least my application (mosquitto / nodered) are operational 
again. I will, maybe tomorrow, clone the SD card and try again 
the update.

quyentruong, 2021-12-20:
I had the same issue before. Micro SD card is basically damaged. Just swap a new one and everything is ok.

MichaIng, 2022-01-04:
Yes that is possible. @erwin-frohsinn were you able to solve it?

erwin-frohsinn, 2022-01-05:
Yes, thanks. I duplicated the SD card and the clone worked again.

closed 2022-01-05 by erwin-frohsinn