MichaIng/DietPi #5103: Ampache [Software ID 40]: Add Lighttpd routing configuration to enable Subsonic API out of the box

MochaMage, 2021-12-20:
Ampache's current install is not currently enabling Subsonic API routing out of the box. Adding the following configuration to a file within the `/etc/lighttpd/conf-enabled` directory will allow the Subsonic API to be accessible. This comes directly from the [Ampache API Documentation](https://github.com/ampache/ampache/wiki/API#for-others) but has been modified to included the `ampache` URI that is used by DietPi's current install.

`url.rewrite-if-not-file += (
  "^/ampache/rest/(([^\?]+)\.view)(\?(.*))?" => "/ampache/rest/index.php?ssaction=$2&$4",
  "^/ampache/play/ssid/([^/]+)/type/([^/]+)/oid/([^/]+)/uid/([^/]+)/client/([^/]+)/noscrobble/([^/]+)/bitrate/([^/]+)/player/([^/]+)/name/([^/]+)(/.*)?$" => "/ampache/play/index.php?ssid=$1&type=$2&oid=$3&uid=$4&client=$5&noscrobble=$6&bitrate=$7&player=$8&name=$9"
)`

MichaIng, 2021-12-20:
Many thanks for your request.

I remember we thought about this once already. This means that Subsonic clients can be used, right? We'd need to add those rewrites for all three webservers, Lighttpd, Nginx and Apache.

MochaMage, 2022-10-07:
@MichaIng Correct, though after testing this, it varies per client as well. This configuration works for Subsonic but not others like Substreamer. Substreamer's requests seem follow a different format that this regex doesn't cover

MochaMage, 2023-10-17:
Related to this enhancement request, I've moved on to a new installation and used Apache2 this time around. I've made the following RewriteRules that, as of test time, work with both DSub and Substreamer applications with the new upgraded regex. I'm not super familiar with Apache so these are currently in my apache.conf file but having these be part of Dietpi Ampache's out-of-the-box installation would be immensely helpful. 

`        RewriteRule "rest/scrobble.view?u=(.*)&p=(.*)&v=(.*)&c=(.*)&id=30+(.*)&submission=(.*)$" "rest/scrobble.view?u=$1&p=$2&v=$3&c=$4&id=$5&submission=$6" [PT]
        RewriteRule "rest/(([^\?]+)\.view)" "rest/index.php?ssaction=$2" [QSA]
        RewriteRule "play/ssid/([^/]+)/type/([^/]+)/oid/([^/]+)/uid/([^/]+)/client/([^/]+)/bitrate/([^/]+)(/cache/([^/]+))?(/transcode_to/([^/]+))?(/action/([^/]+))?/player/([^/]+)/name/([^/]+)(/.*)?$" "play?ssid=$1&type=$2&oid=$3&uid=$4&client=$5&bitrate=$6&cache=$8&transcode_to=mp3&action=$12&player=$13&name=$14"
`