MichaIng/DietPi #5104: DietPi doesn't boot changing Rpi4 board with the same SD

giusanny, 2021-12-21:
#### Required Information
- DietPi version | 7.9.3
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used  5V 3A 
- SDcard used SanDisk ultra 32GB

Having 2 identical Rpi4 with 2 idendical Sunfounder touch screen 10.1'' without any other USB devices plugged into, on the first Rpi4 everything boots and works fine, then if i put the same SD in the second Rpi4 it does not boot: after first boot screen with Raspberry logo the preboot script will not execute (i put some logs with echo into), the screen is disconnected and goes in the sleep mode also the green light stays always on forever.

If i put back again the SD in the first Rpi4 works perfectly again.

All the two Rpi4 and touch work perfectly with another SD card (Samsung 32GB) and Raspbian so there are no hadrware problems

How and where can i read some logs before the preboot script execution to get some other information?

P.S. DietPi already setup to not wait network and NTP to boot

#### Expected behaviour
The SD that work on an Rpi4 has to work also in another Rpi4 with same hardware config 

#### Actual behaviour
DietPi boots only on one Rpi4

MichaIng, 2021-12-21:
Many thanks for your report.

Obviously there is some different between any of the hardware components between the two Pi's 😉. Try to remove the `quite` flag from `cmdline.txt` to see all boot/kernel logs on the screen.

Please re-enable at least network time sync at boot, it is imply strictly required for any successful network connection, it does not delay boot and runs on a much later boot stage anyway.

giusanny, 2021-12-21:
First of all thanks to you and your reply!

I don't expect so many differences but i have your same doubt on the hardware, but i made a new boot as you asked (without quiet) and the monitor stops and all hangs at the point "File System Check"...so it doesn't really hangs but is making a file system check and now explained the green light always ON...

[image]

How can i disable File System Check? I really don't know why it is requested only on the second Rpi4

giusanny, 2021-12-21:
seems like removing `fsck.repair=yes` from the `cmdline.txt` solved my issue this should be useful for others with the same problem

Still don't know why the second Rpi needs a file system check 

thanks a lot @MichaIng for your support

MichaIng, 2021-12-24:
I highly recommend to leave `fsck.repair=yes` in place. There is only two reasons for a forced fsck on boot:
1. An ext4 filesystem every specified amount of mounts, 60 by default. Does not apply here since it is the boot FAT partition which is checked.
2. The filesystem has a dirty bit checked, i.e. the kernel detected a filesystem corruption before the reboot. In this case it is highly recommended to have it repaired ASAP before it causes for issues, hence the auto-repair.

Since it's the FAT partition, you can however do the fsck interactively while its mounted:
```sh
fsck /dev/mmcblk0p1
```

giusanny, 2021-12-26:
I agree with you but I still don't get why the same SD (same kernel) forces fsck only on the first Rpi4

Joulinar, 2021-12-26:
Probably system detects a new hardware and force a check

MichaIng, 2021-12-27:
AFAIK there is no mechanism for forcing an `fsck` based on hardware changes. I guess it was by chance that the `fsck` was done on one Pi, invalidated the full `fsck` flag for the next boot (on the other system), but the unclean shutdown and in case related filesystem error detected on the other system then re-set the flag for next boot. Important to leave it running through to break the loop and avoid the risk for permanent data loss.

closed 2022-02-10 by giusanny