MichaIng/DietPi #5105: wifi problem

geol, 2021-12-21:
when the router's wifi set 802.11w to "Optional" or "Enable", then my rpi4 can not connect to it.

Joulinar, 2021-12-21:
I guess this is an issue of RPi kernel/firmware not supporting ieee80211w/MFP https://github.com/raspberrypi/linux/issues/3619

geol, 2021-12-22:
> I guess this is an issue of RPi kernel/firmware not supporting ieee80211w/MFP [raspberrypi/linux#3619](https://github.com/raspberrypi/linux/issues/3619)

thank u for reply. so, there is no any solutions at the moment?

Joulinar, 2021-12-22:
Not sure if we are able to do anything if it is indeed a kernel/firmware topic. Because we don't do kernel development. You could ask on the GitHub post I linked above if there is a solution to geht ieee80211w/MFP running.

geol, 2021-12-23:
> Not sure if we are able to do anything if it is indeed a kernel/firmware topic. Because we don't do kernel development. You could ask on the GitHub post I linked above if there is a solution to geht ieee80211w/MFP running.

okay, thank you.

closed 2021-12-23 by geol

MichaIng, 2021-12-24:
And there was even a firmware and kernel patch available already, but it hasn't been supplied via any release until today 😞. @geol You may contribute to the issue, to hopefully get it more attention. I'm not sure how frequently `ieee80211w` is used, but especially WPA3 becomes common so plenty reasons to get it supported natively.