MichaIng/DietPi #5106: Dietpi-Dashboard fails to install due to curl cmd failing

dma4620, 2021-12-22:
login as: root
root@192.168.1.3's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v7.9.3 : 16:29 - Wed 12/22/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 38 °C / 100 °F : Cool runnings
 - LAN IP : 192.168.1.3 (eth0)
 - MOTD : WARNING: Learn whether your system is affected by the
          Java Log4j vulnerability CVE-2021-44228 and how to mitigate:
          https://dietpi.com/blog/?p=1172
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: from scratch)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# systemctl status dietpi-dashboard
Unit dietpi-dashboard.service could not be found.
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : php7.4-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:4 https://deb.debian.org/debian bullseye-backports InRelease [43.7 kB]
Get:5 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Get:6 https://deb.debian.org/debian-security bullseye-security/main armhf Packages [99.0 kB]
Get:7 https://deb.debian.org/debian bullseye-backports/main armhf Packages.diff/Index [63.3 kB]
Get:8 https://deb.debian.org/debian bullseye-backports/main armhf Packages T-2021-12-22-1401.51-F-2021-12-22-0201.33.pdiff [980 B]
Get:8 https://deb.debian.org/debian bullseye-backports/main armhf Packages T-2021-12-22-1401.51-F-2021-12-22-0201.33.pdiff [980 B]
Get:9 https://archive.raspberrypi.org/debian bullseye/main armhf Packages [242 kB]
[... 167 lines cut ...]
* stopped the pause stream!
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
* Connection #1 to host objects.githubusercontent.com left intact
curl: (28) Operation timed out after 5187 milliseconds with 0 bytes received
HTTP/2 302
server: GitHub.com
date: Wed, 22 Dec 2021 16:33:30 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/396090981/f59f9599-7f02-4c8e-89db-b931783db968?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211222%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20211222T163330Z&X-Amz-Expires=300&X-Amz-Signature=00aed62e1dea2331bcabe8bc683341c1074a54a46952b27145bfb85fd4dfbf2e&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=396090981&response-content-disposition=attachment%3B%20filename%3Ddietpi-dashboard-armv7l&response-content-type=application%2Foctet-stream
cache-control: no-cache
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
content-length: 662
x-github-request-id: B76E:467F:119D25:32322D:61C3535A

```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2021-12-22:
Hi,

it seems your connection timed out

MichaIng, 2021-12-24:
Does it take that long as well when connecting from console?
```sh
time curl -IL 'https://objects.githubusercontent.com/'
```
As a workaround you can raise the timeout, e.g.:
```sh
G_CONFIG_INJECT 'CONFIG_G_CHECK_URL_TIMEOUT=' 'CONFIG_G_CHECK_URL_TIMEOUT=20' /boot/dietpi.txt
```

MichaIng, 2022-01-04:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2022-01-04 by MichaIng