MichaIng/DietPi #5107: dietpi-drivemanager forgets about an addition partition after power outage and can't mount it later manually

YNikiforov, 2021-12-22:
#### Details:
- Date           | Wed Dec 22 19:04:52 CET 2021
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux RPi 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `mount -o noatime,lazytime,rw /dev/mmcblk0p3 /mnt/data`
- Exit code      | 32
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:
1. SD Card in my RPi has additional third partition with ext4 for userdata (doc, audio, photo, etc)
2. This additional partition was mounted using dietpi-drivemanager to /mnt/data
3. After power outage this additional partition got corrupted (maybe others also, but RPi booted fine anyway)
#### Expected behaviour:
- I'd expect that all partitions mounted with dietpi-drivemanager are checked on start and mounted or a user is notified somehow like in the banner message during login or maybe a mail is sent
#### Actual behaviour:
- partition was not mounted on start after power failure
- subsequent restart via ssh login didn't help to recover from the failure automatically
- attempt to mount using dietpi-drivemanager failed with a message below and a prompt to submit a bug
- after checking the filesystem with `fsck.ext4 /dev/mmcblk0p3` dietpi-drivemanager no longer remembered where this partition was mounted to and proposed a new mount path `/mnt/disk_uuid`
#### Additional logs:
```
mount: /mnt/data: mount(2) system call failed: Structure needs cleaning.
```

ravenclaw900, 2021-12-22:
Hello,
```
Structure needs cleaning.
```
This means that the filesystem on the external drive is corrupted. I see that you already ran `fsck`, what did it say? You might need to run it multiple times to fix everything.

YNikiforov, 2021-12-22:
Hi,

here is what I did and what I saw

```
root@RPi:~# mount -o noatime,lazytime,rw /dev/mmcblk0p3 /mnt/data
mount: /mnt/data: mount(2) system call failed: Structure needs cleaning.
root@RPi:~# fsck.ext4 /dev/mmcblk0p3
e2fsck 1.46.2 (28-Feb-2021)
data: recovering journal
Journal transaction 726 was corrupt, replay was aborted.
data contains a file system with errors, check forced.
Pass 1: Checking inodes, blocks, and sizes
Inode 1048677 extent tree (at level 1) could be shorter.  Optimize<y>? yes
Inode 1048700 extent tree (at level 1) could be shorter.  Optimize<y>? yes
Inode 1048735 extent tree (at level 1) could be narrower.  Optimize<y>? yes
Pass 1E: Optimizing extent trees
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Block bitmap differences:  -(10899456--10911743) -(12615680--12627717) -(12627968--12629679) -(12630016--12636602) -(12638208--12658079) -(12658176--12662667) -(12662784--12677902) -(12679168--12684890) -(12685312--12693375) -(12693504--12712582) -(12713984--12715947) -(12716032--12719729) -(12720128--12744674) -(12744704--12750823) -(12750848--12767079) -(12767232--12803451) -(12804096--12832165) -(12832768--12857214) -(12857344--12867430) -(12867584--12889301) -(12890112--12912222) -(12912640--12940006) -(12941312--12943134) -(12943360--12944945) -(12945408--12969600) -(12969984--12984016) -(12984320--12984920) -(12986368--13018519) -(13019136--13032347) -(13033472--13052901) -(13053952--13085295) -(13086720--13107199) -(13139968--13143412) -(13144064--13144485) -(13146112--13164543)
Fix<y>? yes
Free blocks count wrong for group #385 (2191, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #386 (1477, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #387 (1950, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #388 (511, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #389 (176, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #390 (644, counted=32768).
Fix ('a' enables 'yes' to all) <y>? yes
Free blocks count wrong for group #391 (602, counted=32768).
Fix ('a' enables 'yes' to all) <y>? yes
Free blocks count wrong for group #392 (282, counted=32768).
Fix ('a' enables 'yes' to all) <y>? yes
Free blocks count wrong for group #393 (810, counted=32768).
Fix ('a' enables 'yes' to all) <y>? yes
Free blocks count wrong for group #394 (1947, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #395 (845, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #396 (1750, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #397 (1740, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #398 (1050, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #399 (1424, counted=32768).
Fix<y>? yes
Free blocks count wrong for group #401 (10469, counted=32768).
Fix<y>? yes
Free blocks count wrong (3310637, counted=2711142).
Fix<y>? yes
yInode bitmap differences:  -1097730
Fix<y>? yes
Free inodes count wrong (3424067, counted=3424065).
Fix<y>? yes
yyyyyyyy
data: ***** FILE SYSTEM WAS MODIFIED *****
data: 191/3424256 files (25.7% non-contiguous), 10959514/13670656 blocks
yroot@RPi:~#mount -o noatime,lazytime,rw /dev/mmcblk0p3 /mnt/data
root@RPi:~#
```

Everything worked fine. The files appeared on a network share.

I rebooted the RPi just to check what would happen next outage/reboot. After reboot the mount did not persist, that's kinda expected. I ran dietpi-drivemanager and now it was possible to mount the partition to a manually entered path. Also a line was added to /etc/fstab
`UUID=5f1ac1cd-0349-4203-8c24-f44515dd2907 /mnt/data ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount`
Next reboots work further fine, no issues so far.

Joulinar, 2021-12-22:
That's basically what drives manager does. Creating an entry inside `/etc/fstab`. It's basically just a frontend and not doing the actual mount during boot. This is done by the system/kernel itself based on `/etc/fstab`

YNikiforov, 2021-12-22:
Hi Joulinar, yes, the idea is clear. The question is whether anyone at dietpi team wants the drive manager to handle the case when a filesystem is corrupted and

- dietpi_userdata was there and is not found after restart
- attempt to mount this partition via drive manager results in a prompt to file a bug here

Joulinar, 2021-12-24:
Usually it's a problem to perform `fsck` on a mounted device like rootFS and/or bootFS. If I'm not mistaken, this require the partition to be unmounted. Which might be a challenge on a running system.

But maybe I'm wrong. Probably @MichaIng could give an answer is this is possible.

MichaIng, 2021-12-24:
`ext4` filesystems cannot be checked while mounted, FAT partitions (`/boot`) e.g. can, so it depends on the filesystem type. Since root cannot be unmounted while the system is running, it needs to be checked on reboot.

Many steps in our scripts are error handled, and the error handler dialogue prints the information required to file a bug report, so it is not specific to this particular task 😉. In this case it was however, a valuable step so that you did the important `fsck` now before potential data loss becomes worth.

MichaIng, 2022-02-20:
I'll mark this as closed. Feel free to reopen if required.

Generally adding some more specific menu entries to the error handler on specific commands makes sense, and our error handler has basic functionality which allows this. In this particular case it is however not so trivial, since file system corruption can happen everywhere resulting in various different kernel and command errors outputs or some which do not indicate in this direction at all. Adding a special handling only for the mount command seems hence not reasonable, but instead a generic approach would be better. If I remember right there is a way to get an error counter on ext4. This could be read at some point and an fsck recommended, within drive manager, banner or so. But I couldn't find a way to check this counter currently, `tune2fs` seems to not list it, `dmesg` should contain it if an error happens, need to check next time.

closed 2022-02-20 by MichaIng