MichaIng/DietPi #5108: Upgrade to 7.9.3 -> debian bullseye InRelease: Unknown error executing apt-key

maury77, 2021-12-24:
#### Details:
- Date           | Fri Dec 24 13:59:02 CET 2021
- Bug report     | bae23ae2-11b1-4cfd-b5f2-2cf739605d9c
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux DietPi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 255
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  Unknown error executing apt-key
Err:2 https://archive.raspberrypi.org/debian bullseye InRelease
  Unknown error executing apt-key
Fetched 38.5 kB in 2s (25.6 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://raspbian.raspberrypi.org/raspbian bullseye InRelease: Unknown error executing apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://archive.raspberrypi.org/debian bullseye InRelease: Unknown error executing apt-key
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Unknown error executing apt-key
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Unknown error executing apt-key
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Joulinar, 2021-12-24:
Hmm,

you could try following

```
apt upgrade
apt-get upgrade --fix-missing
dpkg --configure -a
```

reboot your system and check

`apt update`

maury77, 2021-12-24:
I resolve with reinstall of apt install --reinstall apt

closed 2021-12-24 by maury77