MichaIng/DietPi #5114: SSH Refused Your Key ( Solution )

billouetaudrey, 2021-12-25:
Hello,

If since update, you have SSH Error, the fix is to add 
`PubkeyAcceptedKeyTypes +ssh-rsa` 

to /etc/ssh/sshd_config

Best Regards

billouetaudrey, 2021-12-25:
Fixed

closed 2021-12-25 by billouetaudrey

MichaIng, 2021-12-27:
All Debian versions up to Sid and Experimental ship at max OpenSSH v8.7, so this does not apply to DietPi in any iteration, yet. But good to know for the future, as this may become relevant in two years or so.

billouetaudrey, 2021-12-27:
I had this issue, dunno why ... Thought was latest Dietpi update ;)

MichaIng, 2021-12-28:
Stange, probably `PubkeyAcceptedKeyTypes` was set elsewhere, but it is not defined in any default config, hence defaults to accepting RSA, ECDSA and in case Ed25519 keys. One explanation could be that you key used a very old/insecure hash algorithm, too low key length or similar, which is as well defined with the default setting, e.g. on Bullseye:
```
ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,
ssh-ed25519
rsa-sha2-512,rsa-sha2-256,ssh-rsa
```
However, better than enabling legacy key formats, if all hosts support newer ones, is to generate a modern Ed25519 or ECDSA key pair.

billouetaudrey, 2021-12-29:
I don't know what I do, suddenly root log ( dietpi work ) stop working. I try to renew the key, nothing help except edit sshd.conf

MichaIng, 2021-12-29:
However, good that you found a solution and we have a reminder that RSA keys are about to be deprecated. Luckily `libssh2` (used by `curl`) supports ECDSA and even Ed25519 it seems, with latest versions: https://libssh2.org/changes.html

billouetaudrey, 2021-12-29:
Exactly :)

But I think if I close 22 port, don't need extra keys :)