MichaIng/DietPi #5115: Error installing myMPD

supersexy, 2021-12-26:
The install process fails because missing PCRE libraries. 

Also the installation process does not seem to be isolated in a transaction, so it leaves the system in a dirty state, users have to do some apt gymnastics to clean up.

Additionaly the "copy-only-blue-text" feature did not produce any meaningful content inside the blue text area, so I am posting everything.

```

DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing myMPD: fork of ympd with improved features

[  OK  ] DietPi-Software | Checking URL: https://github.com/jcorporation/myMPD/archive/master.tar.gz
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://github.com/jcorporation/myMPD/archive/master.tar.gz -o master.tar.gz
[ INFO ] DietPi-Software | APT install for: make gcc cmake perl libc6-dev libssl-dev libid3tag0-dev libflac-dev liblua5.3-dev pkg-config libpcre3-dev, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.32.
(Reading database ... 44898 files and directories currently installed.)
Preparing to unpack .../00-perl-modules-5.32_5.32.1-4+deb11u2_all.deb ...
Unpacking perl-modules-5.32 (5.32.1-4+deb11u2) ...
Selecting previously unselected package libgdbm6:arm64.
Preparing to unpack .../01-libgdbm6_1.19-2_arm64.deb ...
Unpacking libgdbm6:arm64 (1.19-2) ...
Selecting previously unselected package libgdbm-compat4:arm64.
Preparing to unpack .../02-libgdbm-compat4_1.19-2_arm64.deb ...
Unpacking libgdbm-compat4:arm64 (1.19-2) ...
Selecting previously unselected package libperl5.32:arm64.
Preparing to unpack .../03-libperl5.32_5.32.1-4+deb11u2_arm64.deb ...
Unpacking libperl5.32:arm64 (5.32.1-4+deb11u2) ...
Selecting previously unselected package perl.
Preparing to unpack .../04-perl_5.32.1-4+deb11u2_arm64.deb ...
Unpacking perl (5.32.1-4+deb11u2) ...
Selecting previously unselected package binutils-common:arm64.
Preparing to unpack .../05-binutils-common_2.35.2-2_arm64.deb ...
Unpacking binutils-common:arm64 (2.35.2-2) ...
Selecting previously unselected package libbinutils:arm64.
Preparing to unpack .../06-libbinutils_2.35.2-2_arm64.deb ...
Unpacking libbinutils:arm64 (2.35.2-2) ...
Selecting previously unselected package libctf-nobfd0:arm64.
Preparing to unpack .../07-libctf-nobfd0_2.35.2-2_arm64.deb ...
Unpacking libctf-nobfd0:arm64 (2.35.2-2) ...
Selecting previously unselected package libctf0:arm64.
Preparing to unpack .../08-libctf0_2.35.2-2_arm64.deb ...
Unpacking libctf0:arm64 (2.35.2-2) ...
Selecting previously unselected package binutils-aarch64-linux-gnu.
Preparing to unpack .../09-binutils-aarch64-linux-gnu_2.35.2-2_arm64.deb ...
Unpacking binutils-aarch64-linux-gnu (2.35.2-2) ...
Selecting previously unselected package binutils.
Preparing to unpack .../10-binutils_2.35.2-2_arm64.deb ...
Unpacking binutils (2.35.2-2) ...
Selecting previously unselected package cmake-data.
Preparing to unpack .../11-cmake-data_3.18.4-2+deb11u1_all.deb ...
Unpacking cmake-data (3.18.4-2+deb11u1) ...
Selecting previously unselected package libarchive13:arm64.
Preparing to unpack .../12-libarchive13_3.4.3-2+b1_arm64.deb ...
Unpacking libarchive13:arm64 (3.4.3-2+b1) ...
Selecting previously unselected package libjsoncpp24:arm64.
Preparing to unpack .../13-libjsoncpp24_1.9.4-4_arm64.deb ...
Unpacking libjsoncpp24:arm64 (1.9.4-4) ...
Selecting previously unselected package librhash0:arm64.
Preparing to unpack .../14-librhash0_1.4.1-2_arm64.deb ...
Unpacking librhash0:arm64 (1.4.1-2) ...
Selecting previously unselected package cmake.
Preparing to unpack .../15-cmake_3.18.4-2+deb11u1_arm64.deb ...
Unpacking cmake (3.18.4-2+deb11u1) ...
Selecting previously unselected package libcc1-0:arm64.
Preparing to unpack .../16-libcc1-0_10.2.1-6_arm64.deb ...
Unpacking libcc1-0:arm64 (10.2.1-6) ...
Selecting previously unselected package libitm1:arm64.
[... 219 lines cut ...]
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find PCRE2 (missing: PCRE2_LIBRARY PCRE2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindPCRE2.cmake:27 (find_package_handle_standard_args)
  CMakeLists.txt:92 (find_package)

-- Configuring incomplete, errors occurred!
See also "/tmp/DietPi-Software/myMPD-master/release/CMakeFiles/CMakeOutput.log".
See also "/tmp/DietPi-Software/myMPD-master/release/CMakeFiles/CMakeError.log".

---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

```

Joulinar, 2021-12-26:
@supersexy 
you can bypass the issue by installing a package in addition

```
apt install libpcre2-dev
```

once done, retry myMPD install

BTW: we don't install software in a sandbox mode. We use the standard `apt` package manager as it would be on any other Debian bases system.

---

@MichaIng 
I have seen myMPD offers pre-build packages https://jcorporation.github.io/myMPD/installation/prebuild-packages

Debian: https://download.opensuse.org/repositories/home:/jcorporation/Debian_11/
Raspbian: https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_11/

Question is if we might switch. This reduce dependency and installation time 😃

MichaIng, 2021-12-27:
Interesting, we install `libpcre3-dev` which looks newer than `libpcre2-dev`, but actually it is the other way round, confusing naming 😄. So myMPD migrated to the newer library 👍.

MichaIng, 2022-01-01:
Switched to official APT repository. Awesome is that the setup/installed files match 100% what the build script does, so it is a seamless migration: #5122

Solves issues due to this an future dependency changes, clean purges assured also when new files are added etc.

closed 2022-01-02 by MichaIng