MichaIng/DietPi #5116: Unformated boot partition after flashing rock64 image

thomashollier, 2021-12-26:
When I setup my SD card for the 7.9 rock64 board, the first boot partition comes out un formatted and 16 megs. This means I can not update the dietpi.txt file or make any adjustments to the configs. Is this normal? If so, how do I update my dietpi.txt file?

Adding a screen capture of what windows "dick management" tool shows, with the latest rock64 omage, and also with the latest rpi for comparison.
[image]
[image]

Joulinar, 2021-12-26:
You can't compare RPi images with Rock64 as they have/use a different layout. The Rock64 image contains a single ext4 partition only. There is no FAT partition like RPi images have. You can check it yourself by downloading the image and open the *.img file using a tool like 7zip.

thomashollier, 2021-12-26:
Ah, thanks. Good to know. The thing is that the image I end up with on my sd card doesn't boot, and looking at the log from "rufus.exe" I added to the original post, it shows the one partition as non bootable. 
Am I missing another step?

thomashollier, 2021-12-27:
Apologies: it appears that my power brick was messed up and it all started working once I swapped it to another one.

closed 2021-12-27 by thomashollier