MichaIng/DietPi #5117: installing docker compose with dietpi-software throws error

hueppinr, 2021-12-27:
#### Details:
- Date           | Mon Dec 27 20:39:04 CET 2021
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RalphsDietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `python3 ./get-pip.py`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...install docker compose through dietpi-software
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/usr/local/lib/python3.9/dist-packages/_distutils_hack/__init__.py:24: UserWarning: Distutils was imported before Setuptools, but importing Setuptools also replaces the `distutils` module in `sys.modules`. This may lead to undesirable behaviors or errors. To avoid these issues, avoid using distutils directly, ensure that setuptools is installed in the traditional way (e.g. not an editable install), and/or make sure that setuptools is always imported before distutils.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/_distutils_hack/__init__.py:36: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
Traceback (most recent call last):
  File "/tmp/DietPi-Software/./get-pip.py", line 27081, in <module>
    main()
  File "/tmp/DietPi-Software/./get-pip.py", line 139, in main
    bootstrap(tmpdir=tmpdir)
  File "/tmp/DietPi-Software/./get-pip.py", line 120, in bootstrap
    args = determine_pip_install_arguments()
  File "/tmp/DietPi-Software/./get-pip.py", line 65, in determine_pip_install_arguments
    import setuptools  # noqa
  File "/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py", line 8, in <module>
    import _distutils_hack.override  # noqa: F401
  File "/usr/local/lib/python3.9/dist-packages/_distutils_hack/override.py", line 1, in <module>
    __import__('_distutils_hack').do_override()
  File "/usr/local/lib/python3.9/dist-packages/_distutils_hack/__init__.py", line 73, in do_override
    ensure_local_distutils()
  File "/usr/local/lib/python3.9/dist-packages/_distutils_hack/__init__.py", line 61, in ensure_local_distutils
    assert '_distutils' in core.__file__, core.__file__
AssertionError: /usr/lib/python3.9/distutils/core.py
```

Joulinar, 2021-12-27:
looks like issue is with Python3 reinstall and not with Docker-Compose. I guess you already have Python3 installed. Could you check pls

``
dietpi-software list | grep "ID 130"
````

hueppinr, 2021-12-27:
Reply is:

***@***.***:~# dietpi-software list | grep "ID 130"

ID 130 | =0 | Python 3: Runtime system, pip package installer and development headers | | https://dietpi.com/docs/software/programming/#python-3

***@***.***:~#

hueppinr, 2021-12-27:
I have the following packages installed:

 

***@***.***:~# dietpi-software list | grep "=2 "

ID 17 | =2 | Git: Clone and manage Git repositories locally |

ID 38 | =2 | FreshRSS: self-hosted RSS feed aggregator | +MariaDB +PHP +webserver | https://dietpi.com/docs/software/social/#freshrss

ID 47 | =2 | ownCloud: File sync, sharing and collaboration platform | +MariaDB +PHP +Redis +webserver | https://dietpi.com/docs/software/cloud/#owncloud

ID 50 | =2 | Syncthing: backup and sync server with web interface | | https://dietpi.com/docs/software/cloud/#syncthing

ID 57 | =2 | Baïkal: lightweight caldav + carddav server | +MariaDB +PHP +webserver | https://dietpi.com/docs/software/social/#baikal

ID 73 | =2 | Fail2Ban: prevents brute-force attacks with ip ban | | https://dietpi.com/docs/software/system_security/#fail2ban

ID 78 | =2 | LESP: Nginx    | SQLite  | PHP | +Nginx +SQLite +PHP | https://dietpi.com/docs/software/webserver_stack/#lesp-web-stack

ID 79 | =2 | LEMP: Nginx    | MariaDB | PHP | +Nginx +MariaDB +PHP | https://dietpi.com/docs/software/webserver_stack/#lemp-web-stack

ID 85 | =2 | Nginx: Lightweight webserver | +PHP | https://dietpi.com/docs/software/webserver_stack/#nginx

ID 87 | =2 | SQLite: Persistent single-file database system | | https://dietpi.com/docs/software/databases/#sqlite

ID 88 | =2 | MariaDB: Persistent cached file-per-table database server | | https://dietpi.com/docs/software/databases/#mariadb

ID 89 | =2 | PHP: Hypertext Preprocessor for dynamic web content | | https://dietpi.com/docs/software/webserver_stack/#php

ID 91 | =2 | Redis: Volatile in-memory non-SQL database server | | https://dietpi.com/docs/software/databases/#redis

ID 92 | =2 | Certbot: Obtain and renew Let's Encrypt SSL certs for HTTPS | | https://dietpi.com/docs/software/system_security/#lets-encrypt

ID 93 | =2 | Pi-hole: block adverts for any device on your network | +Git +SQLite +PHP +webserver | https://dietpi.com/docs/software/dns_servers/#pi-hole

ID 103 | =2 | DietPi-RAMlog: Makes /var/log a RAM disk, preserves file structure on reboot | | https://dietpi.com/docs/software/log_system/#dietpi-ramlog

ID 104 | =2 | Dropbear: Lightweight SSH server | | https://dietpi.com/docs/software/ssh/#dropbear

ID 162 | =2 | Docker: Build, ship, and run distributed applications | | https://dietpi.com/docs/software/programming/#docker

ID 182 | =2 | Unbound: validating, recursive, caching DNS resolver | | https://dietpi.com/docs/software/dns_servers/#unbound

ID 185 | =2 | Portainer: Simplifies container management in Docker (standalone host) | +Docker | https://dietpi.com/docs/software/programming/#portainer

ID 200 | =2 | DietPi-Dashboard (beta): Official lightweight DietPi web interface (Rust) | | https://dietpi.com/docs/software/system_stats/#dietpi-dashboard

Joulinar, 2021-12-27:
ok it is not installed. Maybe Python was installed on another way already. On a quick test, removing `setuptools` before installation of Python3 was bypassing the issue. Try following

```
pip3 uninstall setuptools -y
dietpi-software install 134
```

hueppinr, 2021-12-27:
This worked perfectly. Thank you so much. 

Can I buy you a beer?

Joulinar, 2021-12-27:
> Can I buy you a beer?

of course you could do https://dietpi.com/contribute.html

---

@MichaIng 
probably something we might need to look into. There seems to be something changes on `setuptools v60` https://github.com/pypa/pip/issues/10742

MichaIng, 2021-12-28:
Hmm, a "bug" in the way `setuptools`, `distutils` and `get-pip.py` play together 🤔. `setuptools` is basically superseding `distutils` since a long time, so removing it from `pip` dependencies seems to be the most reasonable step to have this resolving itself. Looks like this was aimed by some guys before the step which `setuptools` made now, but obviously it was missed: https://github.com/pypa/pip/issues/8761

If uninstalling `setuptools` is the quickest workaround, we can adapt it until `pip` ships a proper solution.

MichaIng, 2022-01-02:
Applied the workaround: https://github.com/MichaIng/DietPi/commit/d2de5e4
Played a bit around, there seems to be really no other solution. However, uninstalling `setuptools` is a very quick step also the attempt "succeeds" if it is not installed, so no real downsides to do so until it got fixed upstream.

closed 2022-01-02 by MichaIng

Joulinar, 2022-01-02:
hope python devs will find a permanent solution soon.