MichaIng/DietPi #5120: DietPi-Services | Changing a service from DietPi controlled to Systemd controlled doesn't work

ravenclaw900, 2021-12-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='dev'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
bullseye
- Kernel version | `uname -a`
Linux rpi4 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
CanaKit
- SDcard used | (EG: SanDisk ultra)
Samsung EVO+ 32GB

#### Steps to reproduce

1. Mark a service as Systemd controlled instead of DietPi controlled

#### Expected behaviour

- Service should not stop e.g. on software installs

#### Actual behaviour

- Service stops on software installs, after one install goes back to "DietPi Controlled"

MichaIng, 2021-12-30:
Many thanks for your report. This is actually the expected behaviour, though it is confusing, I admit:
- `DietPi controlled` vs `systemd controlled` defines only what/when the service is started at boot. `systemd controlled` has the service enabled, like when you do `systemctl enable <service>`, so that systemd starts it during earlier boot stage, depending on `Before=`/`After=` definitions only. `DietPi controlled` has the service disabled, like when you do `systemctl disable <service>`, so that systemd does not start it, but `dietpi-services` from `dietpi-postboot.service` at a later boot stage in defined order so that all dependencies/backends are started before middleware before frontend/dependant services.
- If you want to stop `dietpi-services` from starting/stopping a service, you need to exclude it, which you can do via `dietpi-services` command, selecting the service, then `Include/Exclude`. To have it still starting at boot, it then needs to be enabled/systemd-controlled.

This behaviour is however indeed confusing and bears some issues as well. We want to change this so that all services are started by systemd at boot, that "disabled" really means "disabled", that APT upgrades keep starting enabled services after a related package upgrade etc. What we need to do for that is adding some `Before=`/`After=` definitions to the service files shipped by Debian packages, otherwise the webserver could start before PHP, PHP before the database, Sonarr/Radarr/Lidarr before the downloader/indexer servers etc, causing at best ugly error messages during boot or even leading to service hangs and crashes. At least for our own service files (those not shipped by Debian packages), this has been done already, but it is an intrusive change, hence requires a full development/version iteration for testing it thoroughly.

ravenclaw900, 2021-12-30:
Ah, maybe a note about that on the 'mode' page. However, it still changes back to 'DietPi Controlled' (and starts up accordingly) after one software install (if not excluded).

MichaIng, 2021-12-30:
> However, it still changes back to 'DietPi Controlled' (and starts up accordingly) after one software install (if not excluded).

That is true, also as failsafe step to have all possibly involved services starting in defined order at boot. Consequently both makes sense in combination 🤔. However, I won't change something on the logic before dropping that "DietPi controlled" thing all together. One of the old behaviours I wanted to remove/align with plain distros for a long time 😉.

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng