MichaIng/DietPi #5121: Transmission Web UI shows no torrents because of a 409 Conflict response from a RPC call

igorkulman, 2021-12-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.9.3`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux`
- SBC model | `RPi 2 Model B (armv7l)`

#### Additional Information (if applicable)
- Software title | `transmission-daemon 2.94 (d8e60ee44f)`
- Bug report ID | `a3ed5e6a-3282-451c-9e14-7446fd21d052`

#### Steps to reproduce

1. Navigate to http://dietpi.local:9091/transmission/web/

#### Expected behaviour
- See a list of torrents

#### Actual behaviour

- The UI is loaded but shows no torrents

#### Extra details
Inspecting the requests shows a 409 Conflict response from a RPC call

[image]

This started happening after the latest `apt upgrade`.

closed 2021-12-31 by igorkulman

MichaIng, 2021-12-31:
Did it solve itself?

igorkulman, 2021-12-31:
Yes I reinstalled transmission and rebooted and that solved it.

Joulinar, 2021-12-31:
Probably something pending reboot after `apt` package installation