MichaIng/DietPi #5123: Update to 7.9.3 failed on Odroid XU4

jcbenoist, 2022-01-02:
#### Details:
- Date           | Sun Jan  2 01:01:15 CET 2022
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | `Linux DietpiXU4 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Start from Dietpi 7.8.2 on Odroid XU4
2. Type dietpi-update
#### Expected behaviour:

- Update successfull
#### Actual behaviour:

- Update failure :  E: The repository 'http://www.ui.com/downloads/unifi/debian stable Release' no longer has a Release file.
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:3 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:4 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Get:5 https://dietpi.com/meveric all InRelease [19.4 kB]
Get:6 https://dietpi.com/meveric buster InRelease [4131 B]
Ign:7 http://www.ui.com/downloads/unifi/debian stable InRelease
Err:8 http://www.ui.com/downloads/unifi/debian stable Release
  404  Not Found [IP: 34.218.155.36 80]
Get:9 https://deb.debian.org/debian-security buster/updates/main armhf Packages [307 kB]
Get:10 https://deb.debian.org/debian buster-backports/main armhf Packages.diff/Index [27.8 kB]
Get:11 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-15-2007.08.pdiff [676 B]
Get:12 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-16-0209.14.pdiff [1139 B]
Get:13 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-21-0202.23.pdiff [449 B]
Get:14 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-21-2001.16.pdiff [1472 B]
Get:15 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-25-0205.37.pdiff [1611 B]
Get:15 https://deb.debian.org/debian buster-backports/main armhf Packages 2021-12-25-0205.37.pdiff [1611 B]
Get:16 https://dietpi.com/meveric all/main armhf Packages [41.1 kB]
Get:17 https://dietpi.com/meveric buster/main armhf Packages [23.0 kB]
Reading package lists...
E: The repository 'http://www.ui.com/downloads/unifi/debian stable Release' no longer has a Release file.
```

Joulinar, 2022-01-02:
Hi,

looks like you added a 3rd party repository `http://www.ui.com/downloads/unifi/debian` and it seems not working or is incorrect. You would need to check this and correct the source configuration. Best is to recreate source file according UniFi docs

https://help.ui.com/hc/en-us/articles/220066768-UniFi-Network-How-to-Install-and-Update-via-APT-on-Debian-or-Ubuntu

MichaIng, 2022-02-06:
Marking as closed due to outstanding reply. Feel free to reopen when required.

closed 2022-02-06 by MichaIng