MichaIng/DietPi #5126: Exit code 132 during upgrade

kuehlakku, 2022-01-02:
### Creating a bug report/issue
#### Details:
- Date           | Sun Jan  2 11:14:15 GMT 2022
- Bug report     | b7f285da-8d69-4c39-98b6-a136bdfa3a6e
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq upgrade`
- Exit code      | 132
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```

Joulinar, 2022-01-02:
Pls can you share full output of

`apt update && apt upgrade`

MichaIng, 2022-01-02:
Not sure if it's related, but a file shipped by the kernel package is malformed:
```
----------
lsmod
----------
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 3: ignoring bad line starting with 'softdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 4: ignoring bad line starting with 'softd%p'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 5: ignoring bad line starting with '3oFtdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 6: ignoring bad line starting with 's^?ftdep`cifs'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 10: ignoring bad line starting with 'sgftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 11: ignoring bad line starting with '�of4deq'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 13: ignoring bad line starting with 'koftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 16: ignoring bad line starting with 'softde�'
```
This file can be re-created with the following command:
```sh
depmod -a
```
And yes, please run the commands suggested by Joulinar. The bug report upload doesn't contain further info about the failed APT upgrade either.

kuehlakku, 2022-01-02:
root@DietPi:~# apt update && apt upgrade
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
16 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Illegal instruction

kuehlakku, 2022-01-02:
root@DietPi:~# lsmod
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 3: ignoring bad line starting with 'softdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 4: ignoring bad line starting with 'softd%p'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 5: ignoring bad line starting with '3oFtdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 6: ignoring bad line starting with 'ftdep`cifs'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 10: ignoring bad line starting with 'sgftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 11: ignoring bad line starting with '▒of4deq'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 13: ignoring bad line starting with 'koftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 16: ignoring bad line starting with 'softde▒'
Module                  Size  Used by
md4                    16384  0
md5                    16384  1
nls_utf8               16384  3
cifs                  802816  2
libarc4                16384  1 cifs
binfmt_misc            24576  1
raspberrypi_hwmon      16384  0
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
aes_neon_bs            28672  0
sha256_generic         16384  1
aes_neon_blk           28672  2 aes_neon_bs
crypto_simd            24576  2 aes_neon_bs,aes_neon_blk
cryptd                 28672  1 crypto_simd
fuse                  131072  1
ip_tables              32768  0
x_tables               45056  1 ip_tables
ipv6                  532480  22
root@DietPi:~# depmod -a
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 3: ignoring bad line starting with 'softdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 4: ignoring bad line starting with 'softd%p'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 5: ignoring bad line starting with '3oFtdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 6: ignoring bad line starting with 'ftdep`cifs'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 10: ignoring bad line starting with 'sgftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 11: ignoring bad line starting with '▒of4deq'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 13: ignoring bad line starting with 'koftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 16: ignoring bad line starting with 'softde▒'
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/oss/snd-pcm-oss.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/snd-timer.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/snd-hrtimer.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/seq/snd-seq-dummy.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/seq/snd-seq-midi-event.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/core/snd-compress.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/drivers/snd-dummy.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/drivers/snd-aloop.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/drivers/snd-virmidi.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/drivers/mpu401/snd-mpu401-uart.ko: Exec format error
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/codecs/snd-soc-adau1977-i2c.ko: Exec format error
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/codecs/snd-soc-cs4265.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/codecs/snd-soc-cs4271-i2c.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/bcm/snd-soc-googlevoicehat-codec.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/bcm/snd-soc-iqaudio-codec.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/bcm/snd-soc-i-sabre-q2m.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/bcm/snd-soc-allo-boss-dac.ko: Invalid argument
[... cut ...] nvalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/sound/soc/bcm/snd-soc-rpi-wm8804-soundcard.ko: Exec format error
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/net/xfrm/xfrm_ipcomp.ko: Invalid argument
depmod: ERROR: failed to load symbols from /lib/modules/5.10.63-v8+/kernel/net/sunrpc/auth_gss/rpcsec_gss_krb5.ko: Invalid argument
depmod: FATAL: Module index: bad character '▒'=0xe4 - only 7-bit ASCII is supported:
)2c:a▒!936

MichaIng, 2022-01-02:
Uff there seem to be more files corrupted. Please try to repair the kernel first that way:
```sh
cd /tmp
wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20211118-3_arm64.deb
dpkg-deb -x raspberrypi-kernel_1.20211118-3_arm64.deb raspberrypi-kernel
rm raspberrypi-kernel_1.20211118-3_arm64.deb
cp -a raspberrypi-kernel/. /
ls -l /boot
lsmod
```

kuehlakku, 2022-01-02:
cd /tmp
wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20211118-3_arm64.deb
dpkg-deb -x raspberrypi-kernel_1.20211118-3_arm64.deb raspberrypi-kernel
rm raspberrypi-kernel_1.20211118-3_arm64.deb
cp -a raspberrypi-kernel/. /
ls -l /boot
lsmod
--2022-01-02 18:32:09--  https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20211118-3_arm64.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.151, 176.126.240.84, 93.93.130.212, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.151|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 21840456 (21M) [application/x-debian-package]
Saving to: ‘raspberrypi-kernel_1.20211118-3_arm64.deb’

raspberrypi-kernel_1.20211118-3_arm64.deb 100%[=====================================================================================>]  20.83M  8.86MB/s    in 2.4s

2022-01-02 18:32:12 (8.86 MB/s) - ‘raspberrypi-kernel_1.20211118-3_arm64.deb’ saved [21840456/21840456]

cp: cannot overwrite non-directory '/./lib' with directory 'raspberrypi-kernel/./lib'
total 29767
-rwxr-xr-x 1 root root   18693 Nov 18 16:22  COPYING.linux
-rwxr-xr-x 1 root root    1594 Dec  5 18:53  LICENCE.broadcom
drwxr-xr-x 2 root root     512 Dec  5 19:36 'System Volume Information'
-rwxr-xr-x 1 root root   27441 Nov 18 16:22  bcm2710-rpi-2-b.dtb
-rwxr-xr-x 1 root root   29558 Nov 18 16:22  bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root   28939 Nov 18 16:22  bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   27429 Nov 18 16:22  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   28100 Nov 18 16:22  bcm2710-rpi-zero-2.dtb
-rwxr-xr-x 1 root root   49833 Nov 18 16:22  bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root   49877 Nov 18 16:22  bcm2711-rpi-400.dtb
-rwxr-xr-x 1 root root   50555 Nov 18 16:22  bcm2711-rpi-cm4.dtb
-rwxr-xr-x 1 root root   52456 Dec  5 18:52  bootcode.bin
-rwxr-xr-x 1 root root     112 Dec  5 19:44  cmdline.txt
-rwxr-xr-x 1 root root    3181 Dec  5 19:44  config.txt
drwxr-xr-x 4 root root    3072 Dec  6 08:59  dietpi
-rwxr-xr-x 1 root root   18092 Oct 21 18:50  dietpi-LICENSE.txt
-rwxr-xr-x 1 root root   14711 Oct 21 18:50  dietpi-README.md
-rwxr-xr-x 1 root root   14969 Dec  5 19:45  dietpi.txt
-rwxr-xr-x 1 root root    7279 Dec  5 18:52  fixup.dat
-rwxr-xr-x 1 root root    5411 Dec  5 18:53  fixup4.dat
-rwxr-xr-x 1 root root    3213 Dec  5 18:53  fixup4cd.dat
-rwxr-xr-x 1 root root    8426 Dec  5 18:53  fixup4db.dat
-rwxr-xr-x 1 root root    8420 Dec  5 18:53  fixup4x.dat
-rwxr-xr-x 1 root root    3213 Dec  5 18:52  fixup_cd.dat
-rwxr-xr-x 1 root root   10269 Dec  5 18:52  fixup_db.dat
-rwxr-xr-x 1 root root   10269 Dec  5 18:52  fixup_x.dat
-rwxr-xr-x 1 root root 7906048 Nov 18 16:22  kernel8.img
drwxr-xr-x 2 root root   20992 Nov 18 16:22  overlays
-rwxr-xr-x 1 root root       0 Dec  5 19:39  ssh
-rwxr-xr-x 1 root root 2965696 Dec  5 18:52  start.elf
-rwxr-xr-x 1 root root 2241504 Dec  5 18:52  start4.elf
-rwxr-xr-x 1 root root  801564 Dec  5 18:52  start4cd.elf
-rwxr-xr-x 1 root root 3737928 Dec  5 18:52  start4db.elf
-rwxr-xr-x 1 root root 2993544 Dec  5 18:53  start4x.elf
-rwxr-xr-x 1 root root  801564 Dec  5 18:52  start_cd.elf
-rwxr-xr-x 1 root root 4809800 Dec  5 18:52  start_db.elf
-rwxr-xr-x 1 root root 3717256 Dec  5 18:52  start_x.elf
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 3: ignoring bad line starting with 'softdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 4: ignoring bad line starting with 'softd%p'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 5: ignoring bad line starting with '3oFtdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 6: ignoring bad line starting with 'ftdep`cifs'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 10: ignoring bad line starting with 'sgftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 11: ignoring bad line starting with '▒of4deq'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 13: ignoring bad line starting with 'koftdep'
libkmod: ERROR ../libkmod/libkmod-config.c:657 kmod_config_parse: /lib/modules/5.10.63-v8+/modules.softdep line 16: ignoring bad line starting with 'softde▒'
Module                  Size  Used by
md4                    16384  0
md5                    16384  1
nls_utf8               16384  3
cifs                  802816  2
libarc4                16384  1 cifs
binfmt_misc            24576  1
raspberrypi_hwmon      16384  0
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
aes_neon_bs            28672  0
sha256_generic         16384  1
aes_neon_blk           28672  2 aes_neon_bs
crypto_simd            24576  2 aes_neon_bs,aes_neon_blk
cryptd                 28672  1 crypto_simd
fuse                  131072  1
ip_tables              32768  0
x_tables               45056  1 ip_tables
ipv6                  532480  22

Joulinar, 2022-01-02:
I guess it did not change anything but all commands to be executed one by one and not together 😉

MichaIng, 2022-01-02:
Ah no the problem is that `/lib` is a symlink to `/usr/lib` and hence cannot be overwritten with a directory on recursive copy. Then it needs to be installed via `dpkg` (as `apt` fails with illegal instruction as well):
```
cd /tmp
wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20211118-3_arm64.deb
dpkg -i raspberrypi-kernel_1.20211118-3_arm64.deb
rm raspberrypi-kernel_1.20211118-3_arm64.deb
lsmod
```

kuehlakku, 2022-01-03:
Many thanks. After performing the mentioned steps, i wasn't able to reboot the system. Don't worry, I did a new installation with the last image file. Everything is running now. Issue can be closed

closed 2022-01-03 by Joulinar

MichaIng, 2022-01-03:
Okay, yeah there were other system files corrupted, so more repairing would have been required to assure that a reboot works find. Probably a wise step to start from scratch.