MichaIng/DietPi #5128: [feature request] YDNS support via DietPi-DDNS tool

edmundlaugasson, 2022-01-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

As [YDNS](https://ydns.io/) is [free and open-source](https://github.com/ydns), then why not include it into DietPi.

#### Describe the solution you'd like:

When opening DietPi-DDNS tool, would like to see YDNS choice there. [Appropriate bash DDNS updater script is here](https://github.com/ydns/bash-updater).

#### Describe alternatives you've considered:

There are alternatives available at DietPi-DDNS tool

#### Additional context

This should work on all supported platforms of DietPi architecture (armhf, aarch64, etc).

MichaIng, 2022-01-03:
Many thanks for your request.

Bare API docs are here: https://ydns.io/api/v1/#update
So instead of invoking a whole bash script, we can do a simple `curl` call like we do in the other cases 🙂.

TDuffinNTU, 2023-10-06:
Would this be a good newcomer issue to tackle? From what I can tell I just need to modify dietpi-ddns:

* Add YDNS to USAGE and as a commandline option via Read()
* Add YDNS update URI to Apply()
* Add YDNS-specific option(s) to other methods.

Any other scripts that need updating to accommodate the change?

MichaIng, 2023-10-07:
Jep, only this one script needs change. Searching for e.g. "DuckDNS" though the script and adding respective YDNS blocks should so it.

MichaIng, 2023-10-14:
Done with: #6674

PR #6674 (2023-10-14): DietPi-DDNS | Add provider YDNS
Closes #5128 

As YDNS expects a single (one and only one) domain in its API request I've added a special case for it in Menu_Domains but this might not be correct (do we submit domains as a  _list_ or is it split by the commas and sent in multiple requests?

There are two options for YDNS updating - One uses an auth header and the other uses an "update link" which is a shortened variant of the update URI but with a random token. I opted for the former since I feel that's more secure and easier to fit into the UX of the application. 

Remaining in draft until I  [... cut]
closed 2023-10-14 by MichaIng