MichaIng/DietPi #5130: docker - mnt docker tmp folder as tmpfs

isarrider, 2022-01-03:
Hi,

has anybody already experimented with the usage of tmpfs for
/mnt/dietpi_userdata/docker-data/tmp

My Pi4 has plenty of RAM, and extracting the containers is otherwise with my SD-Card slow...

MichaIng, 2022-01-03:
Sounds reasonable. Since I do not actively use Docker, can you show an example of the contained data/sizes?

It could be made an own tmpfs mount or linked to `/tmp/docker_tmp` or so. Only downside of this method is that any process which accesses this directory must not run in a systemd context with `PrivateTmp=true`, as the directory would then not be visible 😉.

isarrider, 2022-01-03:
ok, I tried it...
For example when I pull Octoprint, it temporarily holds ~330MB...
(https://hub.docker.com/r/octoprint/octoprint/tags)
Works and speeds up things for me especially as my SD cards are a little older ;)

MichaIng, 2022-01-03:
The whole container image is stored in `/mnt/dietpi_userdata/docker-data/tmp`, or is it other data and just coincidence that both is ~330 MiB? 🤔

isarrider, 2022-01-03:
The container (or to be more specific the ~10 temporary files) are downloaded to /mnt/dietpi_userdata/docker-data/tmp and then extracted to the images folder... (So the tmp folder ist empty again after like 30s)...

MichaIng, 2022-01-03:
Ah I got it, so it is used (as well) as temporary download directory. Jep pretty clearly something where a tmpfs makes much sense. Even when downloading huge containers and using a swap file to cover those, it's better to only write to disk (via swapping) when really needed instead of doing that for every download.

Makes sense to create a dedicated tmpfs then with a size which covers the expected download sizes and in case other uses of this directory, instead of doing a `/tmp` symlink.

isarrider, 2022-01-03:
Perfect...

Joulinar, 2022-01-03:
@MichaIng 
should we create an own `tmpfs` pointing to `/mnt/dietpi_userdata/docker-data/tmp` once docker is installed?

MichaIng, 2022-01-03:
I wouldn't do that by default for now, as it can easily shoot low RAM systems into OOM killer crashes. This may be something to document, or a whiptail prompt which explains the consequences clearly. When one downloads a 1 GiB container while there is less than 1 GiB overall memory left: 💥! A little similar to video streaming serves transcoding directories: Awesome to have in tmpfs, but only when sufficient RAM covers the potentially large usage.

Joulinar, 2022-01-03:
an idea could be to limit this to system with sufficient physical memory, like 2GB+

MichaIng, 2022-01-03:
Though hard to say what sufficient is, as it simply depends on the used containers, when installing/updating them. Similar to like transcoding sizes depend on video quality, where HD and 4k have a massive influence, not really possible to define a generic min memory size.

Joulinar, 2022-01-03:
ok, than let's make a note on the docs

MichaIng, 2022-01-03:
Added an issue at the docs repo: https://github.com/MichaIng/DietPi-Docs/issues/656

I'll mark this as closed then. @isarrider keep us updated if you run into any issues with this, e.g. also if that directory is unexpectedly used by other large data or so, aside of temporary downloads.

closed 2022-01-03 by MichaIng

isarrider, 2022-01-04:
I will...
Looks good so far...

isarrider, 2022-01-28:
ok, it really works smooth
(of course when there is more free RAM than what is need for the complete store & unpack process)

isarrider, 2022-12-05:
ok, as long as there is enough RAM, it really speeds up things...
Never seen any issue so far...

MichaIng, 2022-12-06:
Thanks for reporting. Thinking again about adding this as option right to `dietpi-software`, but still unsure about how large the tmpfs should be and from how much physical RAM we should offer this. Probably giving this in users hands with an input box + `dietpi.txt` setting, simply allowing a tmpfs size up to physical RAM - 100 MiB or so, with two sentences that or course system memory and size of container downloads are limiting factors, and how to solve issues temporarily with `unmount /mnt/dietpi_userdata/docker-data/tmp` and persistently with `sed -i '/docker-data\/tmp/d' /etc/fstab`?