MichaIng/DietPi #5131: DietPi-Software | SAMBA, support for in-kernel ksmbd driver

GvY85, 2022-01-03:
With kernel 5.15 around the corner for Bullseye-Backports and already in Bookworm/Testing next to native NTFS support it also has an [in-kernel SAMBA driver](https://www.techrepublic.com/article/linux-kernel-5-15-is-now-available-and-it-has-something-special-for-ntfs-users/) called [ksmbd](https://www.kernel.org/doc/html/latest/filesystems/cifs/ksmbd.html).

It would be nice if DietPi can support this in the future.

MichaIng, 2022-01-03:
The problem here is that the related userspace tools are available on Bookworm only: https://packages.debian.org/bookworm/ksmbd-tools
And it doesn't ship a service or such yet + [man pages](https://manpages.debian.org/testing/ksmbd-tools/index.html) are minimal, so we need to integrate one our end and play with how to configure and start the daemon properly. Also features are still limited compared to Samba, hence it would be an alternative implementation, not a replacement for Samba.

GvY85, 2022-01-03:
I agree, it should be additional, not a replacement.
Perhaps it will be supported by Backports anytime soon, lets see.