MichaIng/DietPi #5132: Installing AdGuard Home Error

Ianszh, 2022-01-03:
#### Details:
- Date           | Mon Jan  3 17:15:29 CET 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `curl -ILfvm 5 https://static.adguard.com/adguardhome/release/AdGuardHome_linux_armv7.tar.gz`
- Exit code      | 6
- Software title | DietPi-Software
#### Steps to reproduce:

1. ... When Installing AdGuard Home
2. 
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Could not resolve host: static.adguard.com
* Closing connection 0
curl: (6) Could not resolve host: static.adguard.com
```

MichaIng, 2022-01-03:
Works fine here. Did you try the `Retry` option from the error prompt? Often such is a temporary server/network issue only.

Joulinar, 2022-01-03:
could be DNS issue. Are you able to `ping static.adguard.com`

Ianszh, 2022-01-03:
Sorry yes looks like a network problem

closed 2022-01-03 by Ianszh