MichaIng/DietPi #5133: Trying to Install pi-hole

Ianszh, 2022-01-03:
#### Details:
- Date           | Mon Jan  3 18:02:53 CET 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT a39ad391-f456-458e-8fb5-c6e25bab159a.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/bugreport/`
- Exit code      | 6
- Software title | DietPi-BugReport
#### Steps to reproduce:

1. ... Trying to Install pi-hole

#### Additional logs:
```
* Could not resolve host: ssh.dietpi.com
* Closing connection 0
* Could not resolve host: ssh.dietpi.com
* Closing connection 1
curl: (6) Could not resolve host: ssh.dietpi.com
```

Joulinar, 2022-01-03:
same issue as the other one you opened https://github.com/MichaIng/DietPi/issues/5132. 

There seems to be a challenge with DNS resolution on your network. What DNS server you are using?

Ianszh, 2022-01-03:
My desktop is fine, but the Pi has a network problem, both on the same network.
DNS servers are
81.92.97.74
81.92.97.20

Ianszh, 2022-01-03:
That's desktop, Pi is:
81.92.99.8

Joulinar, 2022-01-03:
do you have any firewall that could block DNS resolution? Strange that both use different DNS server. You could try to switch to static IP and change DNS to a global public one like Quad9 `9.9.9.9` or Cloudflare `1.1.1.1`

Ianszh, 2022-01-03:
My router Firewall has these set

[image]

Joulinar, 2022-01-03:
did you tried to change DNS server to a public one?

Ianszh, 2022-01-03:
on the pi yes but no change

Ianszh, 2022-01-03:
[image]

Ianszh, 2022-01-03:
Internet [Failure] Offline

Joulinar, 2022-01-03:
nope that's just the view but the current DNS server is shown on the top right corner. It is local host `127.0.0.1`

can you share `cat /etc/resolv.conf`

Ianszh, 2022-01-03:
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
nameserver 127.0.0.1

Joulinar, 2022-01-03:
do you recall if you installed `resolvconf` manually?

And to be sure, there is no local DNS server correct?

`ss -tulpn | grep LISTEN`

Ianszh, 2022-01-03:
do you recall if you installed `resolvconf` manually?
No I didn't

I did an apt-upgrade things were working fine before

tcp     LISTEN   0        50               0.0.0.0:139            0.0.0.0:*      users:(("smbd",pid=804,fd=35))                                                 
tcp     LISTEN   0        100            127.0.0.1:24559          0.0.0.0:*      users:(("nxserver.bin",pid=469,fd=8))                                          
tcp     LISTEN   0        100            127.0.0.1:26001          0.0.0.0:*      users:(("nxnode.bin",pid=958,fd=7))                                            
tcp     LISTEN   0        1000             0.0.0.0:22             0.0.0.0:*      users:(("dropbear",pid=371,fd=3))                                              
tcp     LISTEN   0        256            127.0.0.1:5335           0.0.0.0:*      users:(("unbound",pid=541,fd=4))                                               
tcp     LISTEN   0        511              0.0.0.0:1880           0.0.0.0:*      users:(("node-red",pid=882,fd=21))                                             
tcp     LISTEN   0        4096           127.0.0.1:8088           0.0.0.0:*      users:(("influxd",pid=482,fd=3))                                               
tcp     LISTEN   0        128            127.0.0.1:7001           0.0.0.0:*      users:(("nxnode.bin",pid=958,fd=19))                                           
tcp     LISTEN   0        100              0.0.0.0:1883           0.0.0.0:*      users:(("mosquitto",pid=862,fd=5))                                             
tcp     LISTEN   0        50               0.0.0.0:445            0.0.0.0:*      users:(("smbd",pid=804,fd=34))                                                 
tcp     LISTEN   0        100              0.0.0.0:4000           0.0.0.0:*      users:(("nxd",pid=953,fd=3))                                                   
tcp     LISTEN   0        100            127.0.0.1:12001          0.0.0.0:*      users:(("nxnode.bin",pid=958,fd=17))                                           
tcp     LISTEN   0        50                  [::]:139               [::]:*      users:(("smbd",pid=804,fd=33))                                                 
tcp     LISTEN   0        128                    *:21                   *:*      users:(("proftpd",pid=744,fd=0))                                               
tcp     LISTEN   0        1000                [::]:22                [::]:*      users:(("dropbear",pid=371,fd=4))                                              
tcp     LISTEN   0        128                [::1]:7001              [::]:*      users:(("nxnode.bin",pid=958,fd=16))                                           
tcp     LISTEN   0        100                 [::]:1883              [::]:*      users:(("mosquitto",pid=862,fd=6))                                             
tcp     LISTEN   0        50                  [::]:445               [::]:*      users:(("smbd",pid=804,fd=32))                                                 
tcp     LISTEN   0        100                 [::]:4000              [::]:*      users:(("nxd",pid=953,fd=4))

Joulinar, 2022-01-03:
let's see if we can remove package `resolvconf`

```
apt-mark auto resolvconf
G_AGA # This should purge resolvconf, if it is not required by any other installed APT package
```

does this remove the package?

Ianszh, 2022-01-03:
Still `DNS = 127.0.0.1`

Joulinar, 2022-01-03:
do now 

```
echo 'nameserver 1.1.1.1 1.0.0.1' > /etc/resolv.conf
reboot
```

Ianszh, 2022-01-03:
Success Online 👍🏼
Thanks a lot, do you think the apt-upgrade cause it?

Joulinar, 2022-01-03:
not sure, usually it should not be possible to install such packages without need. Means, something did a trigger to have it installed. As well setting it to localhost as DNS server is something not happen out of the box. I know this was a behavior of PiHole in past but changed some time back. But I understood the installation did not finished. Probably we will not find out.

Ianszh, 2022-01-03:
Ok, things seem to be installing fine now.
Thanks again for your quick and very helpful response!!

MichaIng, 2022-01-03:
If it changes again, report back, then we need to check which process does this change. `resolvconf` itself does not touch `/etc/resolv.conf`, but other programs (or users), like DHCP clients, can call `resolvconf` to add/remove/change DNS nameserver entries. But those can also directly edit `/etc/resolv.conf`, when `resolvconf` is not available 😉.

Ianszh, 2022-01-03:
Ok will do 🤘🏼

Joulinar, 2022-01-03:
well hard to find out as long as PiHole or AGH are installed. Because both will resolve local DNS request 😉 

OK going to close the issue as it has been solved. Feel free to reopen if needed

MichaIng, 2022-01-03:
Now I remember:
```sh
systemctl status unbound-resolvconf.service
```
This service aims exactly to add `127.0.0.1` as local DNS nameserver. It only does something when the `resolvconf` command is available. I'd say we should always disable this service on Unbound install. Much better to leave the choice whether using Unbound (or Pi-hole/AGH) as local DNS or not to user, to configure actively via network settings, leaving it up to DHCP or whichever method. Having another service which implicitly messes with the local nameservers causes only issues and confusion. What you think?

Joulinar, 2022-01-03:
Fully agree on this. Personally I still think using localhast as DNS resolver is a worse idea 😉

MichaIng, 2022-01-03:
Done: https://github.com/MichaIng/DietPi/commit/b528c19

closed 2022-01-04 by MichaIng