MichaIng/DietPi #5134: error installing sabnzbd via dietpi-software

clarky0000, 2022-01-04:
Trying to install sabnzbd via dietpi-software. 
I perviously had an old version installed - it may have been installed using a different method as listed here: https://sabnzbd.org/wiki/installation/install-debian. I really can't remember. I removed all files located at /etc/sabnzbd and /usr/lib/sabnzbd before trying to install via dietpi-software
--------------------------------------------------------
Getting the following error:

#### Details:
- Date           | Tue Jan  4 18:48:28 AEDT 2022
- Bug report     | 2cd0e752-6173-4664-afbe-5d95a88cc5be
- DietPi version | v6.34.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `python3 ./get-pip.py`
- Exit code      | 1
- Software title | DietPi-Software

#### Additional logs:
```
/usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:25: UserWarning: Distutils was imported before Setuptools, but importing Setuptools also replaces the `distutils` module in `sys.modules`. This may lead to undesirable behaviors or errors. To avoid these issues, avoid using distutils directly, ensure that setuptools is installed in the traditional way (e.g. not an editable install), and/or make sure that setuptools is always imported before distutils.
  "Distutils was imported before Setuptools, but importing Setuptools "
/usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py:36: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
Traceback (most recent call last):
  File "./get-pip.py", line 27081, in <module>
    main()
  File "./get-pip.py", line 139, in main
    bootstrap(tmpdir=tmpdir)
  File "./get-pip.py", line 120, in bootstrap
    args = determine_pip_install_arguments()
  File "./get-pip.py", line 65, in determine_pip_install_arguments
    import setuptools  # noqa
  File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 8, in <module>
    import _distutils_hack.override  # noqa: F401
  File "/usr/local/lib/python3.7/dist-packages/_distutils_hack/override.py", line 1, in <module>
    __import__('_distutils_hack').do_override()
  File "/usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py", line 73, in do_override
    ensure_local_distutils()
  File "/usr/local/lib/python3.7/dist-packages/_distutils_hack/__init__.py", line 61, in ensure_local_distutils
    assert '_distutils' in core.__file__, core.__file__
AssertionError: /usr/lib/python3.7/distutils/core.py
```

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster 1
- Kernel version | Linux DietPi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)

Joulinar, 2022-01-04:
This is a known issue with latest `Setuptools` version https://github.com/MichaIng/DietPi/issues/5117

A permanent fix will become available in next version DietPi v8

A workaround could be following before trying to install/reinstall Python

`pip3 uninstall setuptools -y`

---

Side note
- you are running very old DietPi version v6.34. consider to update by running `dietpi-update`
- as well you are running oldstable Debian Buster. consider to upgrade to Bullseye https://dietpi.com/blog/?p=811

clarky0000, 2022-01-04:
Im getting a command not found. 

root@DietPi:~# pip3 uninstall setuptools -y
-bash: pip3: command not found

What am I missing?

Joulinar, 2022-01-04:
strange, Python 3 is installed and it should work. Can you try 

`pip uninstall setuptools -y`

MichaIng, 2022-01-04:
The issue has been fixed with next `setuptools` release btw: https://github.com/pypa/setuptools/commit/41fa663

Not sure how now `setuptools` is installed while `pip3` is not available. Hammer solution would be:
```sh
rm -R /usr/local/lib/python3.7/dist-packages/setuptools
dietpi-software install 139
```

Joulinar, 2022-01-04:
we have a similar issue (as well Buster) on our forum. https://dietpi.com/phpbb/viewtopic.php?t=9877

Next issue might be `cryptography` Looks like pre-compiled wheel  `cryptography 36.0.0` did not satisfy sabnzbd. Something to test if it really require `36.0.1` which is not available for Python 3.7 https://www.piwheels.org/project/cryptography/

MichaIng, 2022-01-04:
The build on piwheels failed indeed: https://piwheels.org/project/cryptography/

MichaIng, 2022-01-04:
Issue solved, the wheel is available now.

closed 2022-01-04 by MichaIng

clarky0000, 2022-01-04:
> The issue has been fixed with next `setuptools` release btw: [pypa/setuptools@41fa663](https://github.com/pypa/setuptools/commit/41fa663) Watch out for v60.3.0 here: https://pypi.org/project/setuptools/#history
> 
> Not sure how now `setuptools` is installed while the `pip3` command is not available. Hammer solution would be:
> 
> ```shell
> rm -R /usr/local/lib/python3.7/dist-packages/setuptools
> dietpi-software install 139
> ```

That appears to have done the trick - thanks guys. I ran:

pip uninstall setuptools -y
then
 rm -R /usr/local/lib/python3.7/dist-packages/setuptools
dietpi-software install 139

thanks guys!

MichaIng, 2022-01-04:
Great. I accidentally closed the issue as the `cryptography` issue was solved, and forgot that yours was actually the other one with `setuptools`, sorry for that 😅. However, upstream fix for that one should go live soon as well + we have the workaround applied 🙂.

clarky0000, 2022-01-05:
Great. So to apply that update when it does what do I do?

MichaIng, 2022-01-05:
Since the workaround did it for you now, there is no need or benefit to apply any further step. DietPi v8.0 will be released at Saturday, after which a Python 3 reinstall will succeed thanks to the workaround, and as fast as `setuptools` v60.3.0 has been released that workaround won't be required either. I hope this happens until Saturday so that we can remove our workaround again 😉.