MichaIng/DietPi #5135: TinkerBoard image does not work with first generation board without emmmc!

chjan, 2022-01-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...Try on tinkerboard original edition. It stops after a few seconds. Also, the sdcard to which flash is written becomes unreadable by macos, unlike any other flash (android etc) on this same sd card 32gb.

I suspect this build needs a TINKER BOARD S to work because it has MASKROM jumper. The original Tinker board does not. You need to create a version of DietPi for that board.
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2022-01-04:
I guess same topic has been opened on our forum as well https://dietpi.com/phpbb/viewtopic.php?t=9878

MichaIng, 2022-01-04:
Many thanks for your report.

> It stops after a few seconds.

What do you mean by this? Do you see output on the screen which stops to continue?

> Also, the sdcard to which flash is written becomes unreadable by macos

That is expected since it is an `ext4` partition, not natively supported bei macOS or Windows. There is not FAT boot of EFI partition which you may find on other images.

The image _should_ boot from SD card and on Tinker Board S from eMMC as well. The MASKROM jumper is only required on Tinker Board S to force boot from SD card instead of from eMMC, but on non-S this is not required, obviously, since there is no eMMC.

chjan, 2022-01-04:
Thanks for swift reply, this is what happens:

 
The red light is on for some preparing seconds.
Linux startup procedure is seen on the local hdmi terminal 
After less than a page of text, the device turns off, and all goes black, also the red led on the device.
 

I tried several times, both flashing the SD card and starting the device. Notorious behavior.

 

I have a 2A linear PSU (dedicated Raspberry PSU from Audiophonics), and in addition to a 65" TV acting as local terminal, there is also a Logitech wireless keyboard. Not Bluetooth but seen from the device as a wired usb keyboard.

 

Removing either/or TV and keyboard does no change to this sequence, so I assume the device does not turn off due to lack of power? At least, the same PSU can run a raspberry PI3 + HAT DAC without problems.

 

Kind regards

Carl Henrik

Reply to: MichaIng/DietPi ***@***.***>
Date: Tuesday, 4 January 2022 at 14:08
To: MichaIng/DietPi ***@***.***>
Cc: chjan ***@***.***>, Author ***@***.***>

chjan, 2022-01-05:
I now tried the ASUS TinkerOs. Same thing happened, so perhaps there is nothing wrong with your build, but this TinkerBoard? Can you recommend a forum for asking Tinker board questions?

MichaIng, 2022-01-12:
Official forum should be: https://tinker-board.asus.com/forum/

MichaIng, 2022-02-20:
Were you successful at the ASUS forum?

chjan, 2022-02-21:
I have not tried yet, but I will. (I am working on RPI in various configurations at the moment). Thanks for the concern!

MichaIng, 2022-04-03:
As of general issue cleanup, I'll mark this as closed. But feel free to reopen when you got further news/time to debug.

closed 2022-04-03 by MichaIng