MichaIng/DietPi #5137: Open Beta v8.0 | Please help testing and hardening the upcoming release

MichaIng, 2022-01-04:
RC version | v8.0.0
---------- | -------
v8.0.0 PR | #5101
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-01-08**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.0
____
### Beta v8.0.0
_(2022-01-04)_

#### Stretch support
- Support for Debian Stretch has now been removed, which allowed us to do a major code cleanup and migrate to some newer methods in different areas which are only available on Buster and above.

#### Changes
- DietPi-Dashboard | Added an option to view multiple dashboard nodes from one frontend web interface. Related to this, the backend can now be installed only, reducing memory usage and making it impossible to manually access the node outside of the backend API.
- DietPi-Print_large | This new script has been added which can be executed or sourced from /boot/dietpi/func/dietpi-print_large to print the string passed via first argument in large figlet style fonts. It currently only supports the characters a-z, A-Z, 0-9, dot and dash, i.e. those commonly allowed in hostnames.
- DietPi-Banner | Added an option to print the system's hostname in large figlet style fonts, right below the banner header. In case it is enabled as well, the regular/small hostname line will then be skipped. Many thanks to @matellis for implementing this feature: https://github.com/MichaIng/DietPi/pull/5113
- DietPi-Software | Mono: "mono-complete" won't be installed anymore but only "mono-devel" instead. This skips the XSP4 webserver service, which listens on port 8084 by default where it conflicts with File Browser. This only affects new instals. You can manually apply this change via "apt-mark manual mono-devel && apt --autoremove purge mono-complete". Many thanks to @jaguar489 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5093
- DietPi-Software | FuguHub: The outdated official installer has been replaced with a fully automated own setup, removing the obsolete or even harmful interactive dialogues. On fresh installs, an admin account "dietpi" with global software password is created as well.
- DietPi-Software | myMPD: Installation is now done via official APT repository, which means quicker install compared to compiling from source, fewer dependencies and easier updates via "apt upgrade": https://github.com/MichaIng/DietPi/issues/5115
- DietPi-Software | Airsonic: Since the project has been archived by its maintainer and suffers from the critical Log4Shell security vulnerability, we moved to the advertised fork "Airsonic-Advanced" which also enables compatibility with Java 17 and hence Debian Bullseye. Airsonic will be reinstalled as part of the DietPi update to apply the migration. All data and settings will be preserved and migrated automatically.

#### Fixes
- Raspberry Pi | Resolved an issue where unintentionally the turbo mode was enabled. This got backported to v7.9, to our existing RPi images and via live patch. Many thanks to @ayo-x and @whyisthisbroken for reporting this issue: https://github.com/MichaIng/DietPi/issues/5088
- DietPi-Config | Resolved an issue where on Raspberry Pi, when the full KMS display driver is used, the sound card selection could have been falsely applied. With full KMS enabled, for each HDMI port an additional HDMI sound card appears (replacing the regular firmware HDMI sound device), even when no HDMI device is attached and with onboard audio disabled. This lead to an incrementation of external sound card indices. The appearance of the KMS HDMI sound cards is now aligned with the onboard firmware HDMI sound cards, i.e. if not onboard "auto" or HDMI is selected in dietpi-config, now the KMS HDMI sound devices are disabled as well, making use of the "noaudio" device tree overlay parameter.
- DietPi-Software | Resolved an issue where on first boot, choice and preference menu selections triggered an error as the install state file did not exist yet. Many thanks to @bsheeres for reporting this issue: https://github.com/MichaIng/DietPi/issues/5080
- DietPi-Software | Resolved an issue where the deprecated AUTO_SETUP_FILE_SERVER_INDEX dietpi.txt option was applied inverted. "-1" should have installed ProFTPD and "-2" Samba, while it was done the opposite way. However, note that this setting is not present on current images and deprecated. Use AUTO_SETUP_INSTALL_SOFTWARE_ID instead to mark any one or multiple file servers for automatic install. Many thanks to @bsheeres for reporting this issue: https://github.com/MichaIng/DietPi/issues/5081
- DietPi-Software | Browsers: On the ARMv6 Raspberry Pi models Raspberry Pi 1 and Zero (1) since Bullseye, Chromium and Firefox fail to start due to hardware and build limitations which are outside of our control. Since those are currently the only two browser available from dietpi-software, they have been disabled together with the browser preference menu for those systems, until we find and implement a good additional compatible browser. Further info: https://github.com/RPi-Distro/chromium-browser/issues/21
- DietPi-Software | Kodi: Resolved an issue where subsequent Kodi reinstalls created duplicates of the CMA parameter for the KMS overlay setting in config.txt. The duplicates are patched out during DietPi update.
- DietPi-Software | Docker: Resolved an issue where the install was aborted due to an incorrect check for missing kernel modules. Many thanks to @dragonandy for reporting this issue: https://github.com/MichaIng/DietPi/issues/5061
- DietPi-Software | Pi-hole: Resolved an issue where the teleporter import log was not shown but denied by the X-Frame-Options header: https://discourse.pi-hole.net/t/unable-to-restore-teleporter-backup-fresh-install-no-funky-changes-made/51573
- DietPi-Software | Blynk Server: Resolved an issue on ARMv6 RPi models where the install failed as the latest Blynk Server release does not contain a Java 8 build. Since the older Java 8 build does not contain a native Log4Shell vulnerability mitigation, it is added server-wise on these systems.
- DietPi-Software | Mycroft AI: Resolved an issue on Bullseye (and above) systems where "mycroft-cli-client" command fails with a permissions issue, even as root user. Many thanks to @berndverhofstadt for reporting this issue: https://github.com/MichaIng/DietPi/issues/5100
- DietPi-Software | Nukkit: Resolved an issue where the install failed due to a changed download URL. The Jenkins instance has moved to ci.opencollab.dev, where also the Geyser and Floodgate projects are hosted.
- DietPi-Software | FuguHub: Resolved an issue where the uninstall failed as the process was not stopped as intended. Many thanks to @kd9352 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5058
- DietPi-Software | myMPD: Resolved an issue where the installation failed due to an updated dependency. Many thanks to @supersexy for reporting this issue: https://github.com/MichaIng/DietPi/issues/5115
- DietPi-Software | Python 3: Worked around an issue where the pip reinstall failed. Many thanks to @hueppinr for reporting this issue: https://github.com/MichaIng/DietPi/issues/5117
- DietPi-Software | Unbound: Resolved an issue where the "unbound-resolvconf" service applied localhost automatically as local nameserver, if the "resolvconf" package was installed. Since often Unbound is installed in combination with Pi-hole or AdGuard Home, and those are usually used by network clients only, not the server itself, "unbound-resolvconf" is disabled now on Unbound installs. If Unbound is wanted as local resolver as well, it should be explicitly configured, e.g. via dietpi-config network options. Many thanks to @Ianszh for reporting this issue: https://github.com/MichaIng/DietPi/issues/5133

billouetaudrey, 2022-01-05:
Hello, install done, will tell if any bug.

For now, this is the bug when I install it : 
[FAILED] DietPi-Survey | Purging survey data
 - Command: curl --key  -m 20 -sT 80017429-532b-437d-9b75-256a91d206ec.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/

MichaIng, 2022-01-05:
Can you try to run this manually and past the output here?
```sh
cd /tmp
> 80017429-532b-437d-9b75-256a91d206ec.txt
curl --key  -m 20 -vT 80017429-532b-437d-9b75-256a91d206ec.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/
```

billouetaudrey, 2022-01-05:
Seem stuck here : 

`root@raspberrypi:~# cd /tmp
> 80017429-532b-437d-9b75-256a91d206ec.txt
curl --key  -m 20 -vT 80017429-532b-437d-9b75-256a91d206ec.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/
Warning: The file name argument '-m' looks like a flag.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 0.0.0.20:80...
  0     0    0     0    0     0      0      0 --:--:--  0:00:16 --:--:--     0
`

MichaIng, 2022-01-05:
Ah sorry, I forgot the empty `--key` argument:
```sh
cd /tmp
> 80017429-532b-437d-9b75-256a91d206ec.txt
curl --key '' -m 20 -vT 80017429-532b-437d-9b75-256a91d206ec.txt 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/'
```

billouetaudrey, 2022-01-05:
* Closing connection 0
curl: (60) SSL peer certificate or SSH remote key was not OK
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

MichaIng, 2022-01-05:
Is our host key still present?
```sh
grep 'ssh.dietpi.com' /root/.ssh/known_hosts
```

billouetaudrey, 2022-01-05:
Nope :( 

`root@raspberrypi:/tmp# grep 'ssh.dietpi.com' /root/.ssh/known_hosts
grep: /root/.ssh/known_hosts: Aucun fichier ou dossier de ce type
root@raspberrypi:/tmp#
`

MichaIng, 2022-01-05:
Recreate it with this command:
```sh
echo 'ssh.dietpi.com ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' > /root/.ssh/known_hosts
```

Joulinar, 2022-01-05:
@MichaIng 
we had already a couple of cases where the ssh-rsa key was missing. Should we include some error handle within the survey function?

MichaIng, 2022-01-05:
Or we re-create it with a patch on next update? You a check for `grep '^ssh.dietpi.com' /root/.ssh/known_hosts` within survey and bug report scripts is actually a good idea as well, so the update is not attempted in the first place when doomed to fail but a meaningful error message shown instead.

Joulinar, 2022-01-06:
question is it missing in some cases? By accident or by purpose?

MichaIng, 2022-01-06:
Another reason to go with the check + error message, so users who intentionally remove it get an info for what it is required.

bsheeres, 2022-01-06:
I have a RPi 3b+ and a Zero W. Is there any value in running installs on both using all the ARM versions, both fresh and upgrade? If so and apart from opting-in, is there any specific log to return?

Maybe another way ask is: Is there a checklist or guide to follow for testing or is it all just sorta fuzzy?

billouetaudrey, 2022-01-06:
> 

Work now, strange no? I never delete this file... 

Question : Do you have any Telegram Group ? :)

Joulinar, 2022-01-06:
Nope we don't have such a group. There is a small user community on Facebook.

billouetaudrey, 2022-01-06:
> Nope we don't have such a group. There is a small user community on Facebook.

Ok ;)

sheeepdev, 2022-01-06:
> Nope we don't have such a group. There is a small user community on Facebook.

Maybe you could make a discord server or such?

tandy-1000, 2022-01-06:
> > Nope we don't have such a group. There is a small user community on Facebook.
> 
> Maybe you could make a discord server or such?

No thank you, Matrix please, no proprietary services for open source projects :/

sheeepdev, 2022-01-06:
> > > Nope we don't have such a group. There is a small user community on Facebook.
> > 
> > 
> > Maybe you could make a discord server or such?
> 
> No thank you, Matrix please, no proprietary services for open source projects :/

Yeah, I know. Discord is poprietary, but still, [a lot of open source projects use it](https://discord.com/open-source).
I don't like Matrix, its very clunky and not user-friendly at all.

MichaIng, 2022-01-06:
I haven't used it much, but there are a lot of Matrix clients, the benefit of an open protocol. So I guess there will be one which suites your expectations, or is it somehow a limitation of the protocol itself compared to Discord?

Generally an additional platform is not always better, it splits reports and solutions for common problems, making it more difficult for us and for users to quickly find or link solutions or contribute to discussions/issues. And it requires us to watch and react on an additional platform. I have a Discord account, but am, to be true, not very reactive there, as forum and in my case especially GitHub keeps me busy. We try to be very reactive on both platforms, so not sure whether a chat platforms adds so much value. However, we were playing with the idea to host an own Synapse Matrix server, being the obvious solution as our server runs DietPi (of course) and we have Synapse integrated into `dietpi-software` 😉.

sheeepdev, 2022-01-06:
> I haven't used it much, but there are a lot of Matrix clients, the benefit of an open protocol. So I guess there will be one which suites your expectations, or is it somehow a limitation of the protocol itself compared to Discord?
> 
> Generally an additional platform is not always better, it splits reports and solutions for common problems, making it more difficult for us and for users to quickly find or link solutions or contribute to discussions/issues. And it requires us to watch and react on an additional platform. I have a Discord account, but am, to be true, not very reactive there, as forum and in my case especially GitHub keeps me busy. We try to be very reactive on both platforms, so not sure whether a chat platforms adds so much value. However, we were playing with the idea to host an own Synapse Matrix server, being the obvious solution as our server runs DietPi (of course) and we have Synapse integrated into `dietpi-software` wink.

Pretty much all of the matrix clients are the same, just the look is different. It is confusing because of all the codes, rooms, etc. I find discord more user-friendly, you just join the server and pick a channel to talk. In matrix you need to pick a homeserver, then make your account, then join the room (or whatever, I don't know) then join the channel and finally you can chat with others.

It just that to a normal user who wants to chat about DietPI Matrix is too complicated. With discord, you just register and join the server and all the information, channels and etc. are there.

But I agree that Matrix is more private, open and secure. I'm not hating on matrix, it is just too complex for someone who just wants to chat.

Joulinar, 2022-01-06:
As we are 2 people only who do user support, there is quite a good chance that there is nobody you could chat with. Doesn't matter if Matrix or Discord. 😉

tandy-1000, 2022-01-06:
> 

> > I haven't used it much, but there are a lot of Matrix clients, the benefit of an open protocol. So I guess there will be one which suites your expectations, or is it somehow a limitation of the protocol itself compared to Discord?
> > Generally an additional platform is not always better, it splits reports and solutions for common problems, making it more difficult for us and for users to quickly find or link solutions or contribute to discussions/issues. And it requires us to watch and react on an additional platform. I have a Discord account, but am, to be true, not very reactive there, as forum and in my case especially GitHub keeps me busy. We try to be very reactive on both platforms, so not sure whether a chat platforms adds so much value. However, we were playing with the idea to host an own Synapse Matrix server, being the obvious solution as our server runs DietPi (of course) and we have Synapse integrated into `dietpi-software` wink.
> 
> Pretty much all of the matrix clients are the same, just the look is different. It is confusing because of all the codes, rooms, etc. I find discord more user-friendly, you just join the server and pick a channel to talk. In matrix you need to pick a homeserver, then make your account, then join the room (or whatever, I don't know) then join the channel and finally you can chat with others. Also, installing matrix and connecting and verifying your session is also a trouble. You would need to save the codes somewhere, which I can do but then I forget where I stored them and I need to re-do it all again.
> 
> It just that to a normal user who wants to chat about DietPI Matrix is too complicated. With discord, you just register and join the server and all the information, channels and etc. are there.
> 
> But I agree that Matrix is more private, open and secure. I'm not hating on matrix, it is just too complex for someone who just wants to chat.

you can just use the default matrix.org homeserver, so its about as simple..

> As we are 2 people only who do user support, there is quite a good chance that there is nobody you could chat with. Doesn't matter if Matrix or Discord. wink

I agree with this though, any chat platform will add to the burden of maintenance and moderation, the forum / github is probably fine enough until there are people ready to moderate it.

elitearmedforce, 2022-01-06:
Another major version of dietpi, hurray. Thank you.
Luckily had no issue with this beta update. Jump from v7.4 to v8.0. RPi4 - Home media server

MichaIng, 2022-01-06:
For those who are interested, why a major version increment: We decided to do this whenever support for an older Debian version (Stretch in this case) or older DietPi versions (DietPi prior to v6.14 in this case) are dropped, hence follow the intended meaning of major versions in semantic versioning to do a breaking change. Other than that, which allowed us to drop and enhance a lot of code to keep up Stretch compatibility, there is not further meaning of it.

Old DietPi systems however only need to call `dietpi-update` two times to get to v8.0, first migration to the `pre-v6.14` branch, updating to v7.9 is done, second from v7.9 the update to v8.0 from `master` branch is offered 😉. Debian Stretch systems can be quite painlessly upgraded to Buster: https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

m0jek, 2022-01-07:
> Nope we don't have such a group. There is a small user community on Facebook.

We do need something for those that don't use Meta / Facebook. My vote will be Telegram or Discord or Matrix. What is useful with Matrix is that there are bridges for Telegram and Discord ...

MichaIng, 2022-01-07:
> We do need something for those that don't use Meta / Facebook.

GitHub and forum.

> What is useful with Matrix is that there are bridges for Telegram and Discord ...

That is actually very nice, indeed. How does it work, is this something that can/needs to be setup on the Matrix server with a Telegram/Discord account which then exposes/syncs Matrix chats on Telegram/Discord automatically?

billouetaudrey, 2022-01-07:
I think I started a discussion about Telegram etc... lol

I know with Telegram ( certainly with Discord too ) you can sync Github changed

tandy-1000, 2022-01-07:
> That is actually very nice, indeed. How does it work, is this something that can/needs to be setup on the Matrix server with a Telegram/Discord account which then exposes/syncs Matrix chats on Telegram/Discord automatically?

you can selfhost the bridge bot, or use t2bot.io which hosts telegram and discord bots

MichaIng, 2022-01-07:
#### Beta v8.0.1 has been merged with a few changes and fixes: #5141

Mausy5043, 2022-01-08:
My 2 cents regarding hardening: you might want to consider disabling SSH-logins by `root`. The `dietpi` user has `sudo` capabities so `root` really shouldn't be exposed.

MichaIng, 2022-01-08:
`root` login is currently required for first run setup, especially when `AUTO_SETUP_AUTOMATED=1` has been set. Since `dietpi` has full passwordless `sudo` permissions, both imply exactly the same security risk, with the only different that `sudo` calls are generally logged.

Furthermore we plan to remove the `dietpi` user in the future and instead add an option to create a new/custom login user during first run setup. When doing so, we can additionally offer to have `root` SSH logins disabled, or lock any `root` logins completely (`usermod -L root`), given that the new user has `sudo` permissions, of course.

But I agree in any case that hardening SSH logins (disabling root login and especially also enabling key authentication) needs to be more easy/transparent to users. While it can be easily done manually, I guess many don't know about this and leave root password login enabled. For this release it is of course too late, but probably you can open an issue so we can track for next release.

MichaIng, 2022-01-08:
#### Beta v8.0.2 has been merged: #5147
We'll do a last code review and in case some quick tests, before doing the DietPi v8.0 release until midnight.

MichaIng, 2022-01-08:
#### DietPi v8.0 has been released, many thanks to all testers ❤️!

closed 2022-01-08 by MichaIng

elitearmedforce, 2022-01-09:
Congrats again on major release. Just a question. Wasn't sure should I be making another issue for a question. I am planning to have a clean install for Bullseye upgrade.
Question:
1) Is it right time to switch 64bit/v8 or stay with 32bit/v7? (I went through [rpi forum](https://forums.raspberrypi.com/viewtopic.php?t=323303) for kodi, noticed it has support hardware decoding for 64bit)

Details:
- SBC: RPi4 B
- RAM: 4 GB
- SD: 32GB
- Purpose: Home media server
- Software used: Kodi, Emby, Sonarr, Radarr, Deluge, qbittorrent. 
*****Hardware de/encoding is a must

MichaIng, 2022-01-09:
> Congrats again on major release.

Many thanks 🙂.

> 1) Is it right time to switch 64bit/v8 or stay with 32bit/v7? (I went through [rpi forum](https://forums.raspberrypi.com/viewtopic.php?t=323303) for kodi, noticed it has support hardware decoding for 64bit)

Yes you can do that, indeed since Bullseye there is an `arm64` build from the RPi repository with hardware acceleration available. It uses the KMS display driver via GBM and video de/encoding is done via FFmpeg and V4L2 libraries. The same is however true for the 32-bit/`armhf` Kodi package, so the 64-bit image won't give you much benefit as long as you do not need it for other software, or run such software which makes vast use of 64-bit integers. The ARMv7 image should be however preferred over the ARMv6 image in every case where you do not plan to use the same image on ARMv6-only RPi models (RPi 1/Zero).

elitearmedforce, 2022-01-10:
> The same is however true for the 32-bit/`armhf` Kodi package, so the 64-bit image won't give you much benefit as long as you do not need it for other software, or run such software which makes vast use of 64-bit integers. 

Even as per my research on it, didn't find any mentions of huge differences other than small performance differences and package issues. So I think I will stick with 32bit for now.

MichaIng, 2022-01-10:
A smaller number of software titles is not available for 64-bit ARM, a small number not for 32-bit ARM, it may have reached quality. But also from performance point of view, 64-bit is faster only in certain cases, as mentioned when software code deals with 64-bit size integers often, while it consumes usually notably more disk and RAM space, which has again a negative impact on performance. So I'd say one cannot give a generic preference (for RPi 4 where RAM size is not inherently a problem) for either the one or the other, but more important is whether all aimed software titles are supported 😉.

Reminds me that we need to add supported architectures to e.g. the software list in our Wiki, so one can check this before flashing the image.

MichaIng, 2022-01-11:
Jep, on RPi using an unprivileged user for Kodi (via service) should be pretty easy, using `video`, `render` and `audio` groups.