MichaIng/DietPi #5140: Mosquitto fails to install on Bullseye.

thomasmockridge, 2022-01-05:
#### Details:
- Date           | Tue Jan  4 20:57:25 SAST 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | `Linux pi 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install mosquitto`
- Exit code      | 100
- Software title | DietPi-Software
- Power supply used | (EG: 5V 3.5Amp)
- 
- Bug report ID | 2ba09c0d-c9e4-4d04-afac-433f5ef396a4
 
#### Steps to reproduce:

1. Fresh install of DietPi on Pi Zero
2. Attempt to install Mosquitto (Was trying to install Home Assistant but this is the failing software dependency)
#### Expected behaviour:

- Mosquitto should install and run.
#### Actual behaviour:

- Mosquitto installs but fails to run. See command session below, full details lower down:
- root@pi:~# /usr/sbin/mosquitto
**Illegal instruction**

#### Extra details:

- After initial install failed I tried to disable the dietpi repo and re-ran:

root@pi:~# apt purge mosquitto mosquitto-clients
root@pi:~# vi /etc/apt/sources.list.d/dietpi-mosquitto.list  # comment out dietpi repo
root@pi:~# apt update
root@pi:~# apt autoremove
# removes libmosquitto1 <- I think this was my problem trying to install older mosquitto...
root@pi:~# apt --fix-broken install mosquitto
### WORKED!! ###

#### Additional logs:
```Original fail:
Selecting previously unselected package mosquitto.
(Reading database ... 27350 files and directories currently installed.)
Preparing to unpack .../mosquitto_2.0.12-0mosquitto1~bullseye1_armhf.deb ...
Unpacking mosquitto (2.0.12-0mosquitto1~bullseye1) ...
Setting up mosquitto (2.0.12-0mosquitto1~bullseye1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/mosquitto.service → /lib/systemd/system/mosquitto.service.
Job for mosquitto.service failed because a fatal signal was delivered to the control process.
See "systemctl status mosquitto.service" and "journalctl -xe" for details.
invoke-rc.d: initscript mosquitto, action "start" failed.
● mosquitto.service - Mosquitto MQTT Broker
     Loaded: loaded (/lib/systemd/system/mosquitto.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Tue 2022-01-04 20:57:24 SAST; 156ms ago
       Docs: man:mosquitto.conf(5)
             man:mosquitto(8)
    Process: 23290 ExecStartPre=/bin/mkdir -m 740 -p /var/log/mosquitto (code=exited, status=0/SUCCESS)
    Process: 23291 ExecStartPre=/bin/chown mosquitto /var/log/mosquitto (code=exited, status=0/SUCCESS)
    Process: 23292 ExecStartPre=/bin/mkdir -m 740 -p /run/mosquitto (code=exited, status=0/SUCCESS)
    Process: 23293 ExecStartPre=/bin/chown mosquitto /run/mosquitto (code=exited, status=0/SUCCESS)
    Process: 23294 ExecStart=/usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf (code=killed, signal=ILL)
   Main PID: 23294 (code=killed, signal=ILL)
        CPU: 151ms

Jan 04 20:57:25 pi systemd[1]: mosquitto.service: Scheduled restart job, restart counter is at 1.
Jan 04 20:57:25 pi systemd[1]: Stopped Mosquitto MQTT Broker.
Jan 04 20:57:25 pi systemd[1]: Starting Mosquitto MQTT Broker...
dpkg: error processing package mosquitto (--configure):
 installed mosquitto package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 mosquitto
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

thomasmockridge, 2022-01-05:
Sorry about last comment it did not work ie same error - just the apt install reported working... 
root@pi:~# /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf
Illegal instruction
root@pi:~#

Joulinar, 2022-01-05:
Hi, 

This is an issue of mosquitto, not being able to be installed on ARMv6 systems. Latest version working is mosquitto 2.0.12

https://github.com/eclipse/mosquitto/issues/2384

thomasmockridge, 2022-01-05:
Thank you very much,  it looks like the downgrade can still work on bullseye, here are my steps to fix:

# Add repository
echo  "deb https://repo.mosquitto.org/debian buster main" > /etc/apt/sources.list.d/mosquitto-buster.list
# Read repository
apt update
# Clear out any existing mosquitto
apt purge "mosquitto*" "libmosquitto*"
apt autoremove
# Install "old" mosquitto
apt-get install mosquitto=2.0.11-0mosquitto1~buster1  mosquitto-clients=2.0.12-0mosquitto1~buster1 libmosquitto1=2.0.12-0mosquitto1~buster1
# Get apt update to ignore mosquitto
apt-mark hold "mosquitto*"
   mosquitto set on hold.
   mosquitto-clients set on hold.
   mosquitto-dev set on hold.
   mosquitto-clients-dbgsym set on hold.
   mosquitto-dbgsym set on hold._

# allow external mqtt connections (and anonymous if you want)

echo "listener 1883
allow_anonymous true" >> /etc/mosquitto/mosquitto.conf

Now to get back to original homeassistant install :-)

thomasmockridge, 2022-01-05:
Thanks for the prompt support - I have Homeassistant and mosquitto working... thank you again

MichaIng, 2022-01-05:
Let me keep this open to track the upstream issue and in case apply the version freeze as workaround to `dietpi-software`.

MichaIng, 2022-01-07:
Do we add the freeze as workaround or do we hope that it gets resolved soon? What makes the workaround a bit complicated (though not too complicated) is that the version contains the repository, e.g. on Bullseye (where it fails as well): `2.0.12-0mosquitto1~bullseye1`

Joulinar, 2022-01-07:
puhh I don't belive it will be fixed upstream soon. At least there is no reaction on the issue by Mosquitto devs. I guess we would need to go for simplest workaround we could find.

MichaIng, 2022-01-07:
A simple workaround would be to install from Raspbian, but this means ancient version `1.5.7` on Buster. At least we can also easily apply this to existing ARMv6 systems by removing the Mosquitto repository via pre-patch.

Joulinar, 2022-01-08:
not sure if this fit's on the issue but I found 2 more software title not working on RPi1 ARMv6 anymore

**Octoprint** 

```
root@DietPi1:~# sudo -u octoprint /mnt/dietpi_userdata/octoprint/.local/bin/octoprint serve
2022-01-08 01:31:04,623 - octoprint.startup - INFO - ******************************************************************************
2022-01-08 01:31:04,637 - octoprint.startup - INFO - Starting OctoPrint 1.7.2
2022-01-08 01:31:04,648 - octoprint.startup - INFO - ******************************************************************************
2022-01-08 01:31:07,767 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-01-08 01:31:07,770 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-01-08 01:31:07,772 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-01-08 01:31:10,703 - octoprint.plugin.core - INFO - Loading plugins from /mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/octoprint/plugins, /mnt/dietpi_userdata/octoprint/.octoprint/plugins and installed plugin packages...
2022-01-08 01:31:20,614 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 38 hook handlers
Illegal instruction
root@DietPi1:~#
```

**docker-compose v1**

```
root@DietPi1:~# docker-compose version
Illegal instruction
```

For `docker-compose` it's easy to fix as we will move to v2 anyway on DietPi v8.1

```
root@DietPi1:~# curl -fL 'https://github.com/docker/compose/releases/download/v2.2.3/docker-compose-linux-armv6' -o /usr/local/bin/docker-compose
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   663  100   663    0     0   1161      0 --:--:-- --:--:-- --:--:--  1165
100 20.7M  100 20.7M    0     0  2255k      0  0:00:09  0:00:09 --:--:-- 2699k
root@DietPi1:~# chmod +x /usr/local/bin/docker-compose
root@DietPi1:~# docker-compose version
Docker Compose version v2.2.3
root@DietPi1:~#
```

MichaIng, 2022-01-08:
Looks like OctoPrint loads an ARMv7-only plugin/wheel, I mean Python scripts themselves cannot be incompatible, but the compiled wheels can.

In case of Docker Compose I'm completely puzzled as it's pulled from piwheels.org directly and hasn't changed since May last year: https://piwheels.org/project/docker-compose/
I do not believe that piwheels.org accidentally build ARMv7-only wheels. However, indeed not worth to dig deeper due to planned migration, and to be true, running a Docker container on an RPi1/Zero won't be much joy anyway, especially thinking about the additional memory consumption for the Docker engine.

MichaIng, 2022-01-08:
Applied the workaround now: https://github.com/MichaIng/DietPi/commit/bb36da2
- On ARMv6 Mosquitto is now installed from Raspbian repository, and the Mosquitto repo is as well removed via pre-patch.

closed 2022-01-08 by MichaIng

Joulinar, 2022-01-09:
@MichaIng 
for OctoPrint indeed it's one or more of the pre-compiled wheels from `piwheels.org`. I did a test, installed OctoPrint without using `piwheels.org` and installation was successful as all wheels compiled on the RPI1 directly. Question is which wheel it is 🤔

MichaIng, 2022-01-09:
Can you open a new issue about those? I guess then indeed `piwheels.org` accidentally compiled some wheels for ARMv7 only, including Docker Compose and some dependency of OctoPrint. We should quickly report that upstream.

Joulinar, 2022-01-10:
> Can you open a new issue about those?

Will do so