MichaIng/DietPi #5143: Slow DHCP server causing trouble with auto installation (with workaround)

jpeg2600, 2022-01-06:
### Creating a bug report/issue

#### Required Information
- DietPi version: 7.9.3
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model |  Asus Tinker Board S 
- Power supply used | well enough
- SDcard used | onboard eMMC 

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- Smooth installation & selected package installation and postinstall script running

#### Actual behaviour

- Crashes after a long DHCP Discover wait, eventually works when an IP is acquired

#### Extra details

- Tried on two different routers/DHCP servers and was still getting the same problem. Sometimes it worked without the workaround. 

WORKAROUND: 
I use this in my dietpi.txt file: 
  CONFIG_CHECK_CONNECTION_IP=127.0.0.1
  CONFIG_CHECK_DNS_DOMAIN=localhost

The NTP step is right after so it hangs a bit but eventually works when the DHCP process is done in the background.

Joulinar, 2022-01-06:
Did you tried using STATIC IP instead of DHCP? 
Setting connection as well as DNS check to localhost is not really a workaround. You simply bypass the network connection check. Which exactly should detect such network issue.

MichaIng, 2022-01-06:
Many thanks for your report.

Interesting to know would be whether `network-online.target` has been reached, despite DHCP hasn't finished yet. Is it still the first boot session or are you able to replicate and check:
```sh
journalctl -u network-online.target -u ifup@eth0 -u systemd-timesyncd
```
The timestamps and output of these units should indicate whether `systemd-timesyncd` started before or after `network-online.target` and whether `network-online.target` was reached before or after DHCP reported a successful lease.

A few versions ago we changed waiting for network at boot so that only `dietpi-postboot.service` (which does the DietPi update check and service starts) waits for `network-online.target`, while logins are generally possible earlier. In case `network-online.target` indeed wasn't reached yet, we could delay, in case of `AUTO_SETUP_AUTOMATED=1` set, the automatic login via `getty@tty1` as well to wait for `network-online.target`. We already alter the unit in that case for the first boot to perform the autologin, so adding the wait for network would be trivial.

MichaIng, 2022-01-07:
Okay `ifup@.service` seems to wait for DHCP to fully finish, works very well here for `dietpi-postboot.service`:
```
Jan 07 13:52:23 DietPi dhclient[334]: DHCPACK of 192.168.1.32 from 192.168.1.1
Jan 07 13:52:23 DietPi ifup[334]: DHCPACK of 192.168.1.32 from 192.168.1.1
Jan 07 13:52:23 DietPi dhclient[334]: bound to 192.168.1.32 -- renewal in 583045 seconds.
Jan 07 13:52:23 DietPi ifup[334]: bound to 192.168.1.32 -- renewal in 583045 seconds.
Jan 07 13:52:24 DietPi systemd[1]: Finished ifup for wlan0.
Jan 07 13:52:24 DietPi systemd[1]: Reached target Network is Online.
Jan 07 13:52:24 DietPi systemd[1]: Started DietPi-PostBoot.
```
So I'll add the `After=network-online.target` for the login prompt (including autologin) to as well. I have a feeling this may solve #4657/#4651 as well @skavinvarnan for reference.

MichaIng, 2022-01-07:
Done: https://github.com/MichaIng/DietPi/commit/b762a64

This change won't have an effect on current images as the DietPi update is done after this script is called. If you can mount/access the `ext4` filesystem before booting, the `/var/lib/dietpi/services/dietpi-firstboot.bash` needs to be replaced with: https://raw.githubusercontent.com/MichaIng/DietPi/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash

jpeg2600, 2022-01-07:
Hey thanks @MichaIng ! Sorry for not reporting back sooner. Feel free to poke me if you want me to do some testing. Cool project, I'm looking forward contributing more now that I will be using it in some IoT solutions.

jpeg2600, 2022-01-07:
I will try your script and report back. Hopefully this weekend.

MichaIng, 2022-01-07:
Yes it would be great if you could try it with the patch/change applied before booting the image. If you cannot mount/edit `ext4` from another system, I can patch out ASUS Tinker Board image.

Don't read below if you are not interested in deep depth 😄:

I tried to test the difference on my notebook with WiFi and DHCP, but after several attempts forth and back, where it always, even with a long sleep in the network service, waited for network to finish, I realised that there was another user session service/target which was ordered after network and before the login getty, forcefully delaying the autologin after network. Probably it is new on Debian Bookworm (which the notebook runs), so tomorrow I hopefully can verify the difference on a cleaner Buster or Bullseye system. But what I recognised, is that a `Wants=`, hence adding a new service to the boot sequence, is not possible for the running boot sequence. Since `dietpi-postboot.service` pulls `network-online.target` in, that is not a problem, but I'm not sure whether the same applies for `Before=`/`After=` ordering. Hence another test makes sense. Else we can apply this via DietPi-PREP for all images in the first place. This means that also non-automated first run setups will have the first login delayed after network, but this isn't a big problem since of course also for the manual setup, network is required, you just have the ability to manually repeat etc.

jpeg2600, 2022-01-08:
So I just tried with the new bash script and still got an error. It sent out two DISCOVER then says "No DHCPOFFERS". I can try changing the timeout value in /etc/dhcp/dhclient.conf before the first boot and see if it fixes it. I'm under the impression that it's the DHCP client reporting an error too soon. 

My system is on at its first boot right now, I can get some logs. 

Here's what I get from `journalctl -u network-online.target -u ifup@eth0 -u systemd-timesyncd

````
root@AgriM-Test:/var/log# journalctl -u network-online.target -u ifup@eth0 -u systemd-timesyncd
-- Journal begins at Wed 2021-12-15 12:26:32 EST, ends at Fri 2022-01-07 20:25:32 EST. --
Dec 15 12:27:08 AgriM-Test systemd[1]: Reached target Network is Online.
Dec 15 12:27:21 AgriM-Test systemd[1]: Starting Network Time Synchronization...
Dec 15 12:27:21 AgriM-Test systemd[1]: Started Network Time Synchronization.
Jan 07 19:36:58 AgriM-Test systemd-timesyncd[1120]: Initial synchronization to time server 68.69.221.61:123 (0.debian.pool.ntp.org).
Jan 07 19:36:59 AgriM-Test systemd[1]: Stopping Network Time Synchronization...
Jan 07 19:36:59 AgriM-Test systemd[1]: systemd-timesyncd.service: Succeeded.
Jan 07 19:36:59 AgriM-Test systemd[1]: Stopped Network Time Synchronization.
```

Also, it would be useful not to automatically disable the ethernet connection when enabling wifi. In my use case, I use WIFI connectivity to access the SBCs but they are connected to a PLC using ethernet and a static IP.

MichaIng, 2022-01-08:
> I'm under the impression that it's the DHCP client reporting an error too soon.

Ah, probably `ifup@.service` returns in case of an error, including timeout. Indeed we reduce the default DHCP timeout values to 10 seconds timeout and 4 seconds waiting time before next attempt. This has been done since ancient time and I simply never touched this, but probably it's time to skip it and trust Debian/ISC to have a wise default applied already: https://github.com/MichaIng/DietPi/blob/801253f/PREP_SYSTEM_FOR_DIETPI.sh#L1452-L1454
The default is 60/60, so the DHCP client waits 60 seconds for an answer and in case waits for 60 seconds before retrying after a timeout. After a timeout, btw, when there are non-expired leases stored already, they will be used, so on reboot a short timeout can lead to a quicker connectivity, just to mention.

Can you put the autologin into the view, in case you are willing to do another test:
```
journalctl -u network-online.target -u ifup@$(G_GET_NET iface) -u systemd-timesyncd -u getty@tty1
```
As I do not see any interface configuration, `eth0` is probably not the right interface you use for internet connection? I replaced it with a function call which derives the main interface via default gateway 😉.

I will now try to get a definite answer whether reordering of loaded systemd units during the boot sequence is generally possible or not.

MichaIng, 2022-01-08:
Okay made some tests:
1. Before/After ordering can be done effectively for the current boot sequence, of course when the target units did not run already. This even works without reloading systems, though in case of `dietpi-firstboot.service` it is implied when it `systemctl disable dietpi-firstboot` itself at the end.
2. Pulling in new units via `Wants=` indeed is not possible, but not required in this case.
3. But indeed practically this ordering is not required, since `getty@.service` is ordered after `systemd-user-sessions.service` already, which is ordered after `network.target`, reached after `ifup@.service`.
4. When the DHCP client times out, `ifup` indeed returns, hence this is the true reason why in your case the autologin was done before DHCP has successfully applied an IP.
5. #711 is still true: `ifup` waits for DHCP answers until timeout, even if no SSID has been configured yet, which is somehow dumb as the WiFi adapter is not connected to anywhere... However, the only case where this place a role is when scanning for SSIDs after WiFi has been enabled the first time (no SSID configured yet). For SSID scanning, the interface must be "up". It is however possible to `iw l set dev wlan0 up` which enables SSID scanning but does not involve any WPA or DHCP steps, making the short DHCP timeout obsolete.

I applied/party reverted the changes according to these results: https://github.com/MichaIng/DietPi/commit/7cedb8e

Let's see whether the raised, e.g. 60 seconds DHCP timeout really solves it for you, but it all makes totally sense now.

closed 2022-01-08 by MichaIng