MichaIng/DietPi #5144: Error when trying to change out ssh from dropbear to openSSH

PintjesBier, 2022-01-06:
Backstory: tried to change out dropbear for openSSH as I keep getting this error when trying to dietpi-update

Jan 06 13:06:48 PiHole1 systemd[1]: Starting OpenBSD Secure Shell server...
Jan 06 13:06:48 PiHole1 sshd[12706]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Jan 06 13:06:48 PiHole1 sshd[12706]: error: Bind to port 22 on :: failed: Address already in use.
Jan 06 13:06:48 PiHole1 sshd[12706]: fatal: Cannot bind any address.
Jan 06 13:06:48 PiHole1 systemd[1]: ssh.service: Main process exited, code=exited, status=255/EXCEPTION
Jan 06 13:06:48 PiHole1 systemd[1]: ssh.service: Failed with result 'exit-code'.
Jan 06 13:06:48 PiHole1 systemd[1]: Failed to start OpenBSD Secure Shell server.

#### Details:
- Date           | Thu Jan  6 13:06:49 CET 2022
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux PiHole1 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Through dietpi-software change SSH client from dropbear to openSSH
2. Select 'install'
#### Expected behaviour:

- The SSH client should be changed out...
#### Actual behaviour:

- Just getting the error mentioned above
#### Extra details:

- ...
#### Additional logs:
```
Setting up openssh-server (1:7.9p1-10+deb10u2+rpt1) ...
rescue-ssh.target is a disabled or a static unit, not starting it.
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.

MichaIng, 2022-01-06:
Many thanks for your report. Dropbear should be stopped before the OpenSSH install starts. If you still have the logs, do you see a:
```
[  OK  ] systemctl stop dropbear
```
Before the OpenSSH package install starts?

And to be sure that it's Dropbear still running:
```sh
ss -tlpn | grep :22
```

PintjesBier, 2022-01-06:
> Many thanks for your report. Dropbear should be stopped before the OpenSSH install starts. If you still have the logs, do you see a:
> 
> ```
> [  OK  ] systemctl stop dropbear
> ```
> 
> Before the OpenSSH package install starts?
> 
> And to be sure that it's Dropbear still running:
> 
> ```shell
> ss -tlpn | grep :22
> ```
> 
> I see you run a very old DietPi version. I had a look into the current `master` branch code but won't check back for v6.34. So generally when you run into an error, first thing you should try is to upgrade DietPi to the latest version.

Thank you for your quick reply! Manually disabling dropbear before installing dropbear worked! So problem has been solved!

Thanks!

closed 2022-01-06 by PintjesBier