MichaIng/DietPi #5145: Home Assistant 404: Not Found Error

theChef613, 2022-01-07:
Hello,
I installed Homeassistant on dietpi v7.9.3 on a RPi 3 Model B (aarch64) using the instructions here: https://dietpi.com/docs/software/home_automation/ 
I waited for the CPU to go down to 0% and ran sudo systemctl restart home-assistant
However, when I go to http://<my.IP>:8123 all I get is a blank webpage with 404: Not Found. Not the web interface.
I already tried reinstalling home assistant both by the reinstall command and uninstalling then reinstalling. any help you can give would be great! thank you!

Joulinar, 2022-01-07:
Hi,

pls can you share following

```
journalctl -u home-assistant
```

MichaIng, 2022-01-07:
And after the service restart, give it time again until CPU usage goes down to ~0%. On first starts, it installs a bunch of Python modules for the web interface, it often fails (for still unknown reason: https://github.com/home-assistant/core/issues/28361) on first attempt, but succeeds (and takes a while) on second attempt. Until the installs are done, the web interface is unreachable. Follow status via `htop` and `ss -tlpn`, the latter showing you whether power `8123` is listened already.

theChef613, 2022-01-07:
> Hi,
> 
> pls can you share following
> 
> ```
> journalctl -u home-assistant
> ```

Hello, 
Thanks for your reply! attached is the log you asked for.
[homeassistant log.txt](https://github.com/MichaIng/DietPi/files/7829557/homeassistant.log.txt)

theChef613, 2022-01-07:
> And after the service restart, give it time again until CPU usage goes down to ~0%. On first starts, it installs a bunch of Python modules for the web interface, it often fails (for still unknown reason: [home-assistant/core#28361](https://github.com/home-assistant/core/issues/28361)) on first attempt, but succeeds (and takes a while) on second attempt. Until the installs are done, the web interface is unreachable. Follow status via `htop` and `ss -tlpn`, the latter showing you whether power `8123` is listened already.

Thanks for the reply! I did the service restart twice, waiting for the htop cpu usage to go to ~0%. I do see port 8123 being listened to which makes sense because the <my.IP>:8123 does take me to a page... just the only thing it says is 404: Not Found

Thanks again!

MichaIng, 2022-01-07:
I just tested on Odroid N2 (`aarch64`) and it works well there. But your error messages also do not look like the ones related to the required service restart:
```
Jan 07 16:44:24 OdroidN2 homeassistant[18332]: 2022-01-07 16:44:24 ERROR (MainThread) [homeassistant.setup] Setup failed for recorder: Unable to import component: No module named 'sqlalchemy'
Jan 07 16:44:26 OdroidN2 homeassistant[18332]: 2022-01-07 16:44:26 ERROR (MainThread) [homeassistant.setup] Error during setup of component http
```
It's these `no module named` ones, happening on any random module for some reason until restart. But in your case it's:
```
recorder: Unable to import component: cannot import name 'event' from partially initialized module 'sqlalchemy'
```
Can you paste your:
```sh
cat /mnt/dietpi_userdata/homeassistant/deps/lib/python3.9/site-packages/sqlalchemy/__init__.py
```

theChef613, 2022-01-07:
> I just tested on Odroid N2 (`aarch64`) and it works well there. But your error messages also do not look like the ones related to the required service restart:
> 
> ```
> Jan 07 16:44:24 OdroidN2 homeassistant[18332]: 2022-01-07 16:44:24 ERROR (MainThread) [homeassistant.setup] Setup failed for recorder: Unable to import component: No module named 'sqlalchemy'
> Jan 07 16:44:26 OdroidN2 homeassistant[18332]: 2022-01-07 16:44:26 ERROR (MainThread) [homeassistant.setup] Error during setup of component http
> ```
> 
> It's these `no module named` ones, happening on any random module for some reason until restart. But in your case it's:
> 
> ```
> recorder: Unable to import component: cannot import name 'event' from partially initialized module 'sqlalchemy'
> ```
> 
> Can you paste your:
> 
> ```shell
[homeassistant log2.txt](https://github.com/MichaIng/DietPi/files/7829975/homeassistant.log2.txt)

> cat /mnt/dietpi_userdata/homeassistant/deps/lib/python3.9/site-packages/sqlalchemy/__init__.py
> ```

here's the info!
[homeassistant log2.txt](https://github.com/MichaIng/DietPi/files/7829977/homeassistant.log2.txt)

thanks!

MichaIng, 2022-01-07:
Okay that is identical to mine, nothing corrupted, so no circular import within this file 🤔. And you already tried reinstall and uninstall + reinstall, so unlikely that any other random corruption is the reason. However, since I'm otherwise out of ideas, can you check kernel logs:
```sh
dmesg -l emerg,alert,crit,err
```

Joulinar, 2022-01-07:
Probably try to reboot the system.

MichaIng, 2022-01-07:
> Probably try to reboot the system.

Ah right, the no. 1 solution at least on Windows, worth to try before doing the time consuming reinstall.

theChef613, 2022-01-07:
> 

reinstalling and then using the RM command worked! :) thank you!

MichaIng, 2022-01-07:
Okay great, whatever the reason for the faulty `/mnt/dietpi_userdata/homeassistant` was in the first place 😄.

closed 2022-01-08 by MichaIng