MichaIng/DietPi #5148: Missing package leads to failed installation of MyMPD

j-m-hoffmann, 2022-01-08:
#### Required Information
- DietPi version | 7.9.3
- Distro version | bullseye
- Kernel version | 5.10.0
- SBC model | RPi3

#### Additional Information (if applicable)
- Software title | MyMPD

#### Expected behaviour
Install without a hitch

#### Actual behaviour
[image]

#### Extra details
can be fixed by installing the libpcre2-dev package

MichaIng, 2022-01-08:
Many thanks for your report.

Solved already. We migrated to the official myMPD APT repository: https://github.com/MichaIng/DietPi/pull/5122

Joulinar, 2022-01-08:
This is already know and a permanent fix will be provided on next release https://github.com/MichaIng/DietPi/issues/5115#issuecomment-1001251945

MichaIng, 2022-01-08:
Release has been done, issue solved.

closed 2022-01-08 by MichaIng