MichaIng/DietPi #5151: Can't install update

s4tch, 2022-01-09:
#### Details:
- Date           | Sun Jan  9 23:15:55 GMT 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | DietPi
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | `Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update
2. ...
#### Expected behaviour:

- dietpi should find updates
#### Actual behaviour:

[FAILED] DietPi-Update | Unable to get latest version.
[image]
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://security.debian.org/debian-security stretch/updates InRelease
Get:2 http://archive.raspbian.org/raspbian stretch InRelease [15.0 kB]
Hit:3 https://deb.debian.org/debian bullseye InRelease
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Ign:7 https://deb.debian.org/debian stretch InRelease
Hit:8 https://deb.debian.org/debian stretch-updates InRelease
Hit:9 https://deb.debian.org/debian stretch Release
Err:2 http://archive.raspbian.org/raspbian stretch InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9165938D90FDDD2E
Reading package lists...
W: GPG error: http://archive.raspbian.org/raspbian stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 9165938D90FDDD2E
E: The repository 'http://archive.raspbian.org/raspbian stretch InRelease' is not signed.
```

Joulinar, 2022-01-09:
There is something wrong on your source files. You have a mixture of Debian Stretch and Debian Bullseye. Did you manually adjusted them? Or did you tried to upgrade the destro?

MichaIng, 2022-01-09:
Also `curl` is missing:
```sh
apt install curl
```

s4tch, 2022-01-09:
Thanks for the quick response! I'll try to do that tomorrow. Cheers

MichaIng ***@***.***> ezt írta (időpont: 2022. jan. 10., H
0:31):

s4tch, 2022-01-10:
I reflashed my card and update works fine now. Thanks again for the quick reply yesterday.

closed 2022-01-10 by s4tch