MichaIng/DietPi #5153: hotspot

progamer562, 2022-01-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
- Distro version | bullseye 1
- Kernel version | `uname -a`
- SBC model | RPi 4 Model B (armv7l) 
- Power supply used | raspberry pi 
- SDcard used | unkown

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

hotspot 

- Was the software title installed freshly or updated/migrated?

freshly

- Can this issue be replicated on a fresh installation of DietPi?

49ae9ad4-27db-403b-8dc2-472295a80158
i think?

- Bug report ID | `echo $G_HW_UUID`

49ae9ad4-27db-403b-8dc2-472295a80158

#### Expected behaviour

giving me wifi

#### Actual behaviour

not starting after reboot gives an error  

#### Extra details

- ...

closed 2022-01-10 by progamer562

Joulinar, 2022-01-10:
maybe you like to share how you solved your issue.

MichaIng, 2022-01-10:
Generally to debug issues with it, the following can help:
```sh
journalctl -u hostapd -u isc-dhcp-server
```