MichaIng/DietPi #5154: Can't install docker using dietpi-software

s4tch, 2022-01-10:
#### Details:
- Date           | Mon Jan 10 10:15:25 GMT 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | DietPi
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | `Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -qq --allow-change-held-packages install docker-ce`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. dietpi-software
2. select docker from the list, install
#### Expected behaviour:

- docker should be installed
#### Actual behaviour:

- see log + screenshot please. (clean install, formatted-reflashed dietpi allo gui image, zero tweaking, just updated to dietpi v8.0.2 before that)
[image]

#### Extra details:

- ...
#### Additional logs:
```
Setting up docker-ce (5:20.10.12~3-0~debian-bullseye) ...
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Mon 2022-01-10 10:15:25 GMT; 44ms ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
    Process: 3689 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
   Main PID: 3689 (code=exited, status=1/FAILURE)
dpkg: error processing package docker-ce (--configure):
 installed docker-ce package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2022-01-10:
Hi,

this is same issue as this one. https://github.com/MichaIng/DietPi/issues/4480
Reason is the pretty old kernel if I'm not mistaken.

s4tch, 2022-01-10:
Thank you. So bottom line is that sparky is unsupported, right?

Joulinar ***@***.***> ezt írta (időpont: 2022. jan. 10., H
11:42):

MichaIng, 2022-01-10:
Just to be sure whether it's the same issue:
```sh
journalctl -u docker -u containerd
```
But yes, the kernel lacks some features now required for Docker. Sadly there are also not much older packages available anymore: https://download.docker.com/linux/debian/dists/bullseye/pool/stable/armhf/
And no chance to get a newer Linux version running on Sparky SBC from current view.

Looks like we need to disable the Docker install option on Sparky SBC accordingly. If you have any chance to install the software you want to run inside the container natively on the host system, that would be the way to go, we may be able to help with this. At least from memory usage point of view it's preferable anyway.

s4tch, 2022-01-10:
-- Journal begins at Mon 2022-01-10 11:23:28 GMT, ends at Mon 2022-01-10 12:43:56 GMT. --
Jan 10 11:23:35 DietPi systemd[1]: Starting containerd container runtime...
Jan 10 11:23:35 DietPi modprobe[936]: modprobe: FATAL: Module overlay not found in directory /lib/modules/3.10.38
Jan 10 11:23:35 DietPi systemd[1]: Starting Docker Application Container Engine...
Jan 10 11:23:38 DietPi containerd[943]: time="2022-01-10T11:23:38.930375001Z" level=info msg="starting containerd" revision="1.4.12~ds1-1~deb11u1" version="1.4.12~ds1"
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.222228751Z" level=info msg="loading plugin \"io.containerd.content.v1.content\"..." type=io.containerd.content.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.230296751Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.aufs\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.243248001Z" level=info msg="skip loading plugin \"io.containerd.snapshotter.v1.aufs\"..." error="aufs is not supported (modprobe aufs failed: exit status 1 \"modprobe: FATAL: Module aufs not found in directory /lib/modules/3.10.38\\n\"): skip plugin" type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.243538751Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.btrfs\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.245587001Z" level=info msg="skip loading plugin \"io.containerd.snapshotter.v1.btrfs\"..." error="path /var/lib/containerd/io.containerd.snapshotter.v1.btrfs (ext4) must be a btrfs filesystem to be used with the btrfs snapshotter: skip plugin" type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.245800751Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.devmapper\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.246090001Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.246233251Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.native\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.252311501Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.overlayfs\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.258838251Z" level=info msg="loading plugin \"io.containerd.snapshotter.v1.zfs\"..." type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.259991251Z" level=info msg="skip loading plugin \"io.containerd.snapshotter.v1.zfs\"..." error="path /var/lib/containerd/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter: skip plugin" type=io.containerd.snapshotter.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.260193501Z" level=info msg="loading plugin \"io.containerd.metadata.v1.bolt\"..." type=io.containerd.metadata.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.260432001Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.260554751Z" level=info msg="metadata content store policy set" policy=shared
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.279733251Z" level=info msg="loading plugin \"io.containerd.differ.v1.walking\"..." type=io.containerd.differ.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.280494751Z" level=info msg="loading plugin \"io.containerd.gc.v1.scheduler\"..." type=io.containerd.gc.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.282087001Z" level=info msg="loading plugin \"io.containerd.service.v1.introspection-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.282684001Z" level=info msg="loading plugin \"io.containerd.service.v1.containers-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283016501Z" level=info msg="loading plugin \"io.containerd.service.v1.content-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283185751Z" level=info msg="loading plugin \"io.containerd.service.v1.diff-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283374251Z" level=info msg="loading plugin \"io.containerd.service.v1.images-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283547251Z" level=info msg="loading plugin \"io.containerd.service.v1.leases-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283809001Z" level=info msg="loading plugin \"io.containerd.service.v1.namespaces-service\"..." type=io.containerd.service.v1
Jan 10 11:23:39 DietPi containerd[943]: time="2022-01-10T11:23:39.283973001Z" level=info msg="loading plugin \"io.containerd.service.v1.snapshots-service\"..." type=io.containerd.service.v1
[... cut ...]
Jan 10 11:24:10 DietPi systemd[1]: docker.service: Start request repeated too quickly.
Jan 10 11:24:10 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Jan 10 11:24:10 DietPi systemd[1]: Failed to start Docker Application Container Engine.
Jan 10 11:24:22 DietPi systemd[1]: docker.service: Start request repeated too quickly.
Jan 10 11:24:22 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
Jan 10 11:24:22 DietPi systemd[1]: Failed to start Docker Application Container Engine.

s4tch, 2022-01-10:
Thanks for the replies. All I'd want besides the basic DietPi features is to install the tidal connect service.

MichaIng, 2022-01-10:
The same `devicemapper` error indeed.

Not sure whether it's a feasible alternative, but `O!MPD` has Tidal support included, maybe other media streaming servers as well.

s4tch, 2022-01-10:
Thanks @MichaIng , unfortunately I couldn't get docker working. Sparky might not be supported after all.

Joulinar, 2022-01-11:
I'm going to close the issue. Feel free to reopen if needed.

closed 2022-01-11 by Joulinar