MichaIng/DietPi #5155: Manual update to Bullseye doesn't allow for update to 8.0

Kreeblah, 2022-01-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=4
G_GITBRANCH='stretch'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye`
- Kernel version | `uname -a`
`Linux odroidc2-2 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`Odroid C2 (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
5V supply
- SDcard used | (EG: SanDisk ultra)
SanDisk, I think?

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
I updated my SBC from Stretch to Buster, and then from Buster to Bullseye using the instructions at https://dietpi.com/docs/usage/#how-to-upgrade-to-buster and then https://dietpi.com/blog/?p=811#2.2-manual-upgrade

Running `cat /etc/os-release`, I get the following:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

#### Expected behaviour
I'd expect to see the login banner report the system as being on Bullseye, and therefore to be able to upgrade to DietPi 8.x.

#### Actual behaviour
I am, instead, seeing it still report as being on Stretch, and 7.9.4 is reported as the latest DietPi release when I run `dietpi-update`.

#### Extra details
I thought updating `G_GITBRANCH` in `/boot/dietpi/.version` might help, but it didn't.  I'm still seeing the following when running `dietpi-update`:

`[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/stretch/.update/version`

Joulinar, 2022-01-11:
I guess this is an expected behaviour since we migrated all Stretch systems into a separat branch and you would need to switch back to master manually. Something @MichaIng could give a definite answer on 😃 

Usually the following command will change back to master

```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```

MichaIng, 2022-01-11:
We should add this step to the Stretch => Buster howto.

Kreeblah, 2022-01-12:
Thanks for updating that!

closed 2022-01-12 by Kreeblah