MichaIng/DietPi #5156: Unable to complete the DietPi prep script

KalixtShawxo, 2022-01-11:
Hi guys. 
I was trying to run the prep script on a clean install of ARMBIAN 21.08.1 FOR THE Cubieboard2, focal current 5.10.60 version. Created 26 Aug 2021. 
I did the apt update and also the install of curl etc. 
Ran the script per the instructions and when it started to get the files I got errors for the public key signatures. 
Saying they could not be verified. 
[image]
Is there an issue somewhere with the script or the files, or did I screw up in the process? 

Thanks in advance.

Joulinar, 2022-01-11:
Could you share a link of the image used? Because `focal` is an Ubuntu release but DietPi did support Debian only.

KalixtShawxo, 2022-01-11:
Damn it. I knew I screwed up. 
Sigh, I didn't realize the focal version was Ubuntu. 
I've been told already that DietPi won't do Ubuntu. 
I will have to look for the other version. 
I got it from armbian.hosthatch.com/archive/Cubieboard2/archive. 

I'll get these variant namings down pat sooner or later. 

Thanks for the quick reply.

Joulinar, 2022-01-11:
You would need to have a look for Buster or Bullseye

KalixtShawxo, 2022-01-11:
Thanks. I found the right version in Bullseye, I have the update running now. Let this be a lesson in attention to detail, stupid little mistakes. 
I started with Ubuntu on some projects, then switched over to DietPi/Debian and I still get confused every now and again.  Guess I gotta slow down. 

Again, thank you for the quick response pointing me back in the right direction. 
I am trying to use DietPi with the Cubieboard2 for a Plex Server dedicated to displaying pictures, that I want to store on a SATA drive since the Cubieboard2 has an onboard SATA.  We'll see if I can make it work.

Joulinar, 2022-01-11:
ok I'm going to close this one. Feel free to reopen if needed.

closed 2022-01-11 by Joulinar