MichaIng/DietPi #5157: Mosquitto: Websockets not working (possible wrong version of Libwebsocket.so library?)

ilgrank, 2022-01-11:
- DietPi version |
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
- SBC model | Virtualbox VM
- Power supply used | na
- SDcard used | na

#### Additional Information (if applicable)
- Software title | Mosquitto
- Was the software title installed freshly or updated/migrated? **Freshly installed**
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

#### Steps to reproduce

1. edit /etc/mosquitto/mosquitto.conf
2. add the following lines:
```
listener 8081
protocol websockets
```
and then start/restart the mosquitto service
#### Expected behaviour
Mosquitto should start, listening both to the 1833 (mqtt) and 8081 (websockets) ports

#### Actual behaviour
the service fails to start:
```
Jan 11 14:22:20 DietPi systemd[1]: Starting Mosquitto MQTT Broker (DietPi)...
Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: mosquitto version 2.0.12 starting
Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: Config loaded from /etc/mosquitto/mosquitto.conf.
Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: Opening ipv4 listen socket on port 1883.
Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: Opening ipv6 listen socket on port 1883.
Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: Opening websockets listen socket on port 8085.
**Jan 11 14:22:20 DietPi mosquitto[3340]: 1641907340: Error: Unable to create websockets listener on port 8085.**
Jan 11 14:22:20 DietPi systemd[1]: mosquitto.service: Main process exited, code=exited,
status=1/FAILURE
Jan 11 14:22:20 DietPi systemd[1]: mosquitto.service: Failed with result 'exit-code'.
**Jan 11 14:22:20 DietPi systemd[1]: Failed to start Mosquitto MQTT Broker (DietPi).**
```

#### Extra details
**possibly** related to: [https://github.com/eclipse/mosquitto/issues/1924](https://github.com/eclipse/mosquitto/issues/1924) ?

MichaIng, 2022-01-11:
Many thanks for your report.

The `libwebsockets18` package + all Mosquitto packages come from their official APT repository, so that should actually match. There seems to be a new version available (ah no, a 2.0.14 package is present, but not in the package lists 🤔), just to rule that our, can you do a package upgrade:
```sh
apt update
apt upgrade
```
And then show the package versions so that we can see whether the expected ones from Mosquitto repo are installed:
```sh
dpkg -l mosquitto libmosquitto1 libwebsockets18
```
Also does the verbosity flag reveal more details?
```sh
mosquitto -vc /etc/mosquitto/mosquitto.conf
# Hit CTRL+C to stop
```

ilgrank, 2022-01-11:
running mosquito directly, I get:
```
root@DietPi:~/tempquitto# mosquitto -v -c /etc/mosquitto/mosquitto.conf
1641908747: Error: Unable to open log file /var/log/mosquitto.log for writing.
```

MichaIng, 2022-01-11:
🤔 that should be disabled (so that logs are sent to journal instead), otherwise fail within the service the same way. Can you show the full:
```sh
cat /etc/mosquitto/mosquitto.conf
```
Probably running it as root somehow behaves differently, else try:
```sh
sudo -u mosquitto mosquitto -vc /etc/mosquitto/mosquitto.conf
```

ilgrank, 2022-01-11:
```
# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.gz

persistence true
persistence_location /var/lib/mosquitto/
#plugin /usr/local/lib/mosquitto_timestamp.so
allow_anonymous false
password_file /etc/mosquitto/passwords.txt

log_dest file /var/log/mosquitto.log
include_dir /etc/mosquitto/conf.d

listener 1883
protocol mqtt

listener 8081
protocol websockets
#http_dir /home/mosquitto/http/
certfile /root/.acme.sh/myhost.net/myhost.net.cer
keyfile /root/.acme.sh/myhost.net/myhost.net.key
cafile /root/.acme.sh/myhost.net/ca.cer

```

ilgrank, 2022-01-11:
truth to be told, I begin to think this could be a permission issue.
I uninstalled mosquitto, upgraded dietpi from 7.xx to 8.02 and re-installed a fresh mosquitto
I imported the old config, and the problem seems related to the initialization of the SSL Websockets port
removing the certs part, (so using Websockets without SSL) I have no errors.

Joulinar, 2022-01-11:
but this is not the config file we create on installation. As we do not ship the entire `cert` part and we remove the `log_dest` part as well

```
[  OK  ] DietPi-Software | sed -i /^[[:blank:]]*pid_file[[:blank:]]/d /etc/mosquitto/mosquitto.conf
[  OK  ] DietPi-Software | rm -R /run/mosquitto
[  OK  ] DietPi-Software | sed -i /^[[:blank:]]*log_dest[[:blank:]]/d /etc/mosquitto/mosquitto.conf
[  OK  ] DietPi-Software | rm -R /var/log/mosquitto
[  OK  ] DietPi-Software | Added setting password_file /etc/mosquitto/passwd to end of file /etc/mosquitto/mosquitto.conf
[  OK  ] DietPi-Software | Added setting listener 1883 to end of file /etc/mosquitto/mosquitto.conf
```

This is our default `mosquitto.conf`

```
root@DietPi4:~# cat /etc/mosquitto/mosquitto.conf
# Place your local configuration in /etc/mosquitto/conf.d/
#
# A full description of the configuration file is at
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example

persistence true
persistence_location /var/lib/mosquitto/

include_dir /etc/mosquitto/conf.d
password_file /etc/mosquitto/passwd
listener 1883
root@DietPi4:~#
```

Can you share your install log pls?

ilgrank, 2022-01-11:
As I wrote above, I used the config from a previous install

MichaIng, 2022-01-11:
Ah I was already wondering, so it was not `Freshly installed` as you stated 😉. Okay great that it works now. Since Mosquitto runs as user `mosquitto`, you need to assure that it has read permissions for certificate/key at `/root/.acme.sh/myhost.net/`. As it is within the `root` user's home directory, do NOT ever think about granting it direct access there, but instead either create the cert outside of there or copy them to `/etc/mosquitto/` with strict permissions so that (only) `mosquitto` has read permissions to them.

closed 2022-01-11 by ilgrank

MichaIng, 2022-01-11:
I just see we should add the `password_file` and `listener` directives above the `include_dir` so that customisations can be done in `/etc/mosquitto/conf.d`. Though if those directives are present already, they are left untouched to not change any customisations to `mosquitto.conf`. However, as we unset the log file and remove the logging dir, when copying a custom config afterwards, the missing logging dir can be the result, as the service is not permitted to pre-create it directly in `/var/log` + it doesn't even try as of https://github.com/MichaIng/DietPi/issues/5157#issuecomment-1009980867.