MichaIng/DietPi #5158: Can't install softwares

hellotinh03, 2022-01-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.0.2
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | CY-0530 5V 3A 15W
- SDcard used | USB Apacher AH354 16GB 3.0

#### Additional Information (if applicable)
- Software title | LXDE, TigerVNC (in this particular case)
- Was the software title installed freshly or updated/migrated? | Haven't been installed yet.
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | b3bff170-6cc0-4a37-a56a-680ac2e5a49f

#### Steps to reproduce

1. Install DietPi
2. Choose softwares in the DietPi Software menu (in this case LXDE and TigerVNC, I tried the DietPi installation 3 times before and it all ended up like this)
3. Choose install software
4. The thing happened

#### Expected behaviour

- LXDE and TigerVNC are installed

#### Actual behaviour

- This:
`E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libgl1-mesa-dri_20.3.5-1%2brpt3%2brpi1_arm64.deb  Error reading from server. Remote end closed connection [IP: 93.93.130.212 443]
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/x/xorg-server/xserver-xorg-core_1.20.11-1%2brpt1_arm64.deb  Error reading from server. Remote end closed connection [IP: 93.93.135.117 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?` 

#### Extra details

- I did tried `apt-get update` and then running the previous command with `--fix-missing`, ended up as the log below.
- Terminal log: 
[log.txt](https://github.com/MichaIng/DietPi/files/7851229/log.txt)

Joulinar, 2022-01-12:
Hi,

I did a test installation without issues. Probably there was a temp challenge with the Raspbian package server. I would recommend to try again.

hellotinh03, 2022-01-13:
> Hi,
> 
> I did a test installation without issues. Probably there was a temp challenge with the Raspbian package server. I would recommend to try again.

I did a fresh install again, yet I still encounter that same issue
`E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libglapi-mesa_20.3.5-1%2brpt3%2brpi1_arm64.deb  Connection timed out [IP: 46.235.231.111 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?`

Joulinar, 2022-01-13:
It's a different package compare to yesterday. For me it looks like network issues as you have `Connection timed out` messages. Means, connection to the internet seems to be flaky. You are using Ethernet or WiFi to connect to your local network?

MichaIng, 2022-01-13:
Can you run the following commands and show the output:
```sh
cd /tmp
time wget 'https://archive.raspberrypi.org/debian/pool/main/m/mesa/libglapi-mesa_20.3.5-1+rpt3+rpi1_arm64.deb'
apt install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri
 mesa-utils mesa-utils-extra xserver-xorg-video-fbdev
```
Looks like a slow connection to the particular RPi repo server(s), as this is not a CDN/mirror director.

Joulinar, 2022-01-13:
keep an eye on this slow network connection to Raspbian package server. It might hit you again next time.

MichaIng, 2022-01-13:
> Raspbian package server

Raspberry Pi package server 😉: `raspbian.raspberrypi.org` (Raspbian) != `archive.raspberrypi.org` (RPi firmware/kernel repo)