MichaIng/DietPi #5159: can´t install Docker

peaceman1920, 2022-01-12:
#### Details:
- Date           | Wed Jan 12 11:21:23 CET 2022
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux rasp4-DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Get:5 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Reading package lists...
E: Release file for https://archive.raspberrypi.org/debian/dists/bullseye/InRelease is not valid yet (invalid for another 3min 5s). Updates for this repository will not be applied.
```

Joulinar, 2022-01-12:
Hi,

your system time is out of sync. Pls try to perform a manual sync and try again.

```
/boot/dietpi/func/run_ntpd 1
apt update
apt upgrade
```

peaceman1920, 2022-01-12:
Hi,
it worked! You´re amazing and fast as lightning. 
How can I avoid this obstacle in the future?
best regards

Joulinar, 2022-01-12:
usually time sync should be done automatically on boot and once a day during night. if needed this could be changed to Boot + Hourly. You could check time sync done as follow

```
journalctl -u systemd-timesyncd.service
```

keep an eye on your system time if it stay in sync or if it is drifting. You can use `date` command to do so. There might be other issues if it is drifting to much.

peaceman1920, 2022-01-12:
many thanks! Have a nice day
it was the best idea to switch to DietPi

closed 2022-01-12 by peaceman1920