MichaIng/DietPi #5160: Left of Linux screen is to the left of the leftmost part of physical screen

chjan, 2022-01-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...
Just installed DietPi64 on a Raspberry Pi 3b+. I am also a Linux newbie that can be corrected!

1. Booting, the text start to the left of the left screen physical border. I See e.g. DietPi:"# and "wn" for unknown. The HDMI screen is a Samsung 4k TV. Is there a remedy to put in a startupscript or similar to get rid of this when the TV cannot be changed for its screen resolution which is 2880x or something. I did not find a suitable resolution in Linux screen setup. Tried them all.

2. While here, please allow that I ask you please to show me the command to start RoonBridge, which I supposedly installed successfully. 

I tried many things, like: sudo systemctl roon or roonbridge or RoonBridge, etc. The result is "Unknown command verb". What is the exact phrase? After googling around I could not find out. And since the TV screen isnt that cooperative, it is difficult to study the list of installed services.

Thanks.

Joulinar, 2022-01-12:
Going to link the related forum issue https://dietpi.com/phpbb/viewtopic.php?t=9914

Joulinar, 2022-01-12:
By design, a Unix OS will never respond to a command if it was accepted and is correct. To see the status of a service, just do following

```
systemctl status roonbridge.service
journalctl -u roonbridge.service
```

MichaIng, 2022-01-12:
On RPi, you can use `dietpi-config` > Display Options > Overscan to move the output into physical screen borders. This is a natural issue with TVs which do not have a 1:1 pixel mapping, especially when the native resolution of the TV is above the max resolution of the hardware, 1080p in case of RPi 3.

MichaIng, 2022-01-12:
> I assume that any firewall system is handled by the installation of Roon (?).

No, generally software installs do not automatically configure any firewall, often they simply cannot. If you install a firewall (there is none pre-installed on DietPi), either on the server itself or another level of your network, you must take care to keep open required ports and allow the application to use them yourself.

You should be btw able to see the current framebuffer resolution via `tvservice -s` command. I bet it's 1080p. Then try to use mentioned overscan options to move and stretch/shrink the display to fit on the TV. To not change the aspect ratio, try to keep overscan of combined Y to combined X changes at 16 to 9.

chjan, 2022-01-12:
No need for firewall in this case. Now screen is within physical border. Roon still does not show up.  Status says it is running. Back later if I cannot sort it out.

MichaIng, 2022-01-12:
Ah yes, I see I misunderstood and the command output was not outside of the screen but simply no output was done as you expected. Display quality can still be enhanced on TVs using overscan, or editing `/boot/config.txt` manually, setting `hdmi_group` and `htmi_mode` for intended resolution: https://www.raspberrypi.com/documentation/computers/config_txt.html#hdmi_group

How do you expect the Roon Bridge to show up on the RPi? It is a service, running in background, which simply listens to inputs from a Roon application controlled on another system. So you have the GUI on e.g. Windows to control the player on the RPi.

chjan, 2022-01-12:
Long time since I used Roon, back at it recently. Thus, did not grasp that a bridge does not show up in GUI. Please forget that.

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng