MichaIng/DietPi #5161: Can't seem to update.

shanew1694, 2022-01-12:
### Creating a bug report/issue

#### Required Information
- **DietPi version** | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=4
G_GITBRANCH='stretch'
G_GITOWNER='MichaIng'

- **Distro version** | Unknown 0
- **Kernel version** | Linux DietPi 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux
- **SBC model** | RPi 4 Model B (aarch64)
- **Power supply used** | Official RPI4 Powe supply
- **SDcard used** | SanDisk ultra

#### Additional Information (if applicable)
- Software title | dietpi-update

#### Expected behavior
The update should complete.

#### Actual behaviour
The update fails saying "Your Debian version is too low to update to the current branch: Unknown (0)"

[img]https://i.imgur.com/mg5Km6U.png[/img]

#### Extra details
I forced a branch change back to 'master' using "G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt"
but that doesn't do anything.

Joulinar, 2022-01-12:
did you run `dietpi-update` again after switching branch back to master?

shanew1694, 2022-01-12:
yeah I re-ran

Joulinar, 2022-01-12:
can you share `cat /boot/dietpi.txt | grep DEV`

shanew1694, 2022-01-12:
I think it's because my distro version is 'Unkown' and it keeps reverting back to the Stretch branch

DEV_GITBRANCH=stretch
DEV_GITOWNER=MichaIng

if i do "G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt" then the "DEV_GITBRANCH=master" but then i do dietpi-update the branch default back to stetch.

Joulinar, 2022-01-12:
what does following return `cat /etc/os-release `

shanew1694, 2022-01-12:
`cat /etc/os-release`
PRETTY_NAME="Debian GNU/Linux bullseye/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

MichaIng, 2022-01-12:
`bullseye/sid`, did you manually change the APT repositories? Can you show:
```
for i in /etc/apt/sources.list{,.d/*.list}
do
echo "List: $i"
cat "$i"
done
```

Joulinar, 2022-01-12:
ok this is quite confusing. This is showing `bullseye/sid`. This would mean Testing/Unstable repository. Can you give some background on your system? Did you upgrade the Debian version? Changed anything on apt source files?

MichaIng, 2022-01-12:
> This is showing `bullseye/sid`. This would mean Testing/Unstable repository.

Actually it means the the currently stable Bullseye, but the file (`base-files` package) is from a time when it was still the testing branch, hence more than half a year old. That shouldn't be possible with Debian APT lists properly applied since this has been updated several times and the `dietpi-update` implies an APT package upgrade. So something must be wrong with the lists.

Joulinar, 2022-01-12:
@MichaIng 
can you have to the attached screen print on first post. `dietpi-update` seems trying to switch to `jesssie-support` As well apt source seems to be Buster. Which would explain why base file not being updated.

shanew1694, 2022-01-12:
I did edit the source.list to add in a bookworm repo so I could upgrade qbittorrent-nox to 4.3.9. But that's the only thing I update and then commented it back out and did `apt-update`

But now doing looking at the source.list most of the repos are buster

```
List: /etc/apt/sources.list
deb https://deb.debian.org/debian/ buster main contrib non-free
deb https://deb.debian.org/debian/ buster-updates main contrib non-free
deb https://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb https://deb.debian.org/debian/ buster-backports main contrib non-free

#deb http://ftp.de.debian.org/debian bookworm main
List: /etc/apt/sources.list.d/dietpi-mono.list
deb https://download.mono-project.com/repo/debian/ buster main
List: /etc/apt/sources.list.d/raspi.list
deb https://archive.raspberrypi.org/debian/ buster main
List: /etc/apt/sources.list.d/sonarr.list
deb https://apt.sonarr.tv/debian/ buster main
```

MichaIng, 2022-01-12:
@Joulinar 
Yes, the "Unknown" Debian version has ID `0` and is hence interpreted as even earlier than Jessie. We should abort the update, or any script, when this ID was found, an print a meaningful error message 🤔.

@shanew1694 
> #deb http://ftp.de.debian.org/debian bookworm main

Seems you experimented with a manual upgrade at least 😉. But the lists look fine now. The problem is that the `base-files` package has a newer version already and is not automatically downgraded by APT. And I'm pretty sure its not the only package which is on that intermediate Bullseye stage already. I guess easiest would be to upgrade to Bullseye: https://dietpi.com/blog/?p=811

Otherwise, to downgrade `base-files` to Buster:
```sh
apt install --reinstall --allow-downgrades base-files=10.3+deb10u11
```
I just it however fails as of broken dependencies. To get a full picture of which packages may be Bullseye versions already:
```sh
dpkg -l | grep 'deb11'
```

shanew1694, 2022-01-12:
`dpkg -l | grep 'deb11'`
 returns only php7.4

Joulinar, 2022-01-12:
can you check `dpkg -l | grep base-files`

shanew1694, 2022-01-12:
```
ii  base-files                                                  11                                                        arm64        Debian base system miscellaneous files
```

MichaIng, 2022-01-12:
Not all package versions from Bullseye have this `deb11`. PHP pulls a bunch of libraries as dependencies, so if you required it for some reason, I indeed recommend to upgrade the full system to Bullseye.

shanew1694, 2022-01-12:
No didn't do that yet, Might break stuff i'll do a backup before i do that

shanew1694, 2022-01-12:
Ok, did a backup and updated everything to bullseye and successfully updated to v8.0.2.

Think it's all good. Throw a little error about me holding back deluged to 1.3.15 but apart from that we're good.

MichaIng, 2022-01-13:
Bullseye ships Deluge v2.0.3, probably probably it does not do the major upgrade automatically to prevent any backwards incompatibilities or so. You should be able to upgrade via:
```sh
dietpi-software reinstall 45
```
Looks like we need to add this to the list of software titles to reinstall, in our Bullseye upgrade blog post. But not sure whether all torrents and settings are migrated gracefully, as it's a major version upgrade.

shanew1694, 2022-01-13:
Oh no, it probably works fine. I'm manually holding back deluged to 1.3.15. 

I don't like v2, although I haven't used it since it was first released.

MichaIng, 2022-01-13:
Ah sorry, I misunderstood 😅. Then it should be all fine.

shanew1694, 2022-01-14:
Well, after the full update I've just realized my rutorrent no longer works.

'''
Bad response from server: (503 [error,getplugins]) Service Unavailable
'''

I've restarted all services and reboot and even tried a reinstall using `dietpi-software reinstall 107`

rtorrent seems to be running fine its just rutorrent throwing errors again. Why is it always rutorrent? :(

MichaIng, 2022-01-14:
Which webserver do you use? In case of Lighttpd (default), can you please try again to access ruTorrent and then check Lighttpd error logs:
```sh
journalctl -u lighttpd
cat /var/log/lighttpd/error.log
```

shanew1694, 2022-01-14:
Ah, seems to be trying php7.3pm, which I just upgraded to 7.4. 
Yes using Lighttpd.

Journalctl: (This repeats a few time)
```
Jan 14 14:31:19 DietPi systemd[1]: Started Lighttpd Daemon.
Jan 14 14:43:25 DietPi systemd[1]: Stopping Lighttpd Daemon...
Jan 14 14:43:25 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=1/FAILURE
Jan 14 14:43:25 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Jan 14 14:43:25 DietPi systemd[1]: Stopped Lighttpd Daemon.
Jan 14 14:43:25 DietPi systemd[1]: Starting Lighttpd Daemon...
Jan 14 14:43:25 DietPi lighttpd[125140]: 2022-01-14 14:43:25: mod_deflate.c.567) DEPRECATED: compress.filetype replaced with deflate.mimetypes
Jan 14 14:43:25 DietPi lighttpd[125140]: 2022-01-14 14:43:25: mod_deflate.c.580) DEPRECATED: compress.cache-dir replaced with deflate.cache-dir
Jan 14 14:43:25 DietPi systemd[1]: Started Lighttpd Daemon.
```

Cat:
```
2022-01-14 15:30:07: gw_backend.c.238) establishing connection failed: socket: unix:/run/php/php7.3-fpm.sock: No such file or directory
2022-01-14 15:30:07: gw_backend.c.970) all handlers for /rutorrent/php/getplugins.php? on .php are down.
2022-01-14 15:30:09: gw_backend.c.315) gw-server re-enabled: unix:/run/php/php7.3-fpm.sock  0 /run/php/php7.3-fpm.sock
2022-01-14 15:30:51: gw_backend.c.238) establishing connection failed: socket: unix:/run/php/php7.3-fpm.sock: No such file or directory
2022-01-14 15:30:51: gw_backend.c.970) all handlers for /rutorrent/php/getplugins.php? on .php are down.
2022-01-14 15:30:53: gw_backend.c.315) gw-server re-enabled: unix:/run/php/php7.3-fpm.sock  0 /run/php/php7.3-fpm.sock
```

MichaIng, 2022-01-14:
Ah, you need to reinstall Lighttpd, as stated in the [blog](https://dietpi.com/blog/?p=811#2.2.4-software-migrations), here the full list of software titles which need to be reinstalled to apply all required migrations:
```sh
dietpi-software reinstall 31 38 40 47 48 56 83 84 85 88 89 93 114 118 130 134 143
```
Otherwise of course do it manually for Lighttpd:
```
sed -i 's/7.3/7.4/' /etc/lighttpd/conf-available/15-fastcgi-php.conf
sed -Ei '/^compress\..*=[[:blank:]]*["(].*[")]$/d' /etc/lighttpd/lighttpd.conf
sed -i '/^[[:blank:]]*"mod_compress",$/d' /etc/lighttpd/lighttpd.conf
# And if you want deflate compression:
apt install lighttpd-mod-deflate
lighty-enable-mod deflate
systemctl restart lighttpd
```

shanew1694, 2022-01-14:
Yeah still not working. Same error on rutorrent page.

New error message in error.log

```
dietpi@DietPi:~$ sudo cat /var/log/lighttpd/error.log
2022-01-14 16:18:27: server.c.1976) server stopped by UID = 0 PID = 1
2022-01-14 16:19:09: server.c.1513) server started (lighttpd/1.4.59)
2022-01-14 16:19:54: server.c.1976) server stopped by UID = 0 PID = 1
2022-01-14 16:20:51: server.c.1513) server started (lighttpd/1.4.59)
2022-01-14 16:21:10: mod_fastcgi.c.487) FastCGI-stderr:PHP message: PHP Fatal error:  Cannot declare class jpopsukiEngine, because the name is already in use in /var/www/rutorrent/plugins/extsearch/engines/jpopsuki.php on line
2022-01-14 16:22:56: server.c.1976) server stopped by UID = 0 PID = 1
2022-01-14 16:23:44: server.c.1513) server started (lighttpd/1.4.59)
2022-01-14 16:23:52: mod_fastcgi.c.487) FastCGI-stderr:PHP message: PHP Fatal error:  Cannot declare class jpopsukiEngine, because the name is already in use in /var/www/rutorrent/plugins/extsearch/engines/jpopsuki.php on line
```

Joulinar, 2022-01-14:
Can you try to reboot

shanew1694, 2022-01-14:
rebooted same thing.

MichaIng, 2022-01-14:
Hmm, the issue may be that we merge the new `ruTorrent` files into the existing directory, to assure that manually installed plugins and such are preserved. But when file names changed, in theory it would be possible that new scripts load conflicting old scripts. Probably we need to preserve only the plugins explicitly (compare existing to freshly downloaded plugins directory and copy over (only) those which are missing in the fresh download) and otherwise remove the old `/var/www/rutorrent`. @shanew1694 if it is okay for you to have ruTorrent reset, including settings and plugins, the hammer method is:
```sh
rm -R /var/www/ruTorrent
dietpi-software reinstall 107
```
To preserve settings, these are the files to backup first:
```
/var/www/rutorrent/conf/config.php
/var/www/rutorrent/conf/access.ini
/var/www/rutorrent/conf/plugins.ini
```
+ in case manually installed plugins from `/var/www/rutorrent/plugins`.

Joulinar, 2022-01-14:
Probably rutorret would need to be removed cleanly and freshly installed

shanew1694, 2022-01-14:
So I moved `/var/www/rutorrent/plugins/extsearch/engines/jpopsuki.php` to `~/` for testing and now rutorrent works but all the plugins now fail as they require `_task`

MichaIng, 2022-01-14:
I made this a new issue: #5170

Marking this as closed now. Feel free to reopen when required.

closed 2022-01-14 by MichaIng

shanew1694, 2022-01-14:
So I'll post this here rather than in the new thread as I don't know if it's just my install or not.

I got a little more time to mess with things, so I did `rm -R /var/www/rutorrent` and reinstalled rtorrent/rutorrent. But I still had the same problem with all the plugins erroring. 

I then found this in rutorrent `config.php`:
```
$pathToExternals = array(
		"php" 	=> '',			// Something like /usr/bin/php. If empty, will be found in PATH.
		"curl"	=> "/usr/bin/curl",
		"gzip"	=> '',			// Something like /usr/bin/gzip. If empty, will be found in PATH.
		"id"	=> '',			// Something like /usr/bin/id. If empty, will be found in PATH.
		"stat"	=> '',			// Something like /usr/bin/stat. If empty, will be found in PATH.
	);
```

I guess when I upgraded the system to bullseye it can no longer find these in PATH?

I changed `config.php` to this and it fully works now:
```
$pathToExternals = array(
                "php"   => '/usr/bin/php',                      // Something like /usr/bin/php. If empty, will be found>
                "curl"  => '/usr/bin/curl',
                "gzip"  => '',                  // Something like /usr/bin/gzip. If empty, will be found in PATH.
                "id"    => '',                  // Something like /usr/bin/id. If empty, will be found in PATH.
                "stat"  => '',                  // Something like /usr/bin/stat. If empty, will be found in PATH.
                "pgrep" => '/usr/bin/pgrep',
        );
```

MichaIng, 2022-01-14:
Interesting. `/usr/bin` is in the PATH by default, so that shouldn't be required. Can you check the following:
```sh
grep 'PATH' /etc/php/7.4/fpm/pool.d/www.conf
php -r 'echo getenv(PATH);'
```
and the other way round with a test script:
```sh
echo -e '<?php\necho getenv('PATH');' > /var/www/test_path.php
```
and then opening that script in browser.

shanew1694, 2022-01-14:
```
dietpi@DietPi:~$ grep 'PATH' /etc/php/7.4/fpm/pool.d/www.conf
; Pass environment variables like LD_LIBRARY_PATH. All $VARIABLEs are taken from
;env[PATH] = /usr/local/bin:/usr/bin:/bin
```

```
dietpi@DietPi:~$ php7.4 -r 'echo getenv(PATH);'
PHP Warning:  Use of undefined constant PATH - assumed 'PATH' (this will throw an Error in a future version of PHP) in Command line code on line 1
/home/dietpi/.pyenv/plugins/pyenv-virtualenv/shims:/home/dietpi/.pyenv/shims:/home/dietpi/.pyenv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/gamesdietpi
```

```
dietpi@DietPi:~$ ls -l /usr/{sbin,local/{bin,sbin}}/{php,pgrep}
ls: cannot access '/usr/sbin/php': No such file or directory
ls: cannot access '/usr/sbin/pgrep': No such file or directory
ls: cannot access '/usr/local/bin/php': No such file or directory
ls: cannot access '/usr/local/bin/pgrep': No such file or directory
ls: cannot access '/usr/local/sbin/php': No such file or directory
ls: cannot access '/usr/local/sbin/pgrep': No such file or directory
```

```
dietpi@DietPi:~$ command -v php pgrep
/usr/bin/php
/usr/bin/pgrep
```

MichaIng, 2022-01-16:
Can you retry it with `www-data` user:
```sh
sudo -u www-data php8.1 -r "echo getenv('PATH') . PHP_EOL;"
```

shanew1694, 2022-01-16:
I assume you mean 7.4.

```
dietpi@DietPi:~$ sudo -u www-data php7.4 -r "echo getenv('PATH') . PHP_EOL;"
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

MichaIng, 2022-01-16:
Okay so all there, exactly as expected. When you create the test script as suggested above, and open it in browser, does it show this PATH as well?
```sh
echo -e "<?php\necho getenv('PATH');" > /var/www/test_path.php
```

shanew1694, 2022-01-16:
I never did that last time. But I just did `echo -e "<?php\necho getenv('PATH');" > /var/www/test_path.php` it now and nothing is shown.

i did `echo-e "<?php\necho ('Hello');" > /var/www/test_path.php` as a test and that worked.

MichaIng, 2022-01-16:
Strange works fine here, though tested with Apache2. Will test again with Lighttpd.

MichaIng, 2022-01-16:
Verified. With Apache2, the PATH is passed to PHP-FPM, with Lighttpd and Nginx it is not 🤔.

This is more than strange, actually the webserver shouldn't be able to influence this, due to:
```
; Clear environment in FPM workers
; Prevents arbitrary environment variables from reaching FPM worker processes
; by clearing the environment in workers before env vars specified in this
; pool configuration are added.
; Setting to "no" will make all environment variables available to PHP code
; via getenv(), $_ENV and $_SERVER.
; Default Value: yes
;clear_env = no
```
- So by default the environment is cleared and this is not changed when using a different webserver.
- So when PHP-FPM itself obviously does not fallback to a default PATH, how is Apache2 able to pass it?

@shanew1694 
A clean solution for you is:
```sh
G_CONFIG_INJECT 'env\[PATH\]' 'env[PATH] = /usr/local/bin:/usr/bin:/bin' /etc/php/7.4/fpm/pool.d/www.conf
```
This way a defined PATH without super user executables (`sbin`) is used.

shanew1694, 2022-01-16:
Just checked `/etc/php/7.4/fpm/pool.d/www.conf` before doing G_CONFIG_INJECT and file already has it

```
; Pass environment variables like LD_LIBRARY_PATH. All $VARIABLEs are taken from
; the current environment.
; Default Value: clean env
;env[HOSTNAME] = $HOSTNAME
;env[PATH] = /usr/local/bin:/usr/bin:/bin
;env[TMP] = /tmp
;env[TMPDIR] = /tmp
;env[TEMP] = /tmp
```

MichaIng, 2022-01-16:
Ah sorry, the `;` is not known as comment character by `G_CONFIG_INJECT`, remove it manually.

In Lighttpd you can have PATH passed via:
```sh
setenv.add-environment = ( "PATH" => env.PATH )
```
So basically `clear_env` means that the PHP-FPM server itself starts with an empty environment but does not prevent clients from passing environment variables together with scripts for handling. Makes sense. I'll re-add the static `/usr/local/bin:/usr/bin:/bin` path, which we had in the past already and which basically makes sense also to skip `sbin` executables in the first place.

MichaIng, 2022-01-16:
Fixed with: https://github.com/MichaIng/DietPi/commit/a96877a

shanew1694, 2022-01-16:
Yep all good now! 

Did `G_CONFIG_INJECT` and reverted `/var/www/rutorrent/conf/config.php` back to the DietPi default and all the errors are now gone.

MichaIng, 2022-01-16:
Great, thanks for the feedback 🙂. Thanks for reporting this btw, as this may affect other web applications as well. Good to know that Nginx and Lighttpd do not pass certain environment variables OOTB, even better to define the PATH explicitly.