MichaIng/DietPi #5162: apt search is really slow, > 10m for apt search git, on a rpi4

MaxPeal, 2022-01-12:
### Creating a bug report/issue
apt search git | wc -l
takes over 10 minutes

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
```
root@DietPi-01:~# /bin/bash -c "apt update ; time apt search git | wc -l"
Hit:1 https://download.docker.com/linux/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease                                                           
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease                                                   
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

5459

real	10m12.485s
user	10m7.046s
sys	0m4.608s
root@DietPi-01:~# 
```

#### Expected behaviour
```
root@DietPi-01:~# docker run --rm -it debian:11 /bin/bash -c "time apt update ; time apt search git | wc -l"
Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
Get:2 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [103 kB]
Get:5 http://deb.debian.org/debian bullseye/main arm64 Packages [8070 kB]
Get:6 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [2600 B]
Fetched 8375 kB in 3s (3050 kB/s)                         
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

real	0m4.473s
user	0m3.950s
sys	0m0.780s

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

5402

real	0m29.140s
user	0m25.508s
sys	0m3.637s
root@DietPi-01:~# 
```

Joulinar, 2022-01-12:
By default we have compression activated on apt. Means packages would need to be unzip first. This is to reduce r/w operation on SD card. If needed this you could be disabled. As well `APT cache` could be enabled to be stored on disk using `dietpi-config`

MichaIng, 2022-01-12:
Interesting, the cache has (nearly) no effect on full text search, probably it's not even used, but the compression has a massive effect. APT seems to handle natively compressed lists extremely ineffective. I mean decompression and re-compression with `tar` takes a fraction of a second.

`apt show git` btw works much better, still `22.743s` on an RPi 2 with compression enabled and cache disabled. That one would make use of the cache as well. But a full text search with 5459 results isn't so helpful usually 😄. However to disable list compression:
```sh
echo 'Acquire::GzipIndexes "false";' > /etc/apt/apt.conf.d/99no-compression
rm -R /var/lib/dpkg/lists
apt update
time apt search git | wc -l
```

Joulinar, 2022-01-12:
probably doing following could be good as well after disabling the compression

`/boot/dietpi/func/dietpi-set_software apt-cache clean`

MichaIng, 2022-01-12:
`rm -R /var/lib/dpkg/lists` is all which is required, but it is done by `/boot/dietpi/func/dietpi-set_software apt-cache clean` as well, among clearing cache and downloaded package archive.

MaxPeal, 2022-01-13:
mybe its still a problem because of the missing Acquire::CompressionTypes::Order?

https://debian-bugs-dist.debian.narkive.com/ivGMAvhK/bug-607135-gzipindexes-not-working-anymore-with-the-default-compression-order

MichaIng, 2022-01-13:
With `Acquire::GzipIndexes "on";` (default on DietPi as of https://github.com/MichaIng/DietPi/blob/a948d12/rootfs/etc/apt/apt.conf.d/97dietpi), it always stores lists compresses, in our case as `xz`. The [Debian](https://deb.debian.org/debian/dists/bullseye/main/binary-armhf/) and [Raspbian](http://raspbian.raspberrypi.org/raspbian/dists/bullseye/main/binary-armhf/) repositories and most others ship the lists `xz` compressed natively, so no decompression is done but they are kept as is. If the lists are uncompressed or differently compressed, they are in case decompressed and recompressed as `xz`, e.g. the case with the [RPi firmware repo](https://archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/), with ships only `gz`.

So I think the bug you linked has been fixed already.

The only question remains why compressed lists do SO much slowdown parsing, as the decompression takes actually not much time. I guess it is somehow done in chunks, maybe to reduce memory usage or so, but it is highly inefficient in terms of speed, obviously.

MichaIng, 2022-02-06:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-06 by MichaIng