MichaIng/DietPi #5164: Support for `CONFIG_SOUNDCARD=*-plug` in dietpi.txt on first-run (ALSA conversion plugin)

jwatt, 2022-01-13:
I have a script that updates the DietPi configuration after burning the image to disk. I noticed that `CONFIG_SOUNDCARD` in dietpi.txt does not support a `-plug` suffix during the initial setup on first run of DietPi. Specifically, I'm trying to set the value `usb-dac-plug` and it gets changed to `usb-dac` after first run. That means that it's necessary to run `dietpi-config` manually to enable the ALSA conversion plugin after first run, and **then** it is changed to `usb-dac-plug`. It would be really great if the first run stage would support setting this value directly. :-)

Thanks for all the awesome work on DietPi!

jwatt, 2022-01-13:
(FWIW I found out about the conversion plugin from [this post](https://dietpi.com/phpbb/viewtopic.php?t=8682) on the Dragonfly (which I'm also using).)

MichaIng, 2022-01-13:
Many thanks for your report.

Interesting, I would have expected the `-plug` suffix to work for first run step as well. I'll check back and in case fix this.

MichaIng, 2022-04-22:
It works well here. I set:
```sh
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=5
CONFIG_SOUNDCARD=hw:0,0-plug
```
And the auto-conversion plugin was successfully enabled 🤔.

MichaIng, 2022-04-28:
Marking this as closed, as it seems to work well. Feel free to reopen when required.

closed 2022-04-28 by MichaIng