MichaIng/DietPi #5165: DietPi-Software | dietpi-dashboard:rename a blank file, delete it

IIMustangII1151, 2022-01-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 
 bullseye

- Kernel version | 
5.10.0-10-amd64 #1 SMP Debian 5.10.84-1

- SBC model | 
Native PC (x86_64)

- Power supply used |
battery laptop
- SDcard used | 
HDD 160gb

#### Additional Information (if applicable)
- Software title | dietpi-dashboard
- Was the software title installed freshly or updated/migrated?
installed freshly
- Can this issue be replicated on a fresh installation of DietPi?
yes
535e7b02-4f83-4c3e-a095-1a76da643fa2

#### Steps to reproduce

1. renaming a file, leaving the name blank, in dietpi-dashboard using file browser deletes the file from the system, even canceling the rename window

#### Expected behaviour

-renaming a file

#### Actual behaviour

- deletes the file from the system

#### Extra details

- ...

ravenclaw900, 2022-01-14:
Hi,
Please report issues with DietPi-Dashboard at https://github.com/ravenclaw900/DietPi-Dashboard.

Joulinar, 2022-01-14:
Going to close this issue as it has been reported on Dashboard GitHub https://github.com/ravenclaw900/DietPi-Dashboard/issues/117

closed 2022-01-14 by Joulinar